greatscottgadgets/ubertooth #47: installing ubertooth- kali linux

erlichsefi, 2015-01-06:
i'm tring to install ubertooth on my vm kali linux. i followed the steps one by one. 
first i had a problam with livusb:

"CheckFunctionExists.c:(.text+0x15): undefined reference to `pcap_createsrcstr'
make[1]: Leaving directory `/root/ubertooth-2014-02-R2/host/build/CMakeFiles/CMakeTmp'
collect2: error: ld returned 1 exit status
make[1]: *** [cmTryCompileExec1385275149] Error 1
make: *** [cmTryCompileExec1385275149/fast] Error 2
"

so i installed manually libusb 1.0.9.

now i'm have other error in compilling.
the error:
root@kali:~/ubertooth-2014-02-R2/host/build# cmake   .. 
FATAL Pcap include dirs cannot be found
FATALPcap library cannot be found
-- Could NOT find PCAP (missing:  PCAP_INCLUDE_DIRS PCAP_LIBRARIES) 
FATAL Pcap include dirs cannot be found
FATALPcap library cannot be found
-- Could NOT find PCAP (missing:  PCAP_INCLUDE_DIRS PCAP_LIBRARIES) 
-- Configuring done
-- Generating done
-- Build files have been written to: /root/ubertooth-2014-02-R2/host/build
root@kali:~/ubertooth-2014-02-R2/host/build# make
[  7%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
/root/ubertooth-2014-02-R2/host/libubertooth/src/ubertooth.c:30:33: fatal error: bluetooth_le_packet.h: No such file or directory
compilation terminated.
make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o] Error 1
make[1]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
make: *** [all] Error 2
 there is solution?

dominicgs, 2015-01-06:
Which version of Ubertooth are you trying to build?  Are you using the same version of libbtbb?

That first error doesn't look like a libusb error, it references pcap.  Are you sure installing libusb fixed that issue?

The second error is because you need to install libpcap-dev.

The third error is because libbtbb doesn't appear to be installed.

erlichsefi, 2015-01-06:
u where right. i installed libcap-dev.1.62 , ans installed again libbtub with the same version and it compiiled.

"
Scanning dependencies of target specan
[ 92%] Generating build/timestamp
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/specan
copying /root/ubertooth-2014-02-R2/host/python/specan_ui/specan/__init__.py -> build/lib.linux-x86_64-2.7/specan
copying /root/ubertooth-2014-02-R2/host/python/specan_ui/specan/Ubertooth.py -> build/lib.linux-x86_64-2.7/specan
[ 92%] Built target specan
Scanning dependencies of target ubertooth-specan-ui
[ 92%] Built target ubertooth-specan-ui
Scanning dependencies of target dfu
[100%] Generating build/timestamp
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/dfu
copying /root/ubertooth-2014-02-R2/host/python/usb_dfu/dfu/dfu.py -> build/lib.linux-x86_64-2.7/dfu
copying /root/ubertooth-2014-02-R2/host/python/usb_dfu/dfu/__init__.py -> build/lib.linux-x86_64-2.7/dfu
[100%] Built target dfu
Scanning dependencies of target ubertooth-dfu
[100%] Built target ubertooth-dfu"

just one thing, i'm tring to run the scan and as u can see:
root@kali:~# ubertooth-btle -f -c /tmp/pipe
bash: ubertooth-btle: command not found

***thank you for the fast answr  and for a patience for a rookie like me:)

dominicgs, 2015-01-06:
You need to run ```make install``` to install the binaries.  You may also need to run ```ldconfig``` if it complains about finding the libraries.

erlichsefi, 2015-01-07:
dominic you are amazing! 

and another one :+1: 

"root@kali:~# ubertooth-btle -f -c /tmp/pipe
control message unsupported
libUSB Error: Command Error:  (-9)
USB error"

erlichsefi, 2015-01-07:
i just now finish installing again all on a live kali ( bacuse i was reading the ubertooth doesnt working in wm, and before i installed on vm kali). 
the steps to fix all the 3 problam the steps was: installin usblib1.0.6--> libpcap1.6 .

but!! in the vm && live kali it show the same problam: 

"root@kali:~# ubertooth-btle -f -c /tmp/pipe
control message unsupported
libUSB Error: Command Error: (-9)
USB error"

please...

dominicgs, 2015-01-07:
Have you updated the firmware on the Ubertooth?  This looks like the Ubertooth doesn't support a command that is being sent to it.

Unfortunately I don't know much about ubertooth-btle.

erlichsefi, 2015-01-07:
does ubertooth supposed to work on wm?

mikeryan, 2015-01-07:
Can you paste the output of ubertooth-util -v

erlichsefi, 2015-01-07:
on live-linux( not kali):
libUSB Error: Insufficient Permissions:  (-3)
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One
Usage:
	-a[0-7] get/set power amplifier level
	-b get hardware board id number
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-d[0-1] get/set all LEDs
	-e start repeater mode
	-f activate flash programming (DFU) mode
	-h display this message
	-i activate In-System Programming (ISP) mode
	-I identify ubertooth device by flashing all LEDs
	-l[0-1] get/set USR LED
	-m display range test result
	-n initiate range test
	-p get microcontroller Part ID
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-r full reset
	-s get microcontroller serial number
	-S stop current operation
	-t intitiate continuous transmit test
	-U<0-7> set ubertooth device to use
	-v get firmware revision number
	-V get compile info
	-z set squelch level

mikeryan, 2015-01-07:
Ah, looks like the udev script didn't install correctly. Can you paste the output of ```cat /etc/udev/rules.d/40-ubertooth.rules``` and ```id```.

erlichsefi, 2015-01-07:
1 etc/udev/rules.d/40-ubertooth.rules: No such file or directory.
2  uid=999(ubuntu) gid=999(ubuntu) groups=999(ubuntu),4(adm),24(cdrom),27(sudo),30(dip),46(plugdev),108(lpadmin),124(sambashare)

mikeryan, 2015-01-07:
Are you installing 2014-04-R1? From the source tarball, edit the file ```host/libubertooth/40-ubertooth.rules```, replacing ```GROUP="usb"``` with ```GROUP="plugdev"```. Then copy this file into /etc/udev/rules.d.

Then unplug/replug your Ubertooth and run ```ubertooth-util -v```.

Sorry it's so complicated. The next release addresses this.

erlichsefi, 2015-01-07:
output:
Firmware revision: 314

 this is the update firmware?

dominicgs, 2015-01-07:
No, that's very old.

The release package contains an up to date binary firmware file in ```ubertooth-one-firmware-bin/```.  Run ```ubertooth-dfu --write ubertooth-one-firmware-bin/bluetooth_rxtx.dfu``` to update the firmware.

erlichsefi, 2015-01-07:
[image]

so two days of work..and nothing? :(

dominicgs, 2015-01-07:
Just to double check the update - could you unplug/replug the Ubertooth and run ```ubertooth-util -V``` to verify the new firmware version?

erlichsefi, 2015-01-07:
ubertooth 2014-04-R1 (dominicgs@mercury) Mon Apr  7 20:14:49 BST 2014

dominicgs, 2015-01-07:
That's good, it looks like it updated correctly.  Wireshark still doesn't give you any packets?  You have a BLE device known to be transmitting nearby?

erlichsefi, 2015-01-07:
yes...i'm transmitting music to my speakers as we read :)

dominicgs, 2015-01-07:
Music is unlikely to be using BLE.  Please try ```ubertooth-rx```

dominicgs, 2015-01-24:
Closing this issue as I believe that we resolved the issues.

closed 2015-01-24 by dominicgs