greatscottgadgets/ubertooth #49: 'make plugins' fails to runs on OS 10.7.5 "-bundle not allowed with -dynamiclib"

nguillermin, 2015-01-13:
(On a second run of 'make plugins')

    PLUGIN: plugin-alertsyslog/
    g++ -bundle -flat_namespace -undefined suppress  -framework Foundation -framework CoreFoundation -F/System/Library/PrivateFrameworks -framework Apple80211 -framework IOKit -rdynamic -shared -rdynamic alertsyslog.o -o alertsyslog.so -ldl -lpcre -lstdc++ 
    i686-apple-darwin11-llvm-g++-4.2: -bundle not allowed with -dynamiclib
    make[1]: *** [alertsyslog.so] Error 1

    PLUGIN: plugin-btscan/
    g++ -bundle -flat_namespace -undefined suppress  -framework Foundation -framework CoreFoundation -F/System/Library/PrivateFrameworks -framework Apple80211 -framework IOKit -rdynamic -shared -rdynamic packetsource_linuxbt.o tracker_btscan.o dumpfile_btscantxt.o kismet_btscan.o -o btscan.so -ldl -lpcre -lstdc++ -lbluetooth -lpthread
    i686-apple-darwin11-llvm-g++-4.2: -bundle not allowed with -dynamiclib
    make[1]: *** [btscan.so] Error 1

    PLUGIN: plugin-spectools/
    g++ -bundle -flat_namespace -undefined suppress  -framework Foundation -framework CoreFoundation -F/System/Library/PrivateFrameworks -framework Apple80211 -framework IOKit -rdynamic -shared -rdynamic spectool_netclient.o spectool_kismet.o -o spectool_net.so -ldl -lpcre -lstdc++ 
    i686-apple-darwin11-llvm-g++-4.2: -bundle not allowed with -dynamiclib
    make[1]: *** [spectool_net.so] Error 1

    PLUGIN: plugin-syslog/
    g++ -bundle -flat_namespace -undefined suppress  -framework Foundation -framework CoreFoundation -F/System/Library/PrivateFrameworks -framework Apple80211 -framework IOKit -rdynamic -shared -rdynamic kismet-syslog.o -o kismet-syslog.so -ldl -lpcre -lstdc++ 
    i686-apple-darwin11-llvm-g++-4.2: -bundle not allowed with -dynamiclib
    make[1]: *** [kismet-syslog.so] Error 1
    
    make: *** [plugins] Error 2

The rest of kismet installs fine with 'sudo make suidinstall'.

dominicgs, 2015-01-13:
Hi,

None of these plugins are related to Ubertooth.

Dominic

nguillermin, 2015-01-13:
Hah! Sorry. You're right, this is actually kismet problem, cropped up while I was trying to get ubertooth working.

closed 2015-01-13 by nguillermin