greatscottgadgets/ubertooth #50: Error compiling the Kismet "packetsource_ubertooth.o" plugin on Raspbian

michaeljau, 2015-01-15:
Hi,
I have just tried to compile plugins for Kismet following the build instructions at:
https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

I got the following error when the "sudo make && make plugins" command was run. See screenshot.
[image]

I am doing the install on Raspbian (on a Raspberry Pi Model B+).

Any suggestions on how I can fix this or work around it? :)

Thanks

michaeljau, 2015-01-15:
Btw. I did a directory listing (with permissions) of the parent directory (/usr/local/src/ubertooth-2014-02-R2/host/kismet/) and see something odd:
[image]

Note. See "plugin-ubertooth -> ../ubertooth-2014-02-R2/host/kismet/plugin-ubertooth"

I'm not sure if this helps.

dominicgs, 2015-01-24:
Sorry for the delay in getting to this, I have been away from my desk for the past few weeks.

The permissions on the symbolic link are normal, the permissions on the directory are used, so the symlink allows all by default.

What are the permissions on the ubertooth-2014-02-R2/host/kismet/plugin-ubertooth directory?  That is where the permission error is coming from.

michaeljau, 2015-01-26:
Thanks Dominic - no worries I haven't had much chance to look at ubertooth lately anyway.

Your email prompted me to run "make" command again, however this time I gave write permission to everyone (chmod 777) and compilation ran fine. PS. I did set permissions back to normal after (chmod 755).

I have tried running the Ubertooth One on raspberry pi (Raspbian), Ubuntu 14.04 (on VM Player 7 & Virtual Box) it's not working. I get the a couple of entries where it reports the LAP before giving the "uh oh, full_buf not emptied".

From similar posts online it sounds like the suggestion is to use the Ubertooth on a "physical" machine with decent hardware specifications? Any version of Ubuntu you recommend? (it sounds like people have had more luck with Ubuntu 12.04)

dominicgs, 2015-02-14:
@michaeljau The "uh oh, full_buf not emptied" message implies that the Ubertooth is delivering data quicker than the host can process it.  One fix for this is to run ubertooth-rx with the ```-e 1``` option to reduce the level of error correction that it performs.

Using Ubertooth on modern x86 hardware with any Linux distribution from the past few years should not be a problem.  Raspbian on the raspberry pi will often work but reducing the error correction level is usually required; it also helps to reduce the memory usage, which is no bad thing on the pi.

I will close this issue as the permissions problem appears to be solved.  Please let me know if you have any further questions.

closed 2015-02-14 by dominicgs