greatscottgadgets/ubertooth #52: kismet does not output to btbb

joshjcomedy, 2015-02-05:
I have installed and compiled from this build guide
https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

running Ubuntu 14.04

my version is 

sudo ubertooth-util -v
Firmware revision: git-4412704

I can run kismet and use ubertooth. I see devices. Though the output does not save to btbb. I am getting pcap files though.

I have added pcapbtbb to the logtypes line in kismet.conf

any thoughts? Much obliged.

dominicgs, 2015-02-05:
@joshjcomedy Could you post your complete kismet config file?

Are you sure that your kismet.conf file is being picked up by kismet?  Have you tried changing other parts of the file to make sure that they take effect?

joshjcomedy, 2015-02-05:
Is there a recommended change I could make to validate changes are taking place?

config is located at 

/home/josh/kismet-2013-03-R1b/conf

Config is posted below

# Kismet config file
# Most of the "static" configs have been moved to here -- the command line
# config was getting way too crowded and cryptic.  We want functionality,
# not continually reading --help!

# Version of Kismet config
version=2009-newcore

# Name of server (Purely for organizational purposes)
# If commented out, defaults to host name of system
# servername=Kismet Server

# Prefix of where we log (as used in the logtemplate later)
# logprefix=/some/path/to/logs

# Do we process the contents of data frames?  If this is enabled, data
# frames will be truncated to the headers only immediately after frame type
# detection.  This will disable IP detection, etc, however it is likely
# safer (and definitely more polite) if monitoring networks you do not own.
# hidedata=true

# Do we allow plugins to be used?  This will load plugins from the system
# and user plugin directiories when set to true (See the README for the default
# plugin locations).
allowplugins=true

# See the README for full information on the new source format
# ncsource=interface:options
# for example:
# ncsource=wlan0
# ncsource=wifi0:type=madwifi
# ncsource=wlan0:name=intel,hop=false,channel=11

# Comma-separated list of sources to enable.  This is only needed if you defined
# multiple sources and only want to enable some of them.  By default, all defined
# sources are enabled.
# For example, if sources with name=prismsource and name=ciscosource are defined,
# and you only want to enable those two:
# enablesources=prismsource,ciscosource

# Control which channels we like to spend more time on.  By default, the list
# of channels is pulled from the driver automatically.  By setting preferred channels,
# if they are present in the channel list, they'll be set with a timing delay so that
# more time is spent on them.  Since 1, 6, 11 are the common default channels, it makes
# sense to spend more time monitoring them.
# For finer control, see further down in the config for the channellist= directives.
preferredchannels=1,6,11

# How many channels per second do we hop?  (1-10)
channelvelocity=3

# By setting the dwell time for channel hopping we override the channelvelocity
# setting above and dwell on each channel for the given number of seconds.
#channeldwell=10

# Channels are defined as:
# channellist=name:ch1,ch2,ch3
# or
# channellist=name:range-start-end-width-offset,ch,range,ch,...
#
# Channels may be a numeric channel or a frequency
#
[... 216 lines cut ...]
# Format of the pcap dump (PPI or 80211)
pcapdumpformat=ppi
# pcapdumpformat=80211

# Default log title
logdefault=Kismet

# logtemplate - Filename logging template.
# This is, at first glance, really nasty and ugly, but you'll hardly ever
# have to touch it so don't complain too much.
#
# %p is replaced by the logging prefix + '/'
# %n is replaced by the logging instance name
# %d is replaced by the starting date as Mon-DD-YYYY
# %D is replaced by the current date as YYYYMMDD
# %t is replaced by the starting time as HH-MM-SS
# %i is replaced by the increment log in the case of multiple logs
# %l is replaced by the log type (pcapdump, strings, etc)
# %h is replaced by the home directory

logtemplate=%p%n-%D-%t-%i.%l

# Where state info, etc, is stored.  You shouldnt ever need to change this.
# This is a directory.
configdir=%h/.kismet/

dominicgs, 2015-02-05:
Can you try running kismet server with the Ubertooth plugin directly: ```kismet_server -c ubertooth``` 

Or try pointing it at the config file explicitly ```kismet_server -f /home/josh/kismet-2013-03-R1b/conf/kismet.conf```

If neither of these work, could you also show us a copy of the output?

joshjcomedy, 2015-02-05:
output of kismet_server -c ubertooth

INFO: Reading from config file /usr/local/etc/kismet.conf
INFO: No 'dronelisten' config line and no command line drone-listen 
      argument given, Kismet drone server will not be enabled.
INFO: Created alert tracker...
INFO: Creating device tracker...
INFO: Registered 80211 PHY as id 0
ERROR: Failed to open user plugin directory (/root/.kismet//plugins/): No 
       such file or directory
INFO: Loaded info for plugin 'spectool_net.so': Plugin name: 'SPECTOOL' 
      Plugin version: '2013-03-R0' Plugin description: 'Spectool-Net'
INFO: Loaded info for plugin 'alertsyslog.so': Plugin name: 'ALERTSYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'ALERTSYSLOG Plugin'
INFO: Activated plugin '/usr/local/lib/kismet/alertsyslog.so': 
      'ALERTSYSLOG' '2013-03-R0' 
INFO: Loaded info for plugin 'btscan.so': Plugin name: 'BTSCAN' Plugin 
      version: '2013-03-R0' Plugin description: 'Active Bluetooth scanning 
      plugin'
INFO: Activated plugin '/usr/local/lib/kismet/btscan.so': 'BTSCAN' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'ubertooth.so': Plugin name: 'UBERTOOTH' 
      Plugin version: '2013-03-R0' Plugin description: 'Ubertooth plugin 
      for Bluetooth baseband protocol'
INFO: Pcap logging for type pcapbtbb
INFO: Activated plugin '/usr/local/lib/kismet/ubertooth.so': 'UBERTOOTH' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'kismet-syslog.so': Plugin name: 'SYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'SYSLOG Plugin'
ERROR: Kismet-Syslog not redirecting any messages to syslog because no 
       syslogtype= directive found in kismet.conf
INFO: Kismet-Syslog redirecting no messages to syslog
INFO: Activated plugin '/usr/local/lib/kismet/kismet-syslog.so': 'SYSLOG' 
      '2013-03-R0' 
INFO: Kismet will spend extra time on channels 1,6,11
INFO: Kismet will attempt to hop channels at 3 channels per second unless 
      overridden by source-specific options
INFO: Matched source type 'ubertooth' for auto-type source 'ubertooth'
INFO: Using default channel list 'UBERTOOTH' on source 'ubertooth'
INFO: Ubertooth using first USB device that looks like a Ubertooth
INFO: Created source ubertooth with UUID 
      35265874-ad85-11e4-842a-b8072e49dc03
INFO: Disabling channel hopping on source 'ubertooth' because it is not 
      capable of setting the channel.
INFO: Will attempt to reopen on source 'ubertooth' if there are errors
INFO: Created TCP listener on port 2501
INFO: Kismet drone framework disabled, drone will not be activated.
INFO: Inserting basic packet dissectors...
INFO: Allowing Kismet frontends to view WEP keys
INFO: Starting GPS components...
INFO: Enabling reconnection to the GPS device if the link is lost
INFO: Using GPSD server on localhost:2947
ERROR: Could not open OUI file '/etc/manuf': No such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/wireshark/manuf': No 
       such file or directory
INFO: Opened OUI file '/usr/share/wireshark/manuf
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 23950 lines 479 indexes
INFO: Creating network tracker...
INFO: Creating channel tracker...
INFO: Registering dumpfiles...
INFO: Pcap log in PPI format
INFO: Opened pcapdump log file 'Kismet-20150205-16-20-36-1.pcapdump'
INFO: Opened netxml log file 'Kismet-20150205-16-20-36-1.netxml'
INFO: Opened nettxt log file 'Kismet-20150205-16-20-36-1.nettxt'
INFO: Opened gpsxml log file 'Kismet-20150205-16-20-36-1.gpsxml'
INFO: Opened alert log file 'Kismet-20150205-16-20-36-1.alert'
INFO: No spectools= line in config file, will not try to use spectools for 
      spectrum data
INFO: Activated plugin '/usr/local/lib/kismet/spectool_net.so': 'SPECTOOL' 
[... 69 lines cut ...]
INFO: Enabling reconnection to the GPS device if the link is lost
INFO: Using GPSD server on localhost:2947
ERROR: Could not open OUI file '/etc/manuf': No such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/wireshark/manuf': No 
       such file or directory
INFO: Opened OUI file '/usr/share/wireshark/manuf
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 23950 lines 479 indexes
INFO: Creating network tracker...
INFO: Creating channel tracker...
INFO: Registering dumpfiles...
INFO: Pcap log in PPI format
INFO: Opened pcapdump log file 'Kismet-20150205-16-29-32-1.pcapdump'
INFO: Opened netxml log file 'Kismet-20150205-16-29-32-1.netxml'
INFO: Opened nettxt log file 'Kismet-20150205-16-29-32-1.nettxt'
INFO: Opened gpsxml log file 'Kismet-20150205-16-29-32-1.gpsxml'
INFO: Opened alert log file 'Kismet-20150205-16-29-32-1.alert'
INFO: No spectools= line in config file, will not try to use spectools for 
      spectrum data
INFO: Activated plugin '/usr/local/lib/kismet/spectool_net.so': 'SPECTOOL' 
      '2013-03-R0' 
INFO: Kismet starting to gather packets
INFO: No packet sources defined.  You MUST ADD SOME using the Kismet 
      client, or by placing them in the Kismet config file 
      (/usr/local/etc/kismet.conf)

dominicgs, 2015-02-05:
The second set of output shows kismet opening the correct output file. It looks like kismet hasn't been finding the configuration file.

Try copying it to /home/josh/.kismet/

joshjcomedy, 2015-02-05:
copied the .conf from /home/josh/kismet-2013-03-R1b/conf/kismet.conf
to /.kismet

is that correct? ran kismet with sudo kismet still no pcapbtbb output.

ran the second command again

kismet_server -c ubertooth

ERROR: Kismet was started as root, NOT launching external control binary.  
       This is NOT the preferred method of starting Kismet as Kismet will 
       continue to run as root the entire time.  Please read the README 
       file section about Installation & Security and be sure this is what 
       you want to do.
INFO: Reading from config file /usr/local/etc/kismet.conf
INFO: No 'dronelisten' config line and no command line drone-listen 
      argument given, Kismet drone server will not be enabled.
INFO: Created alert tracker...
INFO: Creating device tracker...
INFO: Registered 80211 PHY as id 0
ERROR: Failed to open user plugin directory (/root/.kismet//plugins/): No 
       such file or directory
INFO: Loaded info for plugin 'spectool_net.so': Plugin name: 'SPECTOOL' 
      Plugin version: '2013-03-R0' Plugin description: 'Spectool-Net'
INFO: Loaded info for plugin 'alertsyslog.so': Plugin name: 'ALERTSYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'ALERTSYSLOG Plugin'
INFO: Activated plugin '/usr/local/lib/kismet/alertsyslog.so': 
      'ALERTSYSLOG' '2013-03-R0' 
INFO: Loaded info for plugin 'btscan.so': Plugin name: 'BTSCAN' Plugin 
      version: '2013-03-R0' Plugin description: 'Active Bluetooth scanning 
      plugin'
INFO: Activated plugin '/usr/local/lib/kismet/btscan.so': 'BTSCAN' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'ubertooth.so': Plugin name: 'UBERTOOTH' 
      Plugin version: '2013-03-R0' Plugin description: 'Ubertooth plugin 
      for Bluetooth baseband protocol'
INFO: Pcap logging for type pcapbtbb
INFO: Activated plugin '/usr/local/lib/kismet/ubertooth.so': 'UBERTOOTH' 
      '2013-03-R0' 
INFO: Loaded info for plugin 'kismet-syslog.so': Plugin name: 'SYSLOG' 
      Plugin version: '2013-03-R0' Plugin description: 'SYSLOG Plugin'
ERROR: Kismet-Syslog not redirecting any messages to syslog because no 
       syslogtype= directive found in kismet.conf
INFO: Kismet-Syslog redirecting no messages to syslog
INFO: Activated plugin '/usr/local/lib/kismet/kismet-syslog.so': 'SYSLOG' 
      '2013-03-R0' 
INFO: Kismet will spend extra time on channels 1,6,11
INFO: Kismet will attempt to hop channels at 3 channels per second unless 
      overridden by source-specific options
INFO: Matched source type 'ubertooth' for auto-type source 'ubertooth'
INFO: Using default channel list 'UBERTOOTH' on source 'ubertooth'
INFO: Ubertooth using first USB device that looks like a Ubertooth
INFO: Created source ubertooth with UUID 
      b2e771da-ad8a-11e4-baac-b8072e49dc03
INFO: Disabling channel hopping on source 'ubertooth' because it is not 
      capable of setting the channel.
INFO: Will attempt to reopen on source 'ubertooth' if there are errors
INFO: Created TCP listener on port 2501
INFO: Kismet drone framework disabled, drone will not be activated.
INFO: Inserting basic packet dissectors...
INFO: Allowing Kismet frontends to view WEP keys
INFO: Starting GPS components...
INFO: Enabling reconnection to the GPS device if the link is lost
INFO: Using GPSD server on localhost:2947
ERROR: Could not open OUI file '/etc/manuf': No such file or directory
ERROR: Could not open OUI file '/usr/share/wireshark/wireshark/manuf': No 
       such file or directory
INFO: Opened OUI file '/usr/share/wireshark/manuf
INFO: Indexing manufacturer db
INFO: Completed indexing manufacturer db, 23950 lines 479 indexes
INFO: Creating network tracker...
INFO: Creating channel tracker...
INFO: Registering dumpfiles...
INFO: Pcap log in PPI format
INFO: Opened pcapdump log file 'Kismet-20150205-16-59-54-1.pcapdump'
INFO: Opened netxml log file 'Kismet-20150205-16-59-54-1.netxml'
INFO: Opened nettxt log file 'Kismet-20150205-16-59-54-1.nettxt'
INFO: Opened gpsxml log file 'Kismet-20150205-16-59-54-1.gpsxml'
INFO: Opened alert log file 'Kismet-20150205-16-59-54-1.alert'
INFO: No spectools= line in config file, will not try to use spectools for 
      spectrum data
INFO: Activated plugin '/usr/local/lib/kismet/spectool_net.so': 'SPECTOOL' 
      '2013-03-R0' 
INFO: Kismet starting to gather packets
INFO: Started source 'ubertooth'
rx blocks of 64 bytes in 512 byte transfers
GOT PACKET ch=69 LAP=9e8b33 err=0 clk100ns=3892239
first sighting 9e8b33
GOT PACKET ch=69 LAP=9e8b33 err=0 clk100ns=3892271
new network 9e8b33
GOT PACKET ch=69 LAP=9e8b33 err=0 clk100ns=3892287
GOT PACKET ch=71 LAP=9e8b33 err=0 clk100ns=3892564
GOT PACKET ch=71 LAP=9e8b33 err=0 clk100ns=3892580
GOT PACKET ch=71 LAP=9e8b33 err=0 clk100ns=3892596
GOT PACKET ch=24 LAP=9e8b33 err=0 clk100ns=3892612
GOT PACKET ch=59 LAP=9e8b33 err=0 clk100ns=3895724
GOT PACKET ch=59 LAP=9e8b33 err=0 clk100ns=3895740
GOT PACKET ch=61 LAP=9e8b33 err=0 clk100ns=3896048
GOT PACKET ch=65 LAP=9e8b33 err=0 clk100ns=3896647
GOT PACKET ch=65 LAP=9e8b33 err=0 clk100ns=3896663
GOT PACKET ch=65 LAP=9e8b33 err=0 clk100ns=3896679
GOT PACKET ch=73 LAP=9e8b33 err=0 clk100ns=3897976
^CINFO: Stopped source 'ubertooth'

*** KISMET IS SHUTTING DOWN ***
Shutting down log files...
INFO: Closed pcapdump log file 'Kismet-20150205-16-59-54-1.pcapdump', 0 
      logged.
INFO: Closed netxml log file 'Kismet-20150205-16-59-54-1.netxml', 0 logged.
INFO: Closed nettxt log file 'Kismet-20150205-16-59-54-1.nettxt', 0 logged.
INFO: Closed gpsxml log file 'Kismet-20150205-16-59-54-1.gpsxml', 0 logged.
INFO: Closed alert log file 'Kismet-20150205-16-59-54-1.alert', 0 logged.
INFO: Shutting down plugins...
WARNING: Kismet changes the configuration of network devices.
         In most cases you will need to restart networking for
         your interface (varies per distribution/OS, but 
         usually:  /etc/init.d/networking restart

Kismet exiting.

dominicgs, 2015-02-05:
If you run kismet using ```sudo kismet``` it will run as root and not find the config file (```/home/josh/.kismet/kismet.conf```)

joshjcomedy, 2015-02-05:
gotcha

kismet gave this error

FATAL: !!! CRITICAL ERROR !!! - Could not launch kismet_capture control binary, 
       due to permission errors.  To run Kismet suid-root your user MUST BE IN 
       THE 'kismet' GROUP.  Use the 'groups' command to show what groups your 
       user is in, and consult the Kismet README for more information. - Kismet 
       will not operate correctly.
Kismet client exiting.

I'm assuming I need to reinstall kismet following the https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide commands?

Is there anything I would have screwed up that wouldnt have created and added myself to a kismet group?

dominicgs, 2015-02-05:
You should be able to add yourself to the kismet group in ```/etc/group```

You probably didn't get added because the group wasn't created until you installed Kismet.  You shouldn't need to reinstall, although you may have needed to run ```make suidinstall``` from your kismet source directory.

joshjcomedy, 2015-02-05:
added my self to the group and still causing the same error

dominicgs, 2015-02-14:
@joshjcomedy Did you try to install Kismet using ```make suidinstall```?  This should have given you permissions to run it as a regular user who is part of the kismet group.

mikeryan, 2015-08-13:
@joshjcomedy are you still having this issue?

mikeryan, 2016-02-04:
Closing due to inactivity.

closed 2016-02-04 by mikeryan