greatscottgadgets/ubertooth #54: plugin-ubertooth compile fails

silverki, 2015-02-11:
It fails when using latest Kismet source (commit 1e637714) from https://www.kismetwireless.net/gitweb/?p=kismet.git;a=summary

make[1]: Entering directory `/xxx/ubertooth-2014-02-R2/host/kismet/plugin-ubertooth'
g++ -pthread -I/usr/include/ncurses   -Wall -g -O2 -I/home/silverk/util_src/kismet -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c ubertooth_ui.cc -o ubertooth_ui.o 
ubertooth_ui.cc: In function ‘int panel_plugin_init(GlobalRegistry*, KisPanelPluginData*)’:
ubertooth_ui.cc:208:40: error: ‘class Kis_Main_Panel’ has no member named ‘AddSortSeparator’
ubertooth_ui.cc: In function ‘void UbertoothProtoBTBBDEV(GlobalRegistry*, std::string, std::vector<smart_word_token>*, KisNetClient*, void*)’:
ubertooth_ui.cc:372:40: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
ubertooth_ui.cc: In function ‘int UbertoothDevlistCB(Kis_Panel_Component*, int, void*, GlobalRegistry*)’:
ubertooth_ui.cc:577:44: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
ubertooth_ui.cc: In function ‘int Ubertooth_Details_ButtonCB(Kis_Panel_Component*, int, void*, GlobalRegistry*)’:
ubertooth_ui.cc:595:1: warning: no return statement in function returning non-void [-Wreturn-type]
ubertooth_ui.cc: In member function ‘virtual void Ubertooth_Details_Panel::DrawPanel()’:
ubertooth_ui.cc:640:6: warning: unused variable ‘selected’ [-Wunused-variable]
make[1]: *** [ubertooth_ui.o] Error 1
make[1]: Leaving directory `/xxx/ubertooth-2014-02-R2/host/kismet/plugin-ubertooth'
make: *** [plugins] Error 2cd

dominicgs, 2015-02-11:
If you're using the latest Kismet code from git, then you need to use the phyneutral plugin.

The plugins are the same, but they target different versions of kismet.

dominicgs, 2015-02-13:
@silverki Did you manage to build the phy-neutral plugin for kismet?

I will assume that this was the problem and close this issue tomorrow if I don't hear back from you by then.

Thanks.

silverki, 2015-02-14:
@dominicgs . Thank you for rapid responses. I can compile newest version and plugins only on next Wednesday  or Fiday.
Was kismet client plugin UI changed in phyneutral?. Reason asking it is that there is no ubertooth_ui.cc in source any more.

dominicgs, 2015-02-17:
The kismet client changes, as part of the phyneutral update, remove some of the 802.11 specific UI elements to allow different data to be displayed.  For this reason the plugin no longer needs a separate UI, it can integrate with the main Kismet UI.

I will wait until the weekend before closing this issue.

dominicgs, 2015-06-03:
@silverki Having not heard anything else on this issue, and not seeing other reports of problems, I am going to close this.

closed 2015-06-03 by dominicgs