greatscottgadgets/ubertooth #55: Named pipe, kismet OSX and Ubuntu freeeze

1am, 2015-02-16:
Hi,

I've been trying to set up Ubertooth with Wireshark on both OSX 10.10 and Ubuntu 14.10 under a VM. I've got through the [tutorials](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) and actually got some data into wireshark but the scanning fails very quickly. 
I've got a few questions regarding the whole setup:

1) The whole scanning freezes after ~5 seconds on native OSX and around 60 sec in a VM. This is a bit awkward. How could i make the scanning process work for longer than that? 
2) Trying to run kismet under Ubuntu caused a "_uh oh, full_buf not emptied_" error quite fast.
3) I get a segfault sometimes with kismet server. ```dmesg``` output shows:
```
[  411.271988] kismet_server[2752]: segfault at 200 ip 00007fae5b4e0b79 sp 00007fae59fede00 error 4 in libusb-1.0.so.0.1.0[7fae5b4d8000+15000]
[  454.429413] atkbd serio0: Spurious NAK on isa0060/serio0. Some program might be trying to access hardware directly.
```
4) The ```ubertooth-specan-ui``` program seems to work fine without freezing for 2 minutes. What's the difference between it and the ```mkfifo tmp/pipe``` approach?
5) When Wireshark works it doesn't show "source" and "destination" addresses even though they are shown in the packet details view (in OSX). How do I get to display these values?
6) Under Ubuntu The protocol column shows "PPI". What is that and where it be changed?

dominicgs, 2015-02-16:
Hi,

Could you provide us with the versions of Ubertooth, libbtbb, Kismet and Wireshark that you are using here?

1) When you say that it freezes, does it lock up, exit, or just stop producing results?  Do you mean Kismet or one of the Ubertooth command line tools?

2) What is the spec of your Ubuntu VM?  This error is often a sign of the host not being able to process the data as quickly as the Ubertooth provides it.

3) Is this on the Ubuntu VM?

4) I'm not entirely sure what you mean by the ```mkfifo tmp/pipe``` approach, so correct me if I'm wrong, but is that the ```ubertooth-btle``` usage?  The two protocols are very different and the tools for each are also different.  They do not share a pcap format.

5) Source and destination in Wireshark are terms taken from other networking technologies.  Bluetooth doesn't have the same concept of source and destination, i.e. only the master device's MAC is used in the packets and even then, only 4 of the 6 bytes.  Are the fields blank?  Showing all 0s?

6) Which protocol are you sniffing here?  Basic Rate Bluetooth?  PPI means "Per Packet Information". It means that we have additional information for each packet that isn't part of the Bluetooth packet itself. i.e. a local clock, the address which we were sniffing for, etc.  This information is optional, so we put it in a separate header, thus the format is "PPI" rather than "Bluetooth".

1am, 2015-02-16:
Hi,

The versions are the following:
Ubertooth one with firmware: git-4412704 (downloaded today from [firmware update step](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware) )

1) By freezing i mean that Ubertooth does not show any data anymore. It just stops. I can CTRL+C kismet or ubertooth utilities. I need to disconnect the device and plug it back into USB.
2) It's a 64bit Ubuntu, 1024mb base memory, 1 CPU. After doubling that scanning worked a bit longer but still freezes... I've expected some issues when running in a VM anyways. I hoped for OSX to work rock-solid but similar things happen on both OS'es.
3) Yes, on Ubuntu. I didn't succeed with installing kismet on OSX and it's not the tool i wanted to use initially.
4) By ```mkfifo tmp/pipe``` I'm refering to [Capturing BLE in Wireshark](https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark) page of the wiki, so it's the same as ```ubertooth-btle```. What I notice is that ```ubertooth-btle```causes a freeze quite fast while the ``` ubertooth-specan-ui``` works without problems for at least few minutes.
5) All fields are shown blank but the terminal output shows some addresses. 
6) What I'm seing are the PPI packets and raw advertising data in packet details. I'm aiming more for something close to [this](https://lacklustre.net/bluetooth/btle_sensor_read_name_request.png) but with MAC addresses.

dominicgs, 2015-02-16:
Hi,

1) It may be that there is a problem with the USB interface being passed through to the VM.  This is not uncommon, and it's likely that it will be stopping the data from getting through. If this is the case, I don't think that there is anything we can do to fix it.

2) I'm also surprised about the tools not working on OSX.  Unfortunately I don't have a system to test them with.  Is there any output that could give you a clue about what is happening on OSX? Any errors or exit codes?

3) Of course, I should have known that as it was dmesg output.  The segfault could be issue #18, although I've been assuming that was an problem with the amount of RAM available, which may not apply here.

4) ```ubertooth-specan-ui``` is a much lighter weight protocol, it is returning RSSI values from the Ubertooth, so not as much data. ```ubertooth-btle``` uses the Ubertooth micro-controller to detect packets and then passes them to the host.  Are you seeing this freeze on Ubuntu and OSX?  Does it freeze if you don't use the pipe, i.e. just running ```ubertooth-btle -f```?

I'll leave questions 5 and 6 for @mikeryan, he knows the BLE Wireshark plugin much better than I do (because he wrote it).

mikeryan, 2015-02-16:
What version of Wireshark are you using? Is it dissecting the BLE advertising data?

To get data similar to the screenshot from my site, you'll need to capture a connection. A single Ubertooth only has a 1/3 chance of capturing a given connection, so simply disconnect and reconnect until you see BLE data packets in Wireshark.

The following filter is useful for Wireshark 1.12: ```btle.advertising_header || btle.data_header.length > 0```

1am, 2015-02-17:
@dominicgs:

1) I'd expect that the VM could be the problem but it's the same on OSX which is my native OS (2012 mac Mini, 2.5Ghz i5, 16gb ram, OSX 10.10, no SSD at the moment). Im plugging Ubertooth into the mini's USB port and it's the only device under "USB 3.0 Hi-Speed Bus" listed in system information. 

2) There's nothing in the system console as well as in the terminal. I'll check on a 2011 macbook pro and an old IBM T43 when i set Ubuntu on it.

4) Yes, ```ubertooth-btle -f``` also freezes after a minute or so. No output whatsoever. It doesn't even exit.

Maybe it's caused by the amount of BTLE devices i have around? I've got min 5 devices advertising at around 100ms interval + some Apple TV device owned by someone next door.

@mikeryan: 

I've tried Wireshark 1.12.3 + 1.99.2 on OSX, and 1.12.1 on Ubuntu.
The data is not actually parsed. All of the raw bytes are shown in the details of a packet. What is also shown in packet details is the message: ```User encapsulation not handled: DLT=147, check your Preferences->Protocols->DLT_USER```

I used the image as a reference - it's got all packet data parsed and a lot of meaningful info shown. What I'd love to get to is having the source mac address for advertisement packets. 
This should be possible since the mac address is received by other sniffers I've used (SmartRF Protocol Packet Sniffer from TI, the nRF51822 based from Adafruit and simple monitor I've wrote using the BLED112 dongle). All I need is is just connect it to wireshark somehow to get proper filtering and visibility.

1am, 2015-02-17:
I just took out a second Ubertooth board, plugged it in and just started ```ubertooth-btle -f -c /tmp/pipe``` and it's working for around 5 minutes now on OSX (got 10000 packets into wireshark).
So the one which is working is on firmware **2012-10-R1** and the one which has problems is on **git-4412704**.
I took the device which wasn't working and rolled back to **2014-04-R1** and it got stuck after ~300packets. I'm going to roll back through the [releases](https://github.com/greatscottgadgets/ubertooth/releases).

Maybe I'm writing the wrong firmware? The [instructions](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware) state that I should write **ubertooth-one-bin-firmware.dfu** I don't see anything named like this and I'm using **bluetooth_rxtx.dfu**

1am, 2015-02-17:
Ok, I've downloaded the binary firmware from the working device, signed it with ```ubertooth-dfu --sign``` and written to the non-working device. Seems that it helped since I've already got to >4000packets being displayed in Wireshark.
I then re-downloaded firmware linked on [How To Update Firmware](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware) (named *ubertooth-one-bin-firmware.dfu*) wiki page and the sniffing stopped at 292 packets.

dominicgs, 2015-02-17:
```ubertooth-one-bin-firmware.dfu``` is the name of the binary firmware that we release.
```bluetooth_rxtx.dfu``` is what we call the firmware in git when developing it. They should be the same firmware, although the two files that you have could be different versions.

It looks like a firmware issue, perhaps involving compatibility between host code and the firmware.

Could you confirm the firmware version that is working and the one that stops after ~300 packets. (Using ```ubertooth-util -v``` or ```ubertooth-util -V```)

Please could you also confirm which version of the Ubertooth host code you are using?  Did you follow the build guide and install 2014-02-R2?  Or build directly from git?

1am, 2015-02-17:
Re: ```ubertooth-util -v``` I have tried: *2014-04-R1* and *2014-04-R2* with freezing. The one that works fine is *2012-10-R1* which Ubertooth originally shipped with.

Regarding ```ubertooth-util -V``` I'm getting the same output for both OSes.
```
control message unsupported
error: -9
```

dominicgs, 2015-02-17:
Getting the control message error for -V is expected with those firmware versions, we added it later than the -v option.

Which version of the host code are you using?  The host code version needs to match the firmware version to ensure the best possible outcome.

1am, 2015-02-17:
One more thing I've noticed... There's a significant size difference between the firmware versions which work and not. 

```
-rw-r-----@ 1 pb  staff    35K Feb 17 10:41 ubertooth-one-bin-firmware.dfu
-rw-r--r--  1 pb  staff   240K Feb 17 10:36 ubertooth_2012-10-R2.dfu
```

```
$ ubertooth-dfu --write ubertooth-one-bin-firmware.dfu 
Checking firmware signature
No DFU devices found - attempting to find Ubertooth devices

1) Found 'Ubertooth One' with USB ID: 1d50:6002

Select a device to flash (default:1, exit:0):1
.............................................................................................................................................
Write complete

$ ubertooth-dfu --write ubertooth_2012-10-R2.dfu 
Checking firmware signature
................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
Write complete
```

1am, 2015-02-17:
@dominicgs I've followed the [Ubertooth tools build guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#ubertooth-tools). The ```-V ``` option seems to be available in my host code version but it fails when I run it.
```
$ ubertooth-util -h
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One
Usage:
	-a[0-7] get/set power amplifier level
	-b get hardware board id number
	-c[2400-2483] get/set channel in MHz
	-C[0-78] get/set channel
	-d[0-1] get/set all LEDs
	-e start repeater mode
	-f activate flash programming (DFU) mode
	-h display this message
	-i activate In-System Programming (ISP) mode
	-I identify ubertooth device by flashing all LEDs
	-l[0-1] get/set USR LED
	-m display range test result
	-n initiate range test
	-p get microcontroller Part ID
	-q[1-225 (RSSI threshold)] start LED spectrum analyzer
	-r full reset
	-s get microcontroller serial number
	-S stop current operation
	-t intitiate continuous transmit test
	-U<0-7> set ubertooth device to use
	-v get firmware revision number
	-V get compile info
	-z set squelch level
```

dominicgs, 2015-02-17:
Thanks, the -V option is part of the 2014-02-R2 release, but it's unsupported by the Ubertooth because you are using the 2012-10-R2 firmware.  This is expected.

The 2014-02-R2 archive that you downloaded as part of the build guide should have a directory in it called ```ubertooth-one-firmware-bin```, which will contain a file called ```bluetooth_rxtx.dfu```.

Could you try to use that firmware image and let me know the results?  It must specifically be the one from the release archive.

1am, 2015-02-17:
I just spotted that the release linked in [Ubertooth tools build guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#ubertooth-tools) and in [releases page](https://github.com/greatscottgadgets/ubertooth/releases) is different - theres a .xz and .gz difference. Actually the packages are different and the .gz doesn't have the *ubertooth-one-firmware-bin* directory.

.gz listing:
```
$ ls -la
total 64
drwxr-xr-x  11 pb  staff    374 Feb 20  2014 .
drwxr-xr-x  14 pb  staff    476 Feb 17 11:35 ..
-rw-r--r--   1 pb  staff    390 Feb 20  2014 .gitignore
-rw-r--r--   1 pb  staff  17967 Feb 20  2014 COPYING
-rw-r--r--   1 pb  staff   1410 Feb 20  2014 README.md
-rw-r--r--   1 pb  staff    245 Feb 20  2014 TRADEMARK
drwxr-xr-x  14 pb  staff    476 Feb 20  2014 firmware
drwxr-xr-x  11 pb  staff    374 Feb 20  2014 hardware
drwxr-xr-x  11 pb  staff    374 Feb 20  2014 host
drwxr-xr-x   5 pb  staff    170 Feb 20  2014 tools
drwxr-xr-x  10 pb  staff    340 Feb 20  2014 web
```

.xz listing:
```
$ ls -la
total 88
drwxr-xr-x@  14 pb  staff    476 Feb 17 11:36 .
drwx------+ 174 pb  staff   5916 Feb 17 11:36 ..
-rw-r--r--@   1 pb  staff   6148 Feb 17 11:36 .DS_Store
-rw-r--r--@   1 pb  staff    390 Feb 19  2014 .gitignore
-rw-r--r--@   1 pb  staff  17967 Feb 19  2014 COPYING
-rw-r--r--@   1 pb  staff   1410 Feb 19  2014 README.md
-rw-r--r--@   1 pb  staff   1615 Feb 20  2014 RELEASENOTES
-rw-r--r--@   1 pb  staff    245 Feb 19  2014 TRADEMARK
drwxr-xr-x@  14 pb  staff    476 Feb 20  2014 firmware
drwxr-xr-x@  11 pb  staff    374 Feb 20  2014 hardware
drwxr-xr-x@  11 pb  staff    374 Feb 20  2014 host
drwxr-xr-x@   4 pb  staff    136 Feb 20  2014 tools
drwxr-xr-x@  11 pb  staff    374 Feb 20  2014 ubertooth-one-firmware-bin
drwxr-xr-x@  12 pb  staff    408 Feb 20  2014 web
```

After writing the *ubertooth-2014-02-R2.tar.xz/ubertooth-one-firmware-bin/bluetooth_rxtx.dfu* I get the following
```
$ ubertooth-util -v
Firmware revision: 2014-02-R2

$ ubertooth-util -V
ubertooth 2014-02-R2 (dominicgs@mercury) Thu Feb 20 13:28:01 GMT 2014
```

Ran this version on OSX with ```ubertooth-btle -f -c /tmp/pipe``` and it froze after 140 packets.

1am, 2015-02-17:
Maybe this could have something to do with libbtbb?
The wiki page links to two different versions of it. 
In the [Prerequisites](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#prerequisites) section it points to *2014-02-R4* and later on in [Wireshark](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#wireshark) section it points to 
```
cd libbtbb-2014-02-R2/wireshark/plugins/btbb
```

I have 2014-02-R4

dominicgs, 2015-02-17:
Yes, the .zip and .tar.gz are automatically created from git.  The .xz is something I create to have the binary firmware files included.  I will update the build guide to use the .xz file.

The libbtbb change was for distribution package maintainers only, it doesn't affect functionality.

This is starting to look like a bug that we've introduced in 2014-02-R2, but I'm surprised nobody else is seeing it.  @mikeryan Any ideas?

1am, 2015-02-17:
Since I could progress with Wireshark setup I tried ti set it up again on Ubuntu 14.10 VM (I don't yet know where to get the *wireshark-dev*, *libwireshark-dev* package equivalents for OSX).  
I followed the tutorial and the btbb.so seems to be installed correctly in Wireshark's plugin directory (in my case it's ```/usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1```). 
After running wireshark all of the packages are not parsed in any way. Is there some setting I need to enable in Wireshark preferences to enable the plugin? I can't really find the right spot.

I'm pasting the output for each step of building process:
```
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1 ..
-- The C compiler identification is GNU 4.9.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for one of the modules 'glib-2.0>=2.4.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pb/Desktop/ubertooth/libbtbb-2014-02-R4/wireshark/plugins/btbb/build
```

```
$ make
[ 25%] Generating plugin.c
Scanning dependencies of target btbb
[ 50%] Building C object CMakeFiles/btbb.dir/plugin.c.o
[ 75%] Building C object CMakeFiles/btbb.dir/packet-btbb.c.o
[100%] Building C object CMakeFiles/btbb.dir/packet-btlmp.c.o
/home/pb/Desktop/ubertooth/libbtbb-2014-02-R4/wireshark/plugins/btbb/packet-btlmp.c: In function ‘proto_register_btlmp’:
/home/pb/Desktop/ubertooth/libbtbb-2014-02-R4/wireshark/plugins/btbb/packet-btlmp.c:2262:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/pb/Desktop/ubertooth/libbtbb-2014-02-R4/wireshark/plugins/btbb/packet-btlmp.c:2267:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/pb/Desktop/ubertooth/libbtbb-2014-02-R4/wireshark/plugins/btbb/packet-btlmp.c:2272:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
Linking C shared module btbb.so
[100%] Built target btbb
```

```
$ sudo make install
[100%] Built target btbb
Install the project...
-- Install configuration: ""
-- Installing: /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/btbb.so
```

```
$ ls -la /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/
total 2468
drwxr-xr-x 2 root root   4096 lut 17 12:48 .
drwxr-xr-x 3 root root   4096 lut 17 12:41 ..
-rw-r--r-- 1 root root 123024 lut 17 12:48 btbb.so
-rw-r--r-- 1 root root 259472 lut  5 17:43 docsis.so
-rw-r--r-- 1 root root 143712 lut  5 17:43 ethercat.so
-rw-r--r-- 1 root root  91664 lut  5 17:43 gryphon.so
-rw-r--r-- 1 root root  54744 lut  5 17:43 irda.so
-rw-r--r-- 1 root root  20768 lut  5 17:43 m2m.so
-rw-r--r-- 1 root root  97128 lut  5 17:43 mate.so
-rw-r--r-- 1 root root 234576 lut  5 17:43 opcua.so
-rw-r--r-- 1 root root 472528 lut  5 17:43 profinet.so
-rw-r--r-- 1 root root  10848 lut  5 17:43 stats_tree.so
-rw-r--r-- 1 root root 152696 lut  5 17:43 unistim.so
-rw-r--r-- 1 root root  74720 lut  5 17:43 wimaxasncp.so
-rw-r--r-- 1 root root  91296 lut  5 17:43 wimaxmacphy.so
-rw-r--r-- 1 root root 651568 lut  5 17:43 wimax.so
```

I'm attaching a screenshot of what I get in Wireshark. 

[image]

1am, 2015-02-17:
@dominicgs I just went through the same procedure on a fresh installed native 32 bit Ubuntu 14.10 on an old IBM T43 and I've got the same results. The capture froze after 100 packets on the latest firmware and worked on the older one.

1am, 2015-02-18:
@dominicgs I tried building the bluetooth_rxtx firmware from source and it seems to be working ok now (got to ~10k packets received).

```
$ sudo ubertooth-util -v
Firmware revision: git-c9d3ed7
```
```
$ sudo ubertooth-util -V
ubertooth git-c9d3ed7 (pb@pb-vb) śro, 18 lut 2015, 10:16:11 CET
```

So it seems that the current codebase might be fine.

dominicgs, 2015-02-18:
@1am Ok, this seems to be something that was broken in 2014-02-R2, but has since been fixed.  I'm surprised that no one else has seen this issue, but I'm going to close it for now.

Feel free to re-open it if you see the issue again.

closed 2015-02-18 by dominicgs

1am, 2015-02-18:
@dominicgs Thank you for your support.

jokumusic, 2015-03-13:
Had this same problem with 2014-04-R1. Git cloned the latest version, compiled and flashed firmware. All good now... :)