greatscottgadgets/ubertooth #56: stack smashing detected

Artson, 2015-02-17:
There is a <code>*** stack smashing detected ***</code> error when sniffing with the -r and -f options.
This happens when the device I am sniffing initiates a connection.
Here is the full stack 
<code>*** stack smashing detected ***: /usr/local/bin/ubertooth-btle terminated Program received signal SIGABRT, Aborted. 0x00007ffff75a9cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  0x00007ffff75a9cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff75ad0d8 in __GI_abort () at abort.c:89
#2  0x00007ffff75e6f24 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff76f30cb "*** %s ***: %s terminated\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff767e82c in __GI___fortify_fail (msg=<optimized out>, 
    msg@entry=0x7ffff76f30b3 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff767e7d0 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x00007ffff7947f8a in lell_pcapng_append_packet (h=0x604010, 
    ns=1424141496096431138, sigdbm=-120 '\210', noisedbm=-55 '\311', 
    refAA=2391391958, pkt=0x6046c0)
    at /home/test/libbtbb/lib/src/pcapng-bt.c:474
#6  0x00007ffff7bd398f in cb_btle (args=0x7fffffffe510, rx=0x7fffffffe560, 
    bank=0) at /home/test/ubertooth/host/libubertooth/src/ubertooth.c:694
#7  0x0000000000401ceb in main (argc=4, argv=0x7fffffffe6a8)
    at /home/test/ubertooth/host/ubertooth-tools/src/ubertooth-btle.c:293
(gdb) backtrace full
#0  0x00007ffff75a9cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
        resultvar = 0
        pid = 28422
        selftid = 28422
#1  0x00007ffff75ad0d8 in __GI_abort () at abort.c:89
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x7fffffffe200, 
            sa_sigaction = 0x7fffffffe200}, sa_mask = {__val = {555573836, 2, 
              140737351925404, 140737353983848, 524302793, 3, 14, 
              140737353983848, 140737351927830, 140737347034312, 
              140737488347456, 140737343089960, 140737488347728, 
              140737343100784, 8192231, 140737488347712}}, sa_flags = 6309680, 
          sa_restorer = 0x0}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007ffff75e6f24 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff76f30cb "*** %s ***: %s terminated\n")
    at ../sysdeps/posix/libc_fatal.c:175
        ap = {{gp_offset = 32, fp_offset = 0, 
            overflow_arg_area = 0x7fffffffe350, 
            reg_save_area = 0x7fffffffe2e0}}
        fd = 13
        on_2 = <optimized out>
---Type <return> to continue, or q <return> to quit---
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
        written = <optimized out>
#3  0x00007ffff767e82c in __GI___fortify_fail (msg=<optimized out>, 
    msg@entry=0x7ffff76f30b3 "stack smashing detected") at fortify_fail.c:37
        do_abort = 1
#4  0x00007ffff767e7d0 in __stack_chk_fail () at stack_chk_fail.c:28
No locals.
#5  0x00007ffff7947f8a in lell_pcapng_append_packet (h=0x604010, 
    ns=1424141496096431138, sigdbm=-120 '\210', noisedbm=-55 '\311', 
    refAA=2391391958, pkt=0x6046c0)
    at /home/test/libbtbb/lib/src/pcapng-bt.c:474
        flags = 51
        pcapng_pkt = {blk_header = {block_type = 6, block_total_length = 92, 
            interface_id = 0, timestamp_high = 331583781, 
            timestamp_low = 817404962, captured_len = 53, packet_len = 53, 
            packet_data = 0x7fffffffe3fc}, le_ll_header = {
            rf_channel = 0 '\000', signal_power = -120 '\210', 
            noise_power = -55 '\311', access_address_offenses = 0 '\000', 
            ref_access_address = 2391391958, flags = 51, 
            le_packet = 0x7fffffffe406 "־\211\216\005\"\277\371\310\356\313", <incomplete sequence \340>}, 
---Type <return> to continue, or q <return> to quit---
          le_packet = "־\211\216\005\"\277\371\310\356\313\340\000v\b>\004\000TSeP\322V\243\003\n\000'\000\000\000\320\a\377\377\377\377\037\255\345\032\315\344\377\377\000"}
        retval = 0
#6  0x00007ffff7bd398f in cb_btle (args=0x7fffffffe510, rx=0x7fffffffe560, 
    bank=0) at /home/test/ubertooth/host/libubertooth/src/ubertooth.c:694
        pkt = 0x6046c0
        opts = 0x7fffffffe510
        i = 0
        access_address = 0
        prev_ts = 559938203
        refAA = 2391391958
        sig = -120 '\210'
        noise = -55 '\311'
        nowns = 1424141496096431138
        ts_diff = 32767
        len = 0
#7  0x0000000000401ceb in main (argc=4, argv=0x7fffffffe6a8)
    at /home/test/ubertooth/host/ubertooth-tools/src/ubertooth-btle.c:293
        r = 64
        pkt = {pkt_type = 0 '\000', status = 0 '\000', channel = 0 '\000', 
          clkn_high = 0 '\000', clk100ns = 559943243, rssi_max = -128 '\200', 
          rssi_min = 127 '\177', rssi_avg = 0 '\000', rssi_count = 0 '\000', 
---Type <return> to continue, or q <return> to quit---
          reserved = "\000", 
          data = "־\211\216\005\"\277\371\310\356\313\340\000v\b>\004\000TSeP\322V\243\003\n\000'\000\000\000\320\a\377\377\377\377\037\255\345\032\315l]L\004\222\000"}
        opt = -1
        do_follow = 1
        do_file = 0
        do_promisc = 0
        do_get_aa = 0
        do_set_aa = 0
        do_crc = -1
        do_adv_index = 37
        do_slave_mode = 0
        do_target = 0
        ubertooth_device = -1 '\377'
        cb_opts = {allowed_access_address_errors = 32}
        r = 4197219
        access_address = 32767
        mac_address = "\000\000\000\000\000"

</code>

dominicgs, 2015-02-17:
Thanks for reporting this, I seem to remember seeing a similar problem with the basic rate code a while ago. It came down to a packet being longer than the maximum packet length we'd allowed for.

I didn't think that out BLE implementation could suffer from the same bug, but I'll take a look to see if it shares the same root cause.

troth, 2015-05-26:
This appears to be a buffer overflow bug in libbtbb. I have submitted a fix:

  https://github.com/greatscottgadgets/libbtbb/pull/18

mikeryan, 2015-08-13:
Looks to be fixed. Tentatively closing.

closed 2015-08-13 by mikeryan