greatscottgadgets/ubertooth #57: Building host utils from source

1am, 2015-02-18:
Hi, 

While trying to figure out what's wrong with my setup in [#55](https://github.com/greatscottgadgets/ubertooth/issues/55) I've built the firmware from latest checkout from master branch. The firmware built without any problems but I've also attempted to build latest host utils and it seems that the dependancies are different than in build 2014-02-R2. 

Whan I build host utils form 2014-02-R2 i get the following output:
```
$ make
Scanning dependencies of target ubertooth
[  7%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
[ 15%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_control.c.o
Linking C shared library libubertooth.so
[ 15%] Built target ubertooth
Scanning dependencies of target ubertooth-btle
[ 23%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-btle.dir/ubertooth-btle.c.o
Linking C executable ubertooth-btle
[ 23%] Built target ubertooth-btle
Scanning dependencies of target ubertooth-debug
[ 30%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/ubertooth-debug.c.o
[ 38%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/cc2400.c.o
[ 46%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-debug.dir/arglist.c.o
Linking C executable ubertooth-debug
[ 46%] Built target ubertooth-debug
Scanning dependencies of target ubertooth-dump
[ 53%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-dump.dir/ubertooth-dump.c.o
Linking C executable ubertooth-dump
[ 53%] Built target ubertooth-dump
Scanning dependencies of target ubertooth-follow
[ 61%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-follow.dir/ubertooth-follow.c.o
Linking C executable ubertooth-follow
[ 61%] Built target ubertooth-follow
Scanning dependencies of target ubertooth-rx
[ 69%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-rx.dir/ubertooth-rx.c.o
Linking C executable ubertooth-rx
[ 69%] Built target ubertooth-rx
Scanning dependencies of target ubertooth-scan
[ 76%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-scan.dir/ubertooth-scan.c.o
Linking C executable ubertooth-scan
[ 76%] Built target ubertooth-scan
Scanning dependencies of target ubertooth-util
[ 84%] Building C object ubertooth-tools/src/CMakeFiles/ubertooth-util.dir/ubertooth-util.c.o
Linking C executable ubertooth-util
[ 84%] Built target ubertooth-util
Scanning dependencies of target specan
[ 92%] Generating build/timestamp
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/specan
copying /home/pb/Desktop/ubertooth/ubertooth-2014-02-R2/host/python/specan_ui/specan/Ubertooth.py -> build/lib.linux-x86_64-2.7/specan
copying /home/pb/Desktop/ubertooth/ubertooth-2014-02-R2/host/python/specan_ui/specan/__init__.py -> build/lib.linux-x86_64-2.7/specan
[ 92%] Built target specan
Scanning dependencies of target ubertooth-specan-ui
[ 92%] Built target ubertooth-specan-ui
Scanning dependencies of target dfu
[100%] Generating build/timestamp
running build
running build_py
creating build
creating build/lib.linux-x86_64-2.7
creating build/lib.linux-x86_64-2.7/dfu
copying /home/pb/Desktop/ubertooth/ubertooth-2014-02-R2/host/python/usb_dfu/dfu/__init__.py -> build/lib.linux-x86_64-2.7/dfu
copying /home/pb/Desktop/ubertooth/ubertooth-2014-02-R2/host/python/usb_dfu/dfu/dfu.py -> build/lib.linux-x86_64-2.7/dfu
[100%] Built target dfu
Scanning dependencies of target ubertooth-dfu
[100%] Built target ubertooth-dfu
```

When I try from git sources I get:
```
$ make
Scanning dependencies of target ubertooth
[  7%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
In file included from /home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:30:0:
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:67:8: error: unknown type name ‘btbb_pcap_handle’
 extern btbb_pcap_handle * h_pcap_bredr;
        ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:68:8: error: unknown type name ‘lell_pcap_handle’
 extern lell_pcap_handle * h_pcap_le;
        ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:70:8: error: unknown type name ‘btbb_pcapng_handle’
 extern btbb_pcapng_handle * h_pcapng_bredr;
        ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:71:8: error: unknown type name ‘lell_pcapng_handle’
 extern lell_pcapng_handle * h_pcapng_le;
        ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:62:1: error: unknown type name ‘btbb_pcap_handle’
 btbb_pcap_handle * h_pcap_bredr = NULL;
 ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:63:1: error: unknown type name ‘lell_pcap_handle’
 lell_pcap_handle * h_pcap_le = NULL;
 ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:65:1: error: unknown type name ‘btbb_pcapng_handle’
 btbb_pcapng_handle * h_pcapng_bredr = NULL;
 ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:66:1: error: unknown type name ‘lell_pcapng_handle’
 lell_pcapng_handle * h_pcapng_le = NULL;
 ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c: In function ‘print_version’:
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:70:6: warning: format ‘%s’ expects argument of type ‘char *’, but argument 4 has type ‘int’ [-Wformat=]
      btbb_get_version(), btbb_get_release());
      ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:70:6: warning: format ‘%s’ expects argument of type ‘char *’, but argument 5 has type ‘int’ [-Wformat=]
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c: In function ‘cb_br_rx’:
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:447:16: error: ‘UAP_ANY’ undeclared (first use in this function)
  uint8_t uap = UAP_ANY;
                ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:447:16: note: each undeclared identifier is reported only once for each function it appears in
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:488:34: error: ‘BTBB_MOD_GFSK’ undeclared (first use in this function)
  btbb_packet_set_modulation(pkt, BTBB_MOD_GFSK);
                                  ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:489:33: error: ‘BTBB_TRANSPORT_ANY’ undeclared (first use in this function)
  btbb_packet_set_transport(pkt, BTBB_TRANSPORT_ANY);
                                 ^
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c: In function ‘cb_btle’:
/home/pb/Desktop/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:609:2: error: unknown type name ‘lell_packet’
  lell_packet * pkt;
  ^
libubertooth/src/CMakeFiles/ubertooth.dir/build.make:54: recipe for target 'libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o' failed
make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o] Error 1
CMakeFiles/Makefile2:121: recipe for target 'libubertooth/src/CMakeFiles/ubertooth.dir/all' failed
make[1]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2
```

Should I get some other version of libbtbb?

1am, 2015-02-18:
Ok. Seems that building after installing latest [libbtbb](https://github.com/greatscottgadgets/libbtbb) fixes the problem.

closed 2015-02-18 by 1am

dominicgs, 2015-02-18:
Yes, the latest git version of Ubertooth tracks the latest libbtbb.

(I know you've discovered this and closed the issue, but I'm confirming for future users)