greatscottgadgets/ubertooth #58: Seetting up btbb with Wireshark

1am, 2015-02-18:
@mikeryan I've decided to make a separate ticket from [#55](https://github.com/greatscottgadgets/ubertooth/issues/55) for the Wireshark configuration since I can't really get how it's supposed to work. It's probably one small detail I'm missing but the packet parsing seems not to be working on my side. I've just rebuilt the btbb.so from master branch ob [libbtbb](https://github.com/greatscottgadgets/libbtbb) repository and followed the instructions from [ubertooth wiki Wireshark setup section](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide#wireshark).

What I've got after building btbb.so is a file exactly the same size as one built from libbtbb-2014-02-R4 release:

```
$ ls -la /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1
...
-rw-r--r-- 1 root root 123024 lut 18 10:38 btbb.so
-rw-r--r-- 1 root root 123024 lut 17 12:48 btbb.so-old
...
```

I first thought that it might actually be the same file but checksums seem to differ:
```
$ sudo md5sum /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/btbb.so
e42bc1c54020ca56fa918778014b2ccd  /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/btbb.so
```

```
$ sudo md5sum /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/btbb.so-old 
c79c7f8c9324218d3f7ca58391a55ee1  /usr/lib/x86_64-linux-gnu/wireshark/plugins/1.12.1/btbb.so-old
```

In both cases of btbb I have only unparsed PPI packets shown in Wireshark:
[image]

I even tried running Wireshark as root but it's always PPI.

ikelos, 2015-02-18:
The wireshark dissectors are part of libbtbb, I've filed an issue (#13) in that repo.
https://github.com/greatscottgadgets/libbtbb/issues/13

I believe the dissector works fine, but only for old file formats.  The newer libbtbb writes them to pcap in a different format, which the dissector does not appear to have been updated to handle.

1am, 2015-02-18:
Thank you @ikelos. I'll close this ticket and respond in the libbtbb one.

closed 2015-02-18 by 1am

ikelos, 2015-02-18:
Ah, yep, so that big yellow bar in the data decoding section is pointing out that you need to check Edit -> Preferences -> Protocols -> DLT_USER.

Under there, click Edit under Encapsulation tables, and then click New.  Make sure User0 (DLT=147) is selected, and then enter btle under Payload Protocol.  I believe you can leave the others blank.  Then ok all the dialog boxes, and hopefully you'll no longer see that issue...

1am, 2015-02-18:
Finally! That was what I've been looking for. Thank you @ikelos !

Do you maybe know a way to show Source address in packet list?

12424642, 2016-08-18:
This is exactly what I was looking for too, thanks @ikelos!
@1am Did you manage to find out how to display the source address?