greatscottgadgets/ubertooth #59: kismet-ubertooth double frees in ubertooth::poll

ikelos, 2015-02-18:
Hi there,

Everytime I run kismet_server -c ubertooth, the ubertooth::poll method seems to cause a double free at line 508 in packetsource_ubertooth.cc after it receives its second packet (this happens even if the two LAPs are not the same).  I'm using libbtbb, ubertooth, kismet and kismet-ubertooth all from git, this also happens with non-git versions of the software as well (ubertooth 2014.04.1 and kismet 2013.03.1).  Tested with original/purchase firmware and 2014-04-R1 firmware.

I'm happy to provide any further diagnostics that might help narrow down the issue.  Just let me know what you need.  5:)

INFO: Started source 'ubertooth'
rx blocks of 64 bytes in 512 byte transfers
INFO: kismet_capture pid 21183 synced with Kismet server, starting service 
      loop
GOT PACKET ch=49 LAP=1928c7 err=0 clk100ns=235484
GOT PACKET ch=19 LAP=1928c7 err=0 clk100ns=235757
INFO: Detected new Bluetooth baseband device 00:00:00:19:28:C7
*** Error in `/usr/bin/kismet_server': double free or corruption (out): 0x00007fffec000ac0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x730db)[0x7ffff67b70db]
/lib64/libc.so.6(+0x785ce)[0x7ffff67bc5ce]
/lib64/libc.so.6(+0x78dbb)[0x7ffff67bcdbb]
/usr/lib64/kismet/ubertooth.so(_ZN22PacketSource_Ubertooth4PollEv+0xab)[0x7ffff58a70fb]
/usr/bin/kismet_server(_ZN19Packetsourcetracker4PollER6fd_setS1_+0x31a)[0x468f7a]
/usr/bin/kismet_server(main+0x17ed)[0x42e99d]
/lib64/libc.so.6(__libc_start_main+0xf0)[0x7ffff6763fa0]
/usr/bin/kismet_server[0x42faa0]
======= Memory map: ========
00400000-005aa000 r-xp 00000000 08:02 6861207                            /usr/bin/kismet_server
007a9000-007aa000 r--p 001a9000 08:02 6861207                            /usr/bin/kismet_server
007aa000-007ad000 rw-p 001aa000 08:02 6861207                            /usr/bin/kismet_server
007ad000-00817000 rw-p 00000000 00:00 0                                  [heap]
7fffec000000-7fffec021000 rw-p 00000000 00:00 0 
7fffec021000-7ffff0000000 ---p 00000000 00:00 0 
7ffff2914000-7ffff2915000 ---p 00000000 00:00 0 
7ffff2915000-7ffff3115000 rw-p 00000000 00:00 0                          [stack:21185]
7ffff3115000-7ffff3116000 ---p 00000000 00:00 0 
7ffff3116000-7ffff3916000 rw-p 00000000 00:00 0                          [stack:21184]
7ffff3916000-7ffff392b000 r-xp 00000000 08:02 3020476                    /lib64/libz.so.1.2.8
7ffff392b000-7ffff3b2a000 ---p 00015000 08:02 3020476                    /lib64/libz.so.1.2.8
7ffff3b2a000-7ffff3b2b000 r--p 00014000 08:02 3020476                    /lib64/libz.so.1.2.8
7ffff3b2b000-7ffff3b2c000 rw-p 00015000 08:02 3020476                    /lib64/libz.so.1.2.8
7ffff3b2c000-7ffff3d37000 r-xp 00000000 08:02 6828741                    /usr/lib64/libcrypto.so.1.0.0
7ffff3d37000-7ffff3f36000 ---p 0020b000 08:02 6828741                    /usr/lib64/libcrypto.so.1.0.0
7ffff3f36000-7ffff3f52000 r--p 0020a000 08:02 6828741                    /usr/lib64/libcrypto.so.1.0.0
7ffff3f52000-7ffff3f5e000 rw-p 00226000 08:02 6828741                    /usr/lib64/libcrypto.so.1.0.0
7ffff3f5e000-7ffff3f61000 rw-p 00000000 00:00 0 
7ffff3f61000-7ffff3f74000 r-xp 00000000 08:02 7347906                    /usr/lib64/kismet/aircrack-kismet.so
7ffff3f74000-7ffff4173000 ---p 00013000 08:02 7347906                    /usr/lib64/kismet/aircrack-kismet.so
7ffff4173000-7ffff4174000 r--p 00012000 08:02 7347906                    /usr/lib64/kismet/aircrack-kismet.so
7ffff4174000-7ffff4175000 rw-p 00013000 08:02 7347906                    /usr/lib64/kismet/aircrack-kismet.so
7ffff4175000-7ffff4183000 rw-p 00000000 00:00 0 
7ffff4183000-7ffff418f000 r-xp 00000000 08:02 7348388                    /usr/lib64/kismet/spectool_net.so
7ffff418f000-7ffff438f000 ---p 0000c000 08:02 7348388                    /usr/lib64/kismet/spectool_net.so
7ffff438f000-7ffff4390000 r--p 0000c000 08:02 7348388                    /usr/lib64/kismet/spectool_net.so
7ffff4390000-7ffff4391000 rw-p 0000d000 08:02 7348388                    /usr/lib64/kismet/spectool_net.so
7ffff4391000-7ffff4396000 r-xp 00000000 08:02 7348389                    /usr/lib64/kismet/kismet-syslog.so
7ffff4396000-7ffff4596000 ---p 00005000 08:02 7348389                    /usr/lib64/kismet/kismet-syslog.so
7ffff4596000-7ffff4597000 r--p 00005000 08:02 7348389                    /usr/lib64/kismet/kismet-syslog.so
7ffff4597000-7ffff4598000 rw-p 00006000 08:02 7348389                    /usr/lib64/kismet/kismet-syslog.so
7ffff4598000-7ffff45b6000 r-xp 00000000 08:02 6849727                    /usr/lib64/libbluetooth.so.3.18.1
7ffff45b6000-7ffff47b5000 ---p 0001e000 08:02 6849727                    /usr/lib64/libbluetooth.so.3.18.1
7ffff47b5000-7ffff47b6000 r--p 0001d000 08:02 6849727                    /usr/lib64/libbluetooth.so.3.18.1
7ffff47b6000-7ffff47b9000 rw-p 0001e000 08:02 6849727                    /usr/lib64/libbluetooth.so.3.18.1
7ffff47b9000-7ffff47cc000 r-xp 00000000 08:02 7348217                    /usr/lib64/kismet/btscan.so
7ffff47cc000-7ffff49cc000 ---p 00013000 08:02 7348217                    /usr/lib64/kismet/btscan.so
7ffff49cc000-7ffff49cd000 r--p 00013000 08:02 7348217                    /usr/lib64/kismet/btscan.so
7ffff49cd000-7ffff49ce000 rw-p 00014000 08:02 7348217                    /usr/lib64/kismet/btscan.so
7ffff49ce000-7ffff49d5000 r-xp 00000000 08:02 7347951                    /usr/lib64/kismet/autowep-kismet.so
7ffff49d5000-7ffff4bd4000 ---p 00007000 08:02 7347951                    /usr/lib64/kismet/autowep-kismet.so
7ffff4bd4000-7ffff4bd5000 r--p 00006000 08:02 7347951                    /usr/lib64/kismet/autowep-kismet.so
7ffff4bd5000-7ffff4bd6000 rw-p 00007000 08:02 7347951                    /usr/lib64/kismet/autowep-kismet.so
7ffff4bd6000-7ffff4bda000 r-xp 00000000 08:02 3018580                    /lib64/libattr.so.1.1.0
7ffff4bda000-7ffff4dd9000 ---p 00004000 08:02 3018580                    /lib64/libattr.so.1.1.0
[... 144 lines cut ...]
        locked = <optimized out>
        __func__ = "_int_free"
#5  0x00007ffff58a70fb in PacketSource_Ubertooth::Poll (this=0x7ffff7f43010) at packetsource_ubertooth.cc:508
        newpack = 0x7bb570
        pkt = 0x7fffec000ac0
        rawchunk = 0x7dcd50
        x = 0
        rx = 98 'b'
        read_size = <optimized out>
        process_packet = <optimized out>
#6  0x0000000000468f7a in Packetsourcetracker::Poll (this=0x7b9180, in_rset=..., in_wset=...) at packetsourcetracker.cc:629
        capd = 27
        x = {first = 1, second = }
#7  0x000000000042e99d in main (argc=<optimized out>, argv=<optimized out>, envp=<optimized out>) at kismet_server.cc:1327
        x = 2
        startroot = <optimized out>
        rset = {fds_bits = {134217728, 0 <repeats 15 times>}}
        wset = {fds_bits = {0 <repeats 16 times>}}
        nlwc = <optimized out>
        hdwc = <optimized out>
        chantracker = <optimized out>
        configfilename = <optimized out>
        option_idx = 0
        startup_ipc_id = 6
...

dominicgs, 2015-02-18:
Thank you for the detailed report - the issue appears to be similar to, if not the same as, issue #18.  I'll leave this issue open until I've verified that it's the same or fixed them both.

The fact that it happens after two packets is interesting.  Is that reliably after the second packet each time?

ikelos, 2015-02-18:
Yep, whether the LAP is the same for the two packets or not, I haven't had an occasion where the crash hasn't happened after the second packet.

ohyeahq, 2016-04-28:
The same here.

Antergos latest rolling; others latest stable

Any workaround or quick hack please?

I know double free is notoriously hard to fix, especially in C++; we often have to put null in the pointer explicitly on initialization and right after freeing; we often have to write our own copy constructor and operator = (instead of relying on the auto-generated ones).  

I'm in a hurry and I just commented out line 508.  Leaky, yes, very bad, but working (for now).  

Cheeeeers

dominicgs, 2016-05-13:
I have now merged pull request #193 that should fix this issue.

ohyeahq, 2016-05-17:
Thanks so much for your effort.  I truly appreciate it.  The git clone works fine for me now for this issue.  A bad news is that the 14-bytes issue (#92) is still there.  I'll leave a report on #92.

dominicgs, 2016-05-17:
@ohyeahq #92 is at the top of my todo list and will be completed as soon as possible.

I'm going to close this issue as fixed.

closed 2016-05-17 by dominicgs