greatscottgadgets/ubertooth #60: ubertooth-btle reports strange channels

ikarus23, 2015-02-25:
This is what I get:
* Set the channel to 2404
* Start `ubertooth-btle` in promiscuous mode (`-p`)
* Channel is 2440 (all the time)

```
ut@ut-VirtualBox ~ % ubertooth-util -V
ubertooth git-c9d3ed7 (ut@ut-VirtualBox) Do 12. Feb 17:21:12 CET 2015
ut@ut-VirtualBox ~ % ubertooth-util -r
Resetting ubertooth device number 0
ut@ut-VirtualBox ~ % ubertooth-util -c2402
ut@ut-VirtualBox ~ % ubertooth-util -c    
Current frequency: 2402 MHz (Bluetooth channel 0)
ut@ut-VirtualBox ~ % ubertooth-btle -p    
systime=1424861944 freq=2440 addr=f61f626a delta_t=15159.477 ms
01 00 99 f3 65 
Data / AA f61f626a (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 0  MD: 0

    Data: 
    CRC:   99 f3 65

systime=1424861944 freq=2440 addr=f61f626a delta_t=40.822 ms
01 00 99 f3 65 
Data / AA f61f626a (valid) /  0 bytes
    Channel Index: 17
    LLID: 1 / LL Data PDU / empty or L2CAP continuation
    NESN: 0  SN: 0  MD: 0

    Data: 
    CRC:   99 f3 65

[...]

^C
ut@ut-VirtualBox ~ % ubertooth-util -c    
Current frequency: 2440 MHz (Bluetooth channel 38)
```

Questions:
* Why is `ubertooth-btle` not sniffing on 2404?
* If it had found a connection and now follows it, why it the channel not hopping (always 2440).
* Maybe related to https://github.com/greatscottgadgets/ubertooth/issues/48 ?

dominicgs, 2015-02-25:
The ```ubertooth-util -c``` settings are for Basic Rate Bluetooth channels, they are not the same as BLE channels as the two versions of Bluetooth use different channel numbering and spacing.

I think it's unlikely that this is related to #48 as they will be hitting different parts of the firmware code.

I'll leave @mikeryan to comment on the usage of ```ubertooth-btle```

mikeryan, 2015-02-26:
The promiscuous mode code forces the device onto channel 2440 during the first phases of connection parameter recovery. In the example snippets you've discovered above, the device is not actually following a connection and is still attempting to recover connection parameters. Once the device begins following a connection, ```ubertooth-util -c``` will behave as expected.

closed 2015-02-26 by mikeryan

ikarus23, 2015-02-26:
Thanks for clearing that up, guys!
So [this README](https://github.com/greatscottgadgets/ubertooth/blob/master/host/README.btle#L82) is outdated? Or did I misunderstood something.

Oh, and by the way: Thanks for all the effort you put into this project! I really appreciate it.

mikeryan, 2015-02-26:
Yes, that file is extremely outdated.