greatscottgadgets/ubertooth #61: Wireshark: User encapsulation not handled: DLT=147 on Kali Linux

redirected, 2015-02-27:
Hello all,

I am running with an Ubertooth One on a Kali Linux machine (Debian 3.14.5-1kali1 i686 GNU/Linux)

I have:

pyusb-1.0.0a3
libbtbb-2014-02-R4
ubertooth-2014-02-R2
wireshark-1.10.2

When I load the output generated from ubertooth-btle, wireshark only gives me the PPI dissection.
It says:
"User encapsulation not handled: DLT=147, check your
Preferences->Protocols->DLT_USER"

This is a comon packet:

0000   d6 be 89 8e 40 1d 69 64 18 a8 fd 60 02 01 1a 13  ....@.id...`....
0010   ff 4c 00 0c 0e 00 a4 07 81 db 52 32 2e 7e a3 d3  .L........R2.~..
0020   c7 69 0b 18 cf f4                                .i....

I don't have experience with previous version of the firmware, and I'm working with ubertooth since 2 days. I followed this page: https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide so everything seems to be correctly installed.

Any help is appreciated, thankyou.

MatthiasU, 2015-03-02:
go to Preferences->Protocols->DLT_USER and enter "btle" as protocol 
that should help

redirected, 2015-03-02:
Ok, this solved the problem, thankyou.

closed 2015-03-02 by redirected

jlubeck, 2015-11-11:
I'm getting this error whe addiing btle as protocol:

error in column 'Payload protocol': dissector not found

Any ideas?

tahersb, 2015-12-09:
@jlubeck What is the version of wireshark that you are using? 
From the build guide - "Wireshark version 1.12 and newer includes the Ubertooth BLE plugin by default"

jlubeck, 2015-12-09:
1.10.6. But when I try to do an apt-upgrade, it's saying that wireshark is already the newest version

dominicgs, 2015-12-09:
I've forgotten the filename of the dissector, but it's probably btle.so

Could you take a look for that file and see where it installed to? The output from the build process should show this.

nandra, 2015-12-10:
Have same issue on ubuntu 14.04, then use ppa [0] to install latest stable wireshark and it start working

[0] - https://launchpad.net/~wireshark-dev/+archive/ubuntu/stable

tahersb, 2015-12-11:
Thats correct, Ubuntu 14.04 distributes back ported wireshark versions. The following worked well for me:
1. Follow instructions on the link posted by @nandra under "Adding this PPA to your system" -> Read about installing
2. sudo apt-get update
3. sudo apt-get -VV install wireshark wireshark-dev (inspect the versions. Should be 2.0.0 at this time)
4. Make sure your user is added to wireshark group - sudo usermod -aG wireshark <your_username>
5. Launch Wireshark (GTK+)
6. Follow all instructions at https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark
7. Enjoy with ubertooth-btle -c /tmp/pipe -p

IgorGanapolsky, 2017-04-06:
@MatthiasU I do not see **DLT_USER** as an option.