greatscottgadgets/ubertooth #62: wireshark is only showing L2Cap Fragments

MatthiasU, 2015-03-02:
I use the ubertooth mostly in promiscuous mode. The Ubertooth works great and after a few seconds it enters the following mode after retrieving the hopping interval. If I issue a read request on my device , the  ubertooth response immediately with a L2CAP Fragment. The problem is Wireshark can not decode that packet and shows me the L2CAP fragment and nothing else. 

I noticed that the header length of a package is 24 bytes . I used the ubertooth half a year ago and I compared the packages with those old captured files. I noticed the header length was only 19 bytes . 
Could PCAP be my problem in this scenario 

Thanks !!

MatthiasU, 2015-03-02:
FYI: I use the latest ubertooth master & latest libbtbb master & wireshark 1.12.3

MatthiasU, 2015-03-02:
fixed that issue ... was my fault

closed 2015-03-02 by MatthiasU

Rahul11111, 2016-05-27:
Hello @MatthiasU,  I am facing the same issue. So, can you elaborate how did you solve this issue?

MatthiasU, 2016-05-30:
Hi Rahul,

Which version of wireshark do u use ?

ajgay, 2017-01-16:
I've got the same issue. Wireshark 2.2.3

mikeryan, 2017-01-16:
This is expected behavior in promiscuous mode. Most of the "packets" being logged are just random noise from the air. Only once a connection has been successfully followed will the packets contain meaningful data.

I recommend using follow (-f) mode rather than promiscuous mode. It's much more reliable and is probably The Thing You Want.

ajgay, 2017-01-17:
Odd. I'm actually running in -f rather than -p but am getting a series of unknown source and destination addressed PDUs and SMP messages. I'm not seeing either rand value or the ENC request or response.

alichator, 2017-03-11:
I am getting the same behavior. As far as I understand the unknown source and destination are expected behavior as the packets don't have source or destination addresses. The master and slave just alternate sending packets on the access address (many of these being empty PDUs).

However, the concerning thing is that after a certain amount of follow traffic, nothing further is captured. My situation sounds the exact same as yours and of the person in the link you have provided. I suspect that perhaps ubertooth gets out of sync with the channel selection algorithm, but that is just speculation. 

@mikeryan any thoughts on what could be causing this behavior?

mikeryan, 2017-03-11:
@alichator if you're able to follow a partial connection but packets vanish, what's most likely occurred is that the connection issued a channel map update. This changes the connection to support only a subset of the data channels, and Ubertooth currently does not process these changes. Are you seeing anything like `CHANNEL_MAP_UPDATE_REQ` in your captured data?

alichator, 2017-03-16:
I only see a LL_CONNECTION_UPDATE_REQ. I did find that the connection does seem to be moved to a subset of the data channel when examining with promiscuous mode. I have tested many times, is it possible that a CHANNEL_MAP_UPDATE_REQ would be missed by ubertooth consistently?

The only other thing being sent is read/write requests over ATT, but its not possible to modify the BTLE behavior using these as far as I understand.

pabigot, 2017-07-22:
This is probably what's affecting me too.  In the CONNECT_REQ from the phone app (Nordic Thingy52) I see a custom channel map that wipes out 8-19 inclusive; then I get very spotty fragments from a source the phone sees reliably.

Is this on anybody's task list?  As best I can tell adding support requires these changes to bluetooth_rxtx, which seem fairly straightforward, but I won't have a chance to look at this for at least a few weeks.

* modify connection_follow_cb in LINK_LISTENING to extract the channel map;
* modify le_state_t to record the custom map;
* modify btle_next_hop to respect the custom map.