greatscottgadgets/ubertooth #63: Partial sniffing with Kismet

redirected, 2015-03-02:
I implemented ubertooth + kismet to sniff encrypted and/or unencrypted traffic between two devices (e.g. two smartphones). I suceeded in compiling/installing proper files following this page https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide.

Now I'm able to run Kismet and dump traffic in a .pcapbtbb external file. Wireshark is able to read and decode its content.

The problem is that in this file I always get partial data.  I always get only 14bytes packets and no data payload. 

My expectation was to eventually find the (encrypted) payload of the buetooth connection. Was I wrong?

dominicgs, 2015-03-02:
@redirected The answer to your question depends on what you are trying to sniff.

If you are trying to sniff a BLE connection, then there are no Kismet plugins and you should use the ```ubertooth-btle``` tool to follow connections.

If you are trying to sniff Basic Rate Bluetooth, then the Kismet plugins are a good place to perform a survey of local devices, but are not able to follow a connection and monitor all packets.  We currently do not have any tools that will do this.  The ```ubertooth-follow``` tool gets close, but it is not robust enough to follow a connection for long.  I would suggest trying to use that instead of the Kismet plugins.

dominicgs, 2015-03-02:
These blog posts give a good introduction to the problems with Basic Rate sniffing:
http://ubertooth.blogspot.co.uk/2013/02/motivating-problem.html
http://ubertooth.blogspot.co.uk/2014/06/discovering-bluetooth-uap.html

redirected, 2015-03-02:
As I don't have complete knowledge of devices,  I could not determine if they were using BLE. After various tests I believe they are not using BTLE. There are mainly two reasons supporting this: 
- hcitool lescan doesn't display those devices
- sniffing with ubertooth-btle doesn't give useful output in wireshark.

Supposing they are using Basic Rate Buletooth, what are the differences between remote and local devices? Is ubertooth-follow capable of dumping the traffic in a pcap-like file?

dominicgs, 2015-03-02:
If you are unsure about the devices it is most likely that they are Basic Rate.

The ubertooth-follow tool does not yet output pcap files, but it will do with the next release (later this month).

redirected, 2015-03-02:
I see. Suppose I have two paired smartphones, and I want to sniff their traffic  from my laptop. My laptop has got the ubertooth device and an integrated bluetooth device. My whole laptop has nothing to do with the piconet and the paired devices.
Is this a typical "remote configuration" ? Does ubertooth-follow work with this configuration?

KamahI, 2016-08-25:
Hello, we seem to have a very similar problem like "redirected" posted. Doen anyone have a solution or did you "redirected" find anything helpful?
I know this thread is very old but it fits exactly what we need..

closed 2020-12-03 by mikeryan