greatscottgadgets/ubertooth #69: USB timeout on RX operations

reunion974, 2015-03-30:
Hello,

My ubertooth One have been sitting for 1 year w/o being used. Recently, I've been trying using the various uberooth tools on pentoo and ubuntu 14.04, with 2012 and 2014 firmware. The RX related commands (specan-ui, ubertooth-rx, ubertooth-scan) always fail with a timeout. After such a timeout occurs, even simple commands as sudo ubertooth-util -v fails. I have to unplug re-plug the ubertooth to make it functional again. After I run ubertooth-specan-ui, the RX led remains activated.

Can someone please confirm that this is an issue with the hardware ? Otherwise is there anything I can try ?

Thanks a lot for your support.

Firmeware details :
- Firmware revision: git-4412704
- ubertooth git-4412704 (dominicgs@mercury) Wed Jan 29 20:38:22 GMT 2014
- Serial No: 0606041f535721b64f81a296f5000005

Here are the various terminal captures :

sudo ubertooth-rx
rx_xfer status: Transfer timed out. (2)

sudo ubertooth-scan -s -x
HCI scan
00:23:6C:A4:FB:7C  Anchaing
C0:EE:FB:21:F7:73  OnePlus One
00:22:41:D6:C1:41  MacBook 

Ubertooth scan
rx_xfer status: Transfer timed out. (2)

sudo ubertooth-specan-ui 
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/bin/ubertooth-specan-ui", line 48, in run
    for frequency_axis, rssi_values in frame_source:
  File "/usr/local/lib/python2.7/dist-packages/specan/Ubertooth.py", line 63, in specan
    buffer = self._device.read(0x82, 64)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 670, in read
    self.__get_timeout(timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 780, in bulk_read
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 889, in __read
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 571, in _check
    raise USBError(_str_error[ret], ret, _libusb_errno[ret])
USBError: [Errno 110] Operation timed out

Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 255, in closeEvent
    self._device.close()
  File "/usr/local/lib/python2.7/dist-packages/specan/Ubertooth.py", line 82, in close
    self._device.ctrl_transfer(0x40, 21)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 711, in ctrl_transfer
    self.__get_timeout(timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 836, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 571, in _check
    raise USBError(_str_error[ret], ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out
Traceback (most recent call last):
  File "/usr/local/bin/ubertooth-specan-ui", line 255, in closeEvent
    self._device.close()
  File "/usr/local/lib/python2.7/dist-packages/specan/Ubertooth.py", line 82, in close
    self._device.ctrl_transfer(0x40, 21)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 711, in ctrl_transfer
    self.__get_timeout(timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 836, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 571, in _check
    raise USBError(_str_error[ret], ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

dominicgs, 2015-03-30:
Hi,

Which version of the host code are you using?  Did you download 2014-02-R2 (i.e. the latest release) and follow the [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide)?

What hardware are you using for the host code?  Are you running inside a VM?

Given that control messages appear to work but bulk communication doesn't, I don't think that this sounds like a hardware issue, but I may be wrong.

reunion974, 2015-03-30:
Hi Dominic,

I followed today's version of the Build Guilde (2015/03/30) i.e. I
downloaded 2014-02-R2. There wasn't any problem during the install.
I tried on two differents laptop that showed the same results :
- Ubuntu14.04-64bit/ intel i3
- the latest distrib of Pentoo 32bit on a Core2Duo.

None of them were inside a VM or executing a VM.

I forgot to say that I don't know if the TX operations are working but at
least they don't result in a timeout.

Thanks

dominicgs, 2015-03-30:
Thanks for the extra information.  This isn't a problem that I've seen before.

Could you give me the output of ```lsusb -vd 1d50:6002``` while the Ubertooth is plugged in to the system?

Could you also check dmesg for the messages that get logged when the Ubertooth is plugged in?

dominicgs, 2015-03-30:
Please also try ```ubertooth-btle -p```.  This will help us to diagnose the problem.

reunion974, 2015-03-30:
The command "sudo ubertooth-btle -p" did not return anything, not even an
error. I had to use Ctl^C. I don't think there any BLE frames around so it
sound pretty normal. I'm configuring some BLE dongles to generate a bit of
LE trafic. I'll keep you posted asap.

Here are the output of the lsusb and dmesg.

sudo lsusb -vd 1d50:6002

Bus 002 Device 031: ID 1d50:6002 OpenMoko, Inc.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1d50 OpenMoko, Inc.
  idProduct          0x6002
  bcdDevice            1.00
  iManufacturer           1 http://ubertooth.sourceforge.net/
  iProduct                2 bluetooth_rxtx
  iSerial                 3 00000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

dmesg

[15002.085288] usb 2-1.1: USB disconnect, device number 31
[15008.935315] usb 2-1.1: new full-speed USB device number 32 using ehci-pci
[15009.030386] usb 2-1.1: New USB device found, idVendor=1d50,
idProduct=6002
[15009.030396] usb 2-1.1: New USB device strings: Mfr=1, Product=2,
SerialNumber=3
[15009.030402] usb 2-1.1: Product: bluetooth_rxtx
[15009.030406] usb 2-1.1: Manufacturer: http://ubertooth.sourceforge.net/
[15009.030410] usb 2-1.1: SerialNumber: 00000001

reunion974, 2015-03-30:
Hi Dominic,

Even with some beacon being advertised around, the "sudo ubertooth-btle -p"
command returned no result/error.

Thanks

2015-03-30 20:43 GMT+04:00 Pierre Ugo Tournoux <tournoux@gmail.com>:

dominicgs, 2015-03-30:
Hi,

Please email me (dominicgs@gmail.com) and I will arrange for a replacement Ubertooth to be sent to you.

Dominic

closed 2015-03-30 by dominicgs