greatscottgadgets/ubertooth #72: undefined symbols _decode_le / _le_print

ghazel, 2015-04-22:
libpcap is from homebrew
libbtbb (at https://github.com/greatscottgadgets/libbtbb/commit/a3df4d72865e8aab15ce6f57e3c5d23fb12a5c94 ) is hand compiled. I can verify it's using the correct libpcap:

```
$ otool -L /usr/local/lib/libbtbb.dylib 
/usr/local/lib/libbtbb.dylib:
	libbtbb.0.dylib (compatibility version 0.0.0, current version 0.3.0)
	/usr/local/opt/libpcap/lib/libpcap.A.dylib (compatibility version 1.0.0, current version 1.6.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
``` 

However, compiling ubertooth fails:
```
ghazel@MacBook-Pro:~/projects/ubertooth/ubertooth-2014-02-R2/host/build$ cmake ..
-- The C compiler identification is AppleClang 6.1.0.6020049
-- The CXX compiler identification is AppleClang 6.1.0.6020049
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.28") 
-- Found LIBBTBB: /usr/local/lib/libbtbb.dylib  
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Found LIBUSB: /usr/local/lib/libusb-1.0.dylib  
-- Pcap include dirs set to /usr/local/opt/libpcap/include
-- Pcap library set to  /usr/local/opt/libpcap/lib/libpcap.dylib
-- Looking for pcap_version
-- Looking for pcap_version - found
-- Looking for pcap_open_dead
-- Looking for pcap_open_dead - found
-- Looking for pcap_freecode
-- Looking for pcap_freecode - found
-- Looking for pcap_breakloop
-- Looking for pcap_breakloop - found
-- Looking for pcap_create
-- Looking for pcap_create - found
-- Looking for pcap_datalink_name_to_val
-- Looking for pcap_datalink_name_to_val - found
-- Looking for pcap_datalink_val_to_description
-- Looking for pcap_datalink_val_to_description - found
-- Looking for pcap_datalink_val_to_name
-- Looking for pcap_datalink_val_to_name - found
-- Looking for pcap_findalldevs
-- Looking for pcap_findalldevs - found
-- Looking for pcap_free_datalinks
-- Looking for pcap_free_datalinks - found
-- Looking for pcap_get_selectable_fd
-- Looking for pcap_get_selectable_fd - found
-- Looking for pcap_lib_version
-- Looking for pcap_lib_version - found
-- Looking for pcap_list_datalinks
-- Looking for pcap_list_datalinks - found
-- Looking for pcap_set_datalink
-- Looking for pcap_set_datalink - found
-- Looking for pcap_open
-- Looking for pcap_open - not found
-- Looking for pcap_findalldevs_ex
-- Looking for pcap_findalldevs_ex - not found
-- Looking for pcap_createsrcstr
-- Looking for pcap_createsrcstr - not found
-- Found PCAP: /usr/local/opt/libpcap/include  
-- Pcap include dirs set to /usr/local/opt/libpcap/include
-- Pcap library set to  /usr/local/opt/libpcap/lib/libpcap.dylib
-- checking for module 'bluez'
--   package 'bluez' not found
-- Could NOT find LIBBLUETOOTH (missing:  LIBBLUETOOTH_LIBRARIES LIBBLUETOOTH_INCLUDE_DIR) 
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.9") 
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   ubertooth

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/ghazel/projects/ubertooth/ubertooth-2014-02-R2/host/build
```

```
ghazel@MacBook-Pro:~/projects/ubertooth/ubertooth-2014-02-R2/host/build$ make
Scanning dependencies of target ubertooth
[  9%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
[ 18%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_control.c.o
/Users/ghazel/projects/ubertooth/ubertooth-2014-02-R2/host/libubertooth/src/ubertooth_control.c:612:13: warning: implicit declaration of function 'btbb_gen_syncword' is invalid in C99
      [-Wimplicit-function-declaration]
        syncword = btbb_gen_syncword(address & 0xffffff);
                   ^
1 warning generated.
Linking C shared library libubertooth.dylib
Undefined symbols for architecture x86_64:
  "_decode_le", referenced from:
      _cb_btle in ubertooth.c.o
      _log_packet in ubertooth.c.o
  "_le_print", referenced from:
      _cb_btle in ubertooth.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libubertooth/src/libubertooth.0.2.dylib] Error 1
make[1]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
make: *** [all] Error 2
```

dominicgs, 2015-04-22:
This looks like libbtbb from git and the last release of ubertooth. Is that correct?

The versions need to be matched, so you'll need to remove libbtbb and install the latest release (2014-02-R?).

fractalspace, 2015-04-27:
Did you resolve these issues?
    -- checking for module 'bluez'
    --   package 'bluez' not found
    -- Could NOT find LIBBLUETOOTH (missing:  LIBBLUETOOTH_LIBRARIES  LIBBLUETOOTH_INCLUDE_DIR)

dominicgs, 2015-04-27:
@fractalspace That doesn't look like it's related to this issue.  I don't believe that's a fatal error, just a warning.  You can go ahead and run make.

If you want to build the ubertooth-follow and ubertooth-scan tools and you're one a linux host, then you can install libBluetooth (bluez) from your distribution's packaging system.

closed 2015-04-29 by dominicgs