greatscottgadgets/ubertooth #74: cmake Segmentation fault

hooow, 2015-05-02:
There is a *** Segmentation fault *** error when installing libbtbb.
I just followed the full steps;
wget https://github.com/greatscottgadgets/ubertooth/archive/master.zip
tar xf libbtbb-master.zip
cd libbtbb-master
mkdir build
cd build
cmake ..
However, when I got there, I always get ;
root@kalipen:~/libbtbb-master/build# cmake
Segmentation fault

Please help me figure out what is the problem?
Thx.

krasin, 2015-05-02:
Hi hooow,

disclaimer: I am not a member of ubertooth/libbtbb team, just a follower of the project.

I would recommend to run cmake under strace to see which command actually fails.

```
strace -f -e trace=execve -s1000 cmake 2>&1 | tee output.log
```  

Please, a few last lines from output.log after that.

hooow, 2015-05-02:
Hi krasin,
Please see following info;
////////////////////////////////////////////////
root@kalipen:~/libbtbb-master# cd build/
root@kalipen:~/libbtbb-master/build# cmake
Segmentation fault
root@kalipen:~/libbtbb-master/build# strace -f -e trace=execve -s1000 cmake 2>&1 | tee output.log
execve("/bin/cmake", ["cmake"], [/* 31 vars */]) = -1 EFAULT (Bad address)
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_KERNEL, si_addr=0} ---
+++ killed by SIGSEGV +++
//////////////////////////////////////////////
What is meanning?
I just start to explore the Linux and ubertooth.
Thx.

krasin, 2015-05-02:
Thanks for posting this. The error has nothing to do with ubertooth/libbtbb, and there's something wrong with your /bin/cmake binary itself. It might be that it's corrupted, or resides on a network or virtual file system that is not accessible, or something else.

A simple way to verify that is

```
readelf -a /bin/cmake > /dev/null
```

If readelf is not found, you can install it as a part of elfutils package, like ```sudo apt-get install elfutils```.

Also, it seems that your cmake is located unusually (/bin/cmake), while my system has it in /usr/bin.

hooow, 2015-05-03:
Thank you very much!!! I got the installation finished! That is the cmake installation error.

Thanks a lot!

krasin, 2015-05-03:
I am glad to hear your issue is resolved. :)

hooow, 2015-05-03:
Hi krasin,
I have one more question here.
checking for module 'libusb-1.0'
--   package 'libusb-1.0' not found
-- Could NOT find LIBUSB (missing:  LIBUSB_LIBRARIES LIBUSB_INCLUDE_DIR) 
-- Pcap include dirs set to /usr/local/include
-- Pcap library set to  /usr/local/lib/libpcap.so
-- Pcap include dirs set to /usr/local/include
-- Pcap library set to  /usr/local/lib/libpcap.so
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIBUSB_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /root/ubertooth-2014-02-R2/host/libubertooth/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
   used as include directory in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
LIBUSB_LIBRARIES (ADVANCED)
    linked by target "ubertooth" in directory /root/ubertooth-2014-02-R2/host/libubertooth/src
    linked by target "ubertooth-util" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-rx" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-dump" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-scan" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-debug" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-follow" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src
    linked by target "ubertooth-btle" in directory /root/ubertooth-2014-02-R2/host/ubertooth-tools/src

However, I have already installed the libusb-1.0
root@kalipen:~# sudo apt-get install libusb-1.0-0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0 is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

So how to configure this?
Thank you for your time.

krasin, 2015-05-03:
What does ```pkg-config --libs libusb-1.0``` say?

hooow, 2015-05-03:
I have solved this problem, thanks :)
root@kalipen:~# pkg-config --libs libusb-1.0
-L/usr/local/lib -lusb-1.0  
It almost takes me 1 hours to install libudev...

Finally, I got there.
root@kalipen:~/ubertooth-2014-02-R2/host/build# cmake ..
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Found LIBUSB: /usr/local/lib/libusb-1.0.so  
-- Pcap include dirs set to /usr/local/include
-- Pcap library set to  /usr/local/lib/libpcap.so
-- Pcap include dirs set to /usr/local/include
-- Pcap library set to  /usr/local/lib/libpcap.so
-- Configuring done
-- Generating done
-- Build files have been written to: /root/ubertooth-2014-02-R2/host/build

:)

closed 2015-05-10 by hooow