greatscottgadgets/ubertooth #77: ubertooth-follow error

dwcash, 2015-05-11:
This is the error I am getting when ever I try to use ubertooth-follow. I am running Kali in a VM in VMWare Player.

root@kali:~# ubertooth-follow 
No address given, reading address from device
LAP=b96eee UAP=aa
libUSB Error: Timeout:  (-7)

I updated my firmware with ubertooth-one-bin-firmware.dfu but the version being shown is not what I expected.

root@kali:~# ubertooth-util -v
Firmware revision: git-4412704

Any thoughts???

-David C

dominicgs, 2015-05-11:
That is the firmware version from the last release.  There was a mistake when it was built and we did not correctly replace the version string with the release name.

Do the other functions work?  e.g. ```ubertooth-rx```?

From the output it looks as though you are trying to monitor a connection for which the local Bluetooth controller is the master device.  Is that correct?

dwcash, 2015-05-12:
Yes, ubertooth-rx and ubertooth-btle seem to work.  ubertooth-follow doesn't seem to work even if I specify a LAP with the -l option. It will list a different LAP and then hang or give the Timeout error.

root@kali:~# ubertooth-follow -l 36c709
No address given, reading address from device
LAP=2a405e UAP=06
^C
root@kali:~# ubertooth-follow -l36c709
No address given, reading address from device
LAP=1340ae UAP=b8
libUSB Error: Timeout:  (-7)
^C
root@kali:~# ubertooth-follow -l36c709
No address given, reading address from device
LAP=4da7de UAP=a5
libUSB Error: Timeout:  (-7)

dominicgs, 2015-05-12:
If you specify a LAP you also need to specify a UAP.  I'll update the output to make this more clear.

I've just taken a look through the firmware history and found that it git-4412704 is older than I thought.  Please download the [source archive](https://github.com/greatscottgadgets/ubertooth/releases/download/2014-04-R1/ubertooth-2014-04-R1.tar.xz) and update with the ```ubertooth-one-firmware-bin/bluetooth_rxtx.dfu``` file.

dwcash, 2015-05-14:
It seems to be working now. Thanks for the assist.

dominicgs, 2015-05-14:
No problem.  I'll close this issue.

closed 2015-05-14 by dominicgs