greatscottgadgets/ubertooth #78: Kismet (make plugins) problem

elmalmeida, 2015-06-03:
Hello guys,

Anyone had the issue below? I'm using Ubuntu 14.04.1 on a x64.

make[1]: Entering directory `/usr/local/src/ubertooth-master/host/kismet/plugin-ubertooth'
g++ -I/usr/include/ncurses   -Wall -g -O2 -I/usr/local/src/kismet-2013-03-R1b -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
packetsource_ubertooth.cc: In function ‘void* ubertooth_cap_thread(void*)’:
packetsource_ubertooth.cc:190:32: error: too many arguments to function ‘int cmd_rx_syms(libusb_device_handle*)’
  cmd_rx_syms(ubertooth->devh, 0);
                                ^
In file included from /usr/local/include/ubertooth.h:25:0,
                 from packetsource_ubertooth.h:34,
                 from packetsource_ubertooth.cc:41:
/usr/local/include/ubertooth_control.h:78:5: note: declared here
 int cmd_rx_syms(struct libusb_device_handle* devh);
     ^
make[1]: *** [packetsource_ubertooth.o] Error 1
make[1]: Leaving directory `/usr/local/src/ubertooth-master/host/kismet/plugin-ubertooth'
make: *** [plugins] Error 2

Any help would be highly appreciated.

Thanks,

--
Emanuel

closed 2015-06-03 by elmalmeida

mikeryan, 2015-06-03:
Looks like a mismatch between libbtbb and Ubertooth. How did you resolve the issue?

dominicgs, 2015-06-03:
This is a mismatch between libubertooth and the plugin that I introduced recently.  One of the arguments to cmd_rx_syms() was the packet count, which we never used.

I have now fixed this.