greatscottgadgets/ubertooth #80: Ubertooth-scan doesn't work

Daniel-Katz, 2015-06-22:
I have run several Ubertooth commands with no issues, including Ubertooth-rx and Ubertooth-specan-ui, but when I try to run Ubertooth-scan, I get the following error:

Inspiron-1520:~$ ubertooth-scan
opening socket: No such device

I have the radio USB plugged in, and it seems to be working normally, so I can't understand why it isn't working.

dominicgs, 2015-06-22:
This looks like the same as issue #45 

ubertooth-scan uses a local Bluetooth dongle to query devices that Ubertooth finds.  In order to run it you need a Bluetooth device attached to the same host as the Ubertooth.  You also need permission to create connections from that device (i.e. you may need to use sudo)

Daniel-Katz, 2015-06-22:
I used sudo, and I have a/the dongle (USB with a couple of glowing green and red lights, with an antenna). Yes, this is exactly the same thing as #45, but no one explained how he solved the issue.

dominicgs, 2015-06-22:
You need a Bluetooth dongle as well as the Ubertooth.

Daniel-Katz, 2015-06-23:
there's another device I need, as well as the Ubertooth piece? I thought that WAS a Bluetooth dongle?

dominicgs, 2015-06-23:
No Ubertooth is not a Bluetooth dongle, it is a raw data sniffer for the 2.4GHz ISM band.

You don't have to have a separate Bluetooth dongle for most usage, just for ubertooth-scan and ubertooth-follow.

Daniel-Katz, 2015-06-23:
ahh, thanks. Now I know. Just to be clear, by bluetooth dongle, you mean something that can enable a computer to be able to connect to bluetooth devices, meaning that if I were using a computer that was able to connect to bluetooth devices, I wouldn't need a dongle?

dominicgs, 2015-06-23:
Good point, it doesn't need to be an external USB device.  If you have a Bluetooth controller built in to your computer, that will work.

```ubertooth-scan``` uses Ubertooth to find the addresses of non-discoverable devices and then attempts to use the Bluetooth controller to connect to and query the device.  It's an enhanced version of ```hcitool scan```.

Daniel-Katz, 2015-06-23:
I plugged in a blueooth device, and now it works, but when I try using scan:

Inspiron-1520:~$ sudo ubertooth-scan 
usb_claim_interface error -6

 Any idea what it is going on about?

dominicgs, 2015-06-23:
That's a strange one.  Are you running in a VM?  Do you have another terminal open with something controlling the Ubertooth?

Try unplugging and reconnecting the Ubertooth and try again?

Daniel-Katz, 2015-06-23:
I'm not running a VM. This laptop originally had Windows 7 but then I overwrote it with ubuntu. I'm trying it again with unplugging the ubertooth piece

Daniel-Katz, 2015-06-23:
IT WORKS!!! Thanks a ton man, I finally have this working

dominicgs, 2015-06-23:
No problem.  I will attempt to improve the documentation and error messages in future.

Daniel-Katz, 2015-06-23:
I'm trying to do the HCI scan that was in the Youtube video, using the same code:

sudo ubertooth-scan -b hcil -t 40 -x

but it is coming out differently, and not just showing found MAC addresses and associated names. I don't suppose you know why?

dominicgs, 2015-06-23:
Is ```hci1``` the right identifier for your dongle?
Are there active devices within range?

Some devices do not respond to inquiry from unknown devices (this was a security measure taken in the past few years, notably with iOS devices)

Daniel-Katz, 2015-06-23:
There are active devices around me (my iPhone included) and it found them, the only thing is, there was alot of other data thrown into the mix as well. For example, I got a lot of lines like these before it started showing the Mac Addresses and associated names:

systime=1435071290 ch=43 LAP=e7f022 err=0 clk100ns=612784237 clk1=8486653 s=-17 n=-77 snr=60

 and even after, the devices found are shown like so:

00:00:6F:E7:F0:22  Daniel Katz’s iPhone
	AFH Map=0x244000001ea1020a02bb

and for my testing purposes, I'm only trying to find the MAC addresses and names

Daniel-Katz, 2015-06-23:
Also, I don't know that hci1 is the right one. when I put hcil (lower case "L") since that's what I assumed the code was, it ran as I said above. However, when I put hcl1 it said not found. I just changed it to hci0 and now it ran again, just like before.

Daniel-Katz, 2015-06-23:
also, I tried replacing -x with -s, it showed HCI scan before ubertooth scan, but nothing under it, and then after the ubertooth scan which had the lines starting with systime, it then showed the MAC addresses, names etc

dominicgs, 2015-06-23:
I think that was an update to the code a while back to show more information.  You should be able to filter it with grep as follows: ```sudo ubertooth-scan -b hci0 -t 40 -x | grep '[0-9A-F]\{2\}\(:[0-9A-F]\{2\}\)\{5\}'```

I imagine that hcil failed and defaulted to hci0, whereas hci1 is a valid argument, so you get the error message.

Removing -x should stop the extended scan, this reducing the excess output.

dominicgs, 2015-06-23:
The first two bytes of the MAC will always be 00:00 as Bluetooth does not use them.  This is not a problem as every Bluetooth tool that I have used will ignore the first two bytes anyway.

Daniel-Katz, 2015-06-23:
i think you're right about the hcil failing, but I didn't get an error message. I just tried the code you posted without the -x and I got exactly what I was hoping for. Thanks a ton for your help.

closed 2015-06-23 by Daniel-Katz