greatscottgadgets/ubertooth #81: Ubertooth-rx timeout

rjacobs00, 2015-06-22:
Is there any timeout option when running Ubertooth-rx?

dominicgs, 2015-06-22:
There isn't, it just runs forever.

Do you mean a "run for x seconds before gracefully exiting" feature?

rjacobs00, 2015-06-22:
Yes thats what I was looking for. I was hoping to be able to gracefully exit, one reason being I do get the "uh oh, full_buf not emptied" error. Any thoughts on the best way to do something similar?

dominicgs, 2015-06-22:
The best solution would probably be for me or someone else to add that functionality.  ```ubertooth-scan``` has something similar.

There will be a fix for the "uh oh, full_buf not emptied" error (I can't remember if it's in git already).

rjacobs00, 2015-06-22:
Okay I will see if I can do something with python in the meantime using pyUbertooth. Im also trying to write the RSSI values to a csv file using the ubertooth-specan-ui as a guide.

I was reading an older post about the error this is the firmware version I'm using 
Firmware revision: git-4412704
I'm also running this through virtualbox which is the problem I think since I keep having to re-add the device back in after using ubertooth-rx.

dominicgs, 2015-06-23:
Virtualbox allows you to set up a filter that will automatically pass the Ubertooth to the VM.  You need to specify the USB VID/PID, which is 1D50:6002.  That should stop you from having to pass it every time you need to reset the device.

rjacobs00, 2015-06-23:
Great thanks! I will do that then. Keep up the good work!

elli89, 2015-06-30:
I have added a timeout option to ubertooth-rx. The default timeout value is 0 (no timeout), which means if the -t command line parameter is not provided, ubertooth-rx will sniff forever.

dominicgs, 2015-06-30:
I've checked the code, created a pull request and merged it.

@elli89 It looks great, thank you for the contribution!

closed 2015-06-30 by dominicgs