greatscottgadgets/ubertooth #82: CMakeLists.txt missing

ehumphri, 2015-06-26:
Trying to build the ubertooth tools using the instructions here: https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide

     root@ubuntu:/usr/local/src/ubertooth-2014-02-R2/build# cmake ..
     CMake Error: The source directory "/usr/local/src/ubertooth-2014-02-R2" does not appear to contain    CMakeLists.txt.
     Specify --help for usage, or press the help button on the CMake GUI.
     root@ubuntu:/usr/local/src/ubertooth-2014-02-R2/build#

OS:

     root@ubuntu:/usr/local/src/ubertooth-2014-02-R2# cat /etc/issue
     Ubuntu 14.04.2 LTS \n \l

cmake:

     root@ubuntu:/usr/local/src/ubertooth-2014-02-R2# cmake -version
     cmake version 2.8.12.2
     root@ubuntu:/usr/local/src/ubertooth-2014-02-R2# 

What am I missing?

dominicgs, 2015-06-26:
You're very close, but you're in the wrong directory.  The build directory needs to be created in ```ubertooth-2014-02-R2/host``` rather than ```ubertooth-2014-02-R2```

ehumphri, 2015-06-26:
Yikes, I totally missed that. Thanks sir!

closed 2015-06-26 by dominicgs