greatscottgadgets/ubertooth #84: cb_br_rx() in libubertooth calculates wrong CLKN value

elli89, 2015-06-30:
CLKN is calculated from clk100ns of the oldest received packet. At the bluetooth rx rate of 1 Mbit per second, each bit has a duration of 1 µs or 10 units of 100 ns. If the bluetooth packet starts with an offset in the bitstream, there is also an offset in the corresponding clk100ns value. This clock offset is currently ignored, which sometimes leads to an incorrect CLKN value for the bluetooth packet.

This issue often causes libbtbb to fail to detect the correct hopping sequences. So when running ubertooth-rx, the CLK27-1 detection procedure needs many trys and sometimes even detects the wrong clock values.

This issue applies to the current git master of ubertooth.

dominicgs, 2015-06-30:
Doesn't the addition of the offset in line 501 (https://github.com/greatscottgadgets/ubertooth/blob/master/host/libubertooth/src/ubertooth.c#L501) correct for that?

elli89, 2015-06-30:
As I understand, offset ist the bit offset of the bluetooth packet in the usb rx stream. Since each bit has a duration of 1 µs or 10 units of 100 ns, it looks for me like (offset*10) should be added in order to translate the bit units into time units. Is that correct?

By the way, which clk100ns ist stored in the usb packet? The timestamp of the first bit that the ubertooth MCU receives from the CC2400? Or the timestamp of the last bit?

elli89, 2015-06-30:
I have pushed some changes that make ubertooth-rx detekt CLK27-1 much faster and most of the time only one try is needed. What do you think of it?

I assumed that clk100ns of the usb packet ist the timestamp of the first bit in the stream.

dominicgs, 2015-07-07:
PR #85 fixed this issue

closed 2015-07-07 by dominicgs