greatscottgadgets/ubertooth #86: Ubertooth instructions

Daniel-Katz, 2015-06-30:
Are there any instructions of what commands I can use via ubertooth/kismet and how to effectively use them?

dominicgs, 2015-07-01:
Do you mean a list of the Ubertooth tools and what they do?

There is an introduction to Ubertooth here: https://github.com/greatscottgadgets/ubertooth/wiki/Getting-Started, although it needs updating with the ubertooth-follow and ubertooth-scan tools.

Daniel-Katz, 2015-07-01:
all I really understood from the introduction is that I can use ubertooth-rx to get the LAP of a device, nd 
'ubertooth-rx -l [lap value] to follow that device specifically, but I already knew how to do that with 'ubertooth-scan and 'ubertooth-follow and that gives me more information that way. Are there any other tools that ubertooth has, aside from rx, scan, follow and specan?

dominicgs, 2015-07-01:
The only other tool is ubertooth-btle, which is used to capture traffic from Bluetooth Low Energy devices (a different protocol and radio specification than basic rate Bluetooth).

There is some information on using it with Wireshark here: https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark  but there is not a lot of documentation available for it yet as it is a work in progress.

Daniel-Katz, 2015-07-02:
Thanks. Can follow or scan be used with wireshark?

dominicgs, 2015-07-02:
Not yet, but that is one of the goals for the next release, which I hope will be in August.

Daniel-Katz, 2015-07-02:
Cool. What information can I expect to possibly get from any of the scans (scan/follow/rx/btle)? I mean, I was able to follow my iPhone specifically and I was able to tell when there was activity between my bluetooth headset an my phone, but can I do anything with the data?

closed 2015-07-08 by Daniel-Katz