greatscottgadgets/ubertooth #89: ubertooth-btle not changing channel

TorstenRobitzki, 2015-07-07:
Hello,
I have a Uberthooth one with firmware release 2014-04-R1. I want to use ubertooth to verify a advertising BLE device. If my device advertises on all three advertising channels, I see advertising packages on channel 37. When I start ubertooth-btle and change to channel 38 (`ubertooth-btle -fA38`), I still see just packages on channel 37. If I change my BLE device to just advertise on channel 38, I see no advertising packages.

I've checked, that the function `cmd_set_channel()` gets called with 2426 and that `libusb_control_transfer()` returns no error. To me it seems like the firmware never realize that it should switch to a different channel, or that this channel information is ignore.

Any idea what could be wrong here? I have a JTag adapter here and could debug the firmware. Is the JTAG connector a "Cortex Debug Connector"? Any advice on where to get something to adapt this to a JLinkPro?

cheers,
Torsten

mikeryan, 2015-07-07:
Using the latest released firmware I can confirm that running the following command will cause the device to change to the specified channel:

    ubertooth-btle -f -A 38

TorstenRobitzki, 2015-07-07:
Hi Mike,
thank you for having a look at this. By latest released firmware, do you mean 2014-04-R1? I observe that if I'm using uberthooth-btle with `-A x`, I can see a changed channel when using `uberthooth-util -c`:

    $ ubertooth-tools/src/ubertooth-btle -f -A 38
    ^C
    $ ubertooth-tools/src/ubertooth-util -c
    Current frequency: 2426 MHz (Bluetooth channel 24)
    $ ubertooth-tools/src/ubertooth-btle -f -A 37
    ^C
    $ ubertooth-tools/src/ubertooth-util -c
    Current frequency: 2402 MHz (Bluetooth channel 0)

but when I scan for advertising packets on channel 38:

    $ ubertooth-tools/src/ubertooth-btle -f -A 38
    Channel: 2426nr: 0
    systime=1436271300 freq=2402 addr=8e89bed6 delta_t=17250.713 ms
    40 16 4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65 bd 88 5c 
    Advertising / AA 8e89bed6 (valid)/ 22 bytes
        Channel Index: 37
        Type:  ADV_IND
        AdvA:  c0:0f:e5:e7:0f:4e (random)
        AdvData: 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
            Type 01 (Flags)
               00000110
            Type 09 (Complete Local Name)
               Temperature

        Data:  4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
        CRC:   bd 88 5c

I get packets on channel 37.

TorstenRobitzki, 2015-07-13:
The reason why we see different behaviour, is because the device does not switch the channel when it is already in MODE_BT_FOLLOW_LE mode:

		if (mode != MODE_BT_FOLLOW_LE) {
			channel = requested_channel;
			requested_channel = 0;

			/* CS threshold is mode-dependent. Update it after
			 * possible mode change. TODO - kludgy. */
			cs_threshold_calc_and_set();
		}

from line bluetooth_rxtx.c:540

Maybe someone can remember, why changing channel is ignored, when in MODE_BT_FOLLOW_LE mode?

cheers,
Torsten

dominicgs, 2015-07-13:
Isn't that check there because once we're in MODE_BT_FOLLOW_LE mode, we
want to have the firmware control the channel rather than to host?

I haven't used BLE enough to know if there's a bug or not, but I think
that's the reasoning behind that code.

TorstenRobitzki, 2015-07-13:
But this way, if you first try to follow a connection on channel 37, stop ubertooth-btle (with ctrl-c) and than got to search for advertisings on channel 38, you end up with still sniffing on channel 37. This is easy to workaround, by reseting the firmware, but it seems somehow wrong to me ;-)

dominicgs, 2015-07-13:
Perhaps the right answer is to set MODE_IDLE at the end (or start) of
ubertooth-btle?

dominicgs, 2015-07-13:
Or maybe we should worry less about setting the channel from the host.

TorstenRobitzki, 2015-07-13:
Even if the ubertooth is currently following a connection request and thus changing the channel, I could not see changing the channel shouldn't be allowed if the host is requesting it. Maybe @mikeryan could add some details on this?

mikeryan, 2015-08-13:
It turns out this was fixed in fb20144629284637d1c035b1741e103a4b9dc0f0, which occurred just after release. I'm going to go ahead and add some belt-and-suspenders that sets the mode to MODE_IDLE when ubertooth-btle quits.

mikeryan, 2015-08-13:
@TorstenRobitzki this will be available in the next release. In the meantime you can try running code from master, which is actually quite stable.

closed 2015-08-13 by mikeryan

TorstenRobitzki, 2015-08-13:
@mikeryan I can confirm that the current trunk fixed the issue. Thank you very much!

commit ca2a1667 (2015-08-13): le: reset to MODE_IDLE when exiting connection following

fixes #89