greatscottgadgets/ubertooth #90: Using the data

Daniel-Katz, 2015-07-08:
I'm doing a research project on the insecurities of Bluetooth, and so far I've managed to successfully use ubertooth-scan and ubertooth-follow to get information on specific devices. My question is, with the resulting data that can be seen using ubertooth-follow when the device is active, do you know if I can decipher anything from it? 
Assuming that I know what kind of Bluetooth device is being used, I can make certain assumptions if I see that the device is actively communicating (for example, if it is connected to wireless headphones and active, I can safely assume that the person is nearby, and is either on a phone call, or listening to music), but do you know if/how I can be more specific?

dominicgs, 2015-07-08:
@Daniel-Katz This question would be better directed to the Ubertooth mailing list (https://lists.sourceforge.net/lists/listinfo/ubertooth-general) as there are a large number of people able to respond to posts there but not many who monitor the GitHub issue tracker.  I would also prefer to keep the issue tracker for bugs rather than questions on usage of the tools.

Daniel-Katz, 2015-07-08:
Okay, will do.

closed 2015-07-08 by Daniel-Katz