greatscottgadgets/ubertooth #91: I have to reset the ubertooth one after doing ctrl+c while running ubertooth-btle

scottdarch, 2015-08-04:
I just bought my ubertooth 1 and  flashed it to Firmware revision: 2014-04-R1. 

When I run ubertooth-btle -p and then stop it using ctrl+c ubertooth-btle won't work again until I reseat the ubertooth device. The second time I run ubertooth-btle -p it just logs nothing.

scottdarch, 2015-08-05:
I saw this problem on osx and also on Ubuntu 12 running on the same machine within a virt.

For osx I can get ubertooth-btle to run 1more time by doing ubertooth-util -r (reset) but the second time I try to do a reset I get:

```
usb_claim_interface error -3
```

scottdarch, 2015-08-05:
I'm running libusb-1.0.19 if that helps.

dominicgs, 2015-08-05:
The main issue here has been fixed in git but has not been released yet.

The ```usb_claim_interface error -3``` issue is surprising to me.  I thought that ```ubertooth-util -r``` should work in most cases.  Could you try to do it again but first enable libusb debugging in your environment?  ```export LIBUSB_DEBUG=4```

scottdarch, 2015-08-06:
[ 3.333342] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.333344] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[ 3.333346] [00000513] libusb: debug [op_handle_events] checking fd 8 with revents = 1
[ 3.333349] [00000513] libusb: debug [darwin_handle_callback] handling control completion with kernel status 0
[ 3.333351] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcbfa40c768 has callback 0x10b4d059c
[ 3.333354] [00000513] libusb: debug [sync_transfer_cb] actual_length=1
[ 3.334041] [00000513] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.334048] [00000513] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.334052] [00000513] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.334173] [00000a0f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 3.334187] [00000513] libusb: debug [handle_events] poll() returned 1
[ 3.334190] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.334192] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[ 3.334194] [00000513] libusb: debug [op_handle_events] checking fd 8 with revents = 1
[ 3.334197] [00000513] libusb: debug [darwin_handle_callback] handling control completion with kernel status 0
[ 3.334200] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcbfa40c768 has callback 0x10b4d059c
[ 3.334202] [00000513] libusb: debug [sync_transfer_cb] actual_length=1
[ 3.334886] [00000513] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.334893] [00000513] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.334896] [00000513] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.334976] [00000a0f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 3.334990] [00000513] libusb: debug [handle_events] poll() returned 1
[ 3.334993] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.334995] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[ 3.334997] [00000513] libusb: debug [op_handle_events] checking fd 8 with revents = 1
[ 3.335000] [00000513] libusb: debug [darwin_handle_callback] handling control completion with kernel status 0
[ 3.335003] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcbfa40c768 has callback 0x10b4d059c
[ 3.335006] [00000513] libusb: debug [sync_transfer_cb] actual_length=1
[ 3.335688] [00000513] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.335695] [00000513] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.335697] [00000513] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.335853] [00000a0f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 3.335865] [00000513] libusb: debug [handle_events] poll() returned 1
[ 3.335868] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.335870] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[ 3.335872] [00000513] libusb: debug [op_handle_events] checking fd 8 with revents = 1
[ 3.335875] [00000513] libusb: debug [darwin_handle_callback] handling control completion with kernel status 0
[ 3.335878] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcbfa40c768 has callback 0x10b4d059c
[ 3.335880] [00000513] libusb: debug [sync_transfer_cb] actual_length=1
[ 3.336564] [00000513] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.336570] [00000513] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.336573] [00000513] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.336693] [00000a0f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 3.336706] [00000513] libusb: debug [handle_events] poll() returned 1
[ 3.336709] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.336711] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[ 3.336713] [00000513] libusb: debug [op_handle_events] checking fd 8 with revents = 1
[ 3.336716] [00000513] libusb: debug [darwin_handle_callback] handling control completion with kernel status 0
[ 3.336719] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fcbfa40c768 has callback 0x10b4d059c
[ 3.336721] [00000513] libusb: debug [sync_transfer_cb] actual_length=1
[ 3.337401] [00000513] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 3.337407] [00000513] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.337410] [00000513] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.337524] [00000a0f] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 3.337535] [00000513] libusb: debug [handle_events] poll() returned 1
[ 3.337538] [00000513] libusb: debug [op_handle_events] checking fd 4 with revents = 0
[ 3.337541] [00000513] libusb: debug [op_handle_events] checking fd 6 with revents = 0
[... cut ...]
[ 2.093252] [00000513] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fed3bd00268 has callback 0x10cb0359c
[ 2.093257] [00000513] libusb: debug [sync_transfer_cb] actual_length=0
[ 2.093267] [00000513] libusb: debug [libusb_release_interface] interface 0
[ 2.093273] [00000513] libusb: debug [libusb_close]
[ 2.093279] [00000513] libusb: debug [usbi_remove_pollfd] remove fd 10
[ 2.093395] [00000513] libusb: debug [libusb_exit]
[ 2.093410] [00000513] libusb: debug [libusb_exit] not destroying default context

scottdarch, 2015-08-07:
So I'm trying to build the firmware to pickup the fix but I'm missing dependencies in my ubuntu 12 environment:

```
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: cannot find -lstdc++
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: cannot find -lsupc++
```

I installed the arm toolchain and all the prerequisites for building the ubertooth tools. What other dependencies are there for building the firmware?

dominicgs, 2015-08-11:
You need to install the arm libstdc++ version.  On my Ubuntu 15.04 system the package is called ```libstdc++-arm-none-eabi-newlib```, although I'm not sure if that package is available for 12.04.

scottdarch, 2015-08-11:
Sorry, I'm on Ubuntu 14.04. That said; there appears to be a bug that prevents installation of libstdc++-arm-none-eabi-newlib on 14.04. When will next firmware release be? I'm having to decide to wait to use my ubertooth or build an Ubuntu 15 image so I can build the firmware.

dominicgs, 2015-08-11:
The next release will likely be next month.  If you are in need of the latest firmware, there is a test version here: https://www.dominicspill.com/ubertooth/bluetooth_rxtx.bin.dfu

scottdarch, 2015-08-12:
Ah, thanks for the dfu but I was able to update my Ubuntu to get the libstdc++-arm-none-eabi-newlib fix. I built and verified that this issues is fixed in the trunk. I also opened a PR to update the README with the libstc++ dependency. Thanks!

commit 825bc890 (2015-08-25): Remove requirement for libstdc++ for non-C++ firmware code
Fixes #91 in an alternative way
commit e05a8041 (2015-10-12): Remove requirement for libstdc++ for non-C++ firmware code
Fixes #91 in an alternative way
closed 2015-08-12 by scottdarch