greatscottgadgets/ubertooth #92: just can see 14 bytes payload

lewisha, 2015-08-10:
hi,i have met one problem. The wireshark just show 14bytes packets,no payload. i just installed  the libbtbb,ubertooth,kismet and wireshark follow the description in https://github.com/greatscottgadgets/ubertooth/wiki. The ubuntu system is 14.0.2

mikeryan, 2015-08-13:
Can you provide a screenshot?

mcheah-, 2015-08-13:
I’ve got the same problem – it’s only ever showing 14 byte packets – screenshot of the packets in wireshark here:

http://creative.coventry.ac.uk/~madeline/ubertooth_scr.png

Using Ubuntu 14.04 native.

dominicgs, 2015-08-13:
This looks to be the same as / related to libbtbb issue #21.

The ubertooth is only capturing access codes because it doesn't have a UAP or CLK.

lewisha, 2015-08-13:
how to resolve it？

mcheah-, 2015-08-17:
Hi,

I’ve just re-run Kismet/ubertooth whilst sniffing traffic, and it appears that the UAP and CLK were successfully detected (see example screenshot). However I’m still getting the same problem when trying to view in Wireshark – all packets are only 14 bytes in size.

SCREENSHOT: http://creative.coventry.ac.uk/~madeline/screenshot_kismet.png

I’ve tried it with Ubertooth firmware versions 2014-02-R2 and 2014-04-R1, with two different physical Ubertooth devices (one at a time).  Kismet version is kismet-2013-03-R1b, and I’ve tried it on Kali Linux 1.0 and 2.0 and Ubuntu 14.04 LTS (both as VMs in VMWare and as native installs) and it all ends up with the same thing.  All the other ubertooth-tools works as it should.

Is there anything else I can do? Or is there something I’m still not doing correctly?

Thanks.

mp3weenie, 2015-11-24:
I am having the same issue. 

Ubuntu 14.04 LTS 
Firmware revision: git-8b7ee77 
libbtbb-2015-10-R1 
ubertooth-2015-10-R1 
kismet-2013-03-R1b 
Wireshark Version 1.10.6 (v1.10.6 from master-1.10)

ohyeahq, 2016-04-28:
Exactly the same here.  

Antergos latest rolling
others latest stable

Any clue, quick hack, or workaround please?

And one newbie Q: I also tried to feed a direct output (pcap or pcapng) from
ubertooth-rx -l [LAP] -q foo.pcap (or -r foo.pcapng)
into wireshark, but it did not understand BT stuff in it.  Why is kismet (i.e. pcapbtbb) 
mandatory?

I'm asking this because kismet-ubertooth double free [issue #59] is also constantly 
killing me, so I want to avoid kismet if possible.  

Cheeeeers

ohyeahq, 2016-05-17:
Hi, the issue is still bothering me.  I googled "ubertooth 14 bytes" to find out that pretty many people are suffering too.  I'll leave here what I did (lengthy), so please let me know if I've done it wrong.  

BTW, my main sniffing target is BT audio (not LE) between a smartphone and a wireless headphone; I try to figure out the codec (SBC or aptX) and the bitrate.  BT stacks are so user-UNFRIENDLY for this kind of matter.  

Antergos latest rolling; others latest stable; Dell Latitude D600 (kinda old)

cf. yaourt -S ubertooth  -> incomplete, no kismet support, avoided ; go DIY
[get python modules & wireshark ; arm stuff]
yaourt -S python-numpy python-pyside wireshark-qt arm-none-eabi-gcc arm-none-eabi-newlib

[set library path /usr/local/lib (silly)]
su -
echo "/usr/local/lib" >/etc/ld.so.conf.d/usr_local_lib_why_not_here.conf
exit
sudo ldconfig

[check usb configration ; know udev rule]
sudo lsusb
-> Bus 003 Device 002: ID 1d50:6002 OpenMoko, Inc. 
sudo lsusb -d 1d50:6002 -v
sudo udevadm info -a -p $(udevadm info -q path -n /dev/bus/usb/003/002)
-> SUBSYSTEM=="usb", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6002" etc

[permission stuff]
sudo groupadd ubertooth
sudo gpasswd -a MYNAME ubertooth
sudo groupadd kismet
sudo gpasswd -a MYNAME kismet
sudo groupadd wireshark
sudo gpasswd -a MYNAME wireshark
su -
echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="1d50", ATTR{idProduct}=="6002", GROUP="ubertooth", MODE="0660"' > /etc/udev/rules.d/99-ubertooth.rules
exit
sudo udevadm control --reload-rules

[make ubertooth etc]
cd ~
mkdir github
cd github
git clone https://github.com/greatscottgadgets/ubertooth
git clone https://github.com/greatscottgadgets/libbtbb

cd ~/github/libbtbb/
mkdir build
cd build
cmake ..
make
sudo make install

cd ~/github/ubertooth/host/
mkdir build
cd build
cmake ..
make
sudo make install

cd ~/github/ubertooth/firmware/
make clean all
make
cd ./bluetooth_rxtx/
sudo ubertooth-dfu -r -d bluetooth_rxtx.dfu

ubertooth-util -V
-> firmware updated, OK

[make wireshark plugins]
cd ~/github/libbtbb/wireshark/plugins/btbb/
mkdir build
cd build
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/wireshark/plugins/2.0.3/ ..
make
sudo make install

cd ~/github/libbtbb/wireshark/plugins/btbredr/
mkdir build
cd build
cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/wireshark/plugins/2.0.3/ ..
make
sudo make install

[make kismet from tarball 2016; tar xvf ...] 
cd ~/ubertooth/kismet-2016-01-R1/
ln -s ~/github/ubertooth/host/kismet/plugin-ubertooth ~/ubertooth/kismet-2016-01-R1
./configure
make
make plugins
sudo make suidinstall
sudo make plugins-install

cd ~/github/ubertooth/host/kismet/plugin-ubertooth/
KIS_SRC_DIR=~/ubertooth/kismet-2016-01-R1 make
sudo KIS_SRC_DIR=~/ubertooth/kismet-2016-01-R1 make install

sudo nano /usr/local/etc/kismet.conf
logtypes=pcapdump,gpsxml,netxml,nettxt,alert
-> logtypes=pcapdump,gpsxml,netxml,nettxt,alert,pcapbtbb

nano ~/.kismet/kismet_ui.conf
plugindir=/usr/lib/kismet_client/
-> plugindir=/usr/local/lib/kismet_client/

[kismet ; start server ; add source interface=ubertooth ; 
hit '~'; Kismet - Plugins - Select Plugin - ubertooth_ui.so ; click space to activate; tab; click close
sniff BT audio (not LE) for a while and quit]

[open the output file (Kismet...pcapbtbb) with wireshark]
-> all 14 bytes, too bad...

Cheeeeeeers

dominicgs, 2016-05-17:
This is because the Ubertooth is only identifying LAPs and finding packets, it isn't able to decode and interpret packets without more information.

This isn'tt a bug, it's related to the way that Bluetooth works.  We cannot decode the packet header without the UAP and therefore cannot tell how long the packet is going to be.  We are only about to decode the first 14bytes until we have the UAP.

ohyeahq, 2016-05-17:
dominicgs: Thanks for a lightening-fast reply!  It cleared out my sight.  Two more quick questions:

(1) I got LAP (and sometimes UAP) by staring at ubertooth-rx output.  Is there a way to specify the LAP/UAP info inside the kismet/ubertooth combo, so that I can go beyond 14 bytes?  

(2) I got pcap (or pcapng or dump) file from

ubertooth-rx -l LAP -q output.pcap

etc, so I tried to feed it to wireshark, but wireshark didn't understand a byte of these payloads.  Theoretically these bytes should contain BT protocols (AVDTP, A2DP etc), so I hoped, at least, I could get the codec/bitrate from them.  But I cannot hand-decode them either.  Any clue?  

Cheeeeeeeers

ohyeahq, 2016-05-18:
I reviewed logs and noticed that my problem is also the same as the issues (#36 and #38).  

The comments there are illuminating, but the core issue is still unresolved: even after UAP is found (as is seen in kismet/ubertooth combo's server window), all the entries in pcapbtbb file are still 14 bytes of zero.  

I tried in a wireless-clean room with just a smartphone/wireless headphone pair for hours.  The kismet BT window clearly showed counts of 1,000 - 10,000 packets just between the two (including the initial connecting phase).  Got "correct CRC" and "UAP" messages.  The resulting pcapbtbb file is up to 400kB or so.  All 14 bytes.  

Cheeeeeeeers

dominicgs, 2016-05-18:
1) There's no way to specify a LAP/UAP combination to Kismet.  This may be an option in the future, but the core Kismet codebase is under significant development and I'd prefer to let that stabilise before putting more development time in to the plugin.

Having said that, Kismet is a survey tool that is designed to enumerate devices in the surrounding area rather than target a specific device, so I think you may be better off using `ubertooth-rx` with the LAP/UAP specified in order to capture more traffic.

The other big change that is coming is that the pcap format has changed and we will no longer be using pcapbtbb, but rather bt_bredr.

2) The packets captured with ubertooth-rx will use the newer pcap format, which will require you to build and install the plugins for Wireshark that come with libbtbb.

The problem you will find is that the connection is likely to be encrypted and without sniffing the pairing there is no chance of breaking it.  It is also very unlikely that you're picking up all of the packets in the connection.  Ubertooth attempts to follow the connection once it has calculated the clock value, but features such as adaptive frequency hopping and multi-slot packets make this unreliable.

Yes, issues #36 and #38 (libbtbb issues, not Ubertooth) are the same bug as this logged against libbtbb rather than Ubertooth.

ohyeahq, 2016-05-20:
Thanks again.  Now I can see the situation much better.  I commented on libbtbb #38 about how I got more than 14 bytes for now.  

So, if the payload is encrypted, there is no (easy) way to break it, even if it is a weak one, i.e., BT classic's PIN, "Mode 4 (SSP) Level 0 or 2", or "Just works" thing?  

Cheeeeeeeeers

closed 2020-12-03 by mikeryan