greatscottgadgets/ubertooth #96: hard to reproduce, wrong delta_t

TorstenRobitzki, 2015-08-13:
Occasionally, ubertooth-btle shows wrong times:

    systime=1439467854 freq=2402 addr=8e89bed6 delta_t=1009.997 ms
    40 16 4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65 bd 88 5c 
    Advertising / AA 8e89bed6 (valid)/ 22 bytes
        Channel Index: 37
        Type:  ADV_IND
        AdvA:  c0:0f:e5:e7:0f:4e (random)
        AdvData: 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
            Type 01 (Flags)
               00000110
            Type 09 (Complete Local Name)
               Temperature

        Data:  4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
        CRC:   bd 88 5c

    systime=1439467855 freq=2402 addr=8e89bed6 delta_t=102822.727 ms
    40 16 4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65 bd 88 5c 
    Advertising / AA 8e89bed6 (valid)/ 22 bytes
        Channel Index: 37
        Type:  ADV_IND
        AdvA:  c0:0f:e5:e7:0f:4e (random)
        AdvData: 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
            Type 01 (Flags)
               00000110
            Type 09 (Complete Local Name)
               Temperature

        Data:  4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
        CRC:   bd 88 5c

    systime=1439467856 freq=2402 addr=8e89bed6 delta_t=1001.997 ms
    40 16 4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65 bd 88 5c 
    Advertising / AA 8e89bed6 (valid)/ 22 bytes
        Channel Index: 37
        Type:  ADV_IND
        AdvA:  c0:0f:e5:e7:0f:4e (random)
        AdvData: 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
            Type 01 (Flags)
               00000110
            Type 09 (Complete Local Name)
               Temperature

        Data:  4e 0f e7 e5 0f c0 02 01 06 0c 09 54 65 6d 70 65 72 61 74 75 72 65
        CRC:   bd 88 5c

The second delta_t should be around ~1000ms too. I've watched on the screen while those packets where received and there where just one second between the printouts. I've double checks this and piped the output of ubertooth-btle through grep and found such an extrem value a second time:

    systime=1439468492 freq=2402 addr=8e89bed6 delta_t=999.997 ms
    systime=1439468493 freq=2402 addr=8e89bed6 delta_t=1006.997 ms
    systime=1439468494 freq=2402 addr=8e89bed6 delta_t=1003.030 ms
    systime=1439468495 freq=2402 addr=8e89bed6 delta_t=1009.965 ms
    systime=1439468496 freq=2402 addr=8e89bed6 delta_t=1005.997 ms
    systime=1439468497 freq=2402 addr=8e89bed6 delta_t=1001.997 ms
    systime=1439468500 freq=2402 addr=8e89bed6 delta_t=3014.992 ms
    systime=1439468501 freq=2402 addr=8e89bed6 delta_t=1007.997 ms
    systime=1439468502 freq=2402 addr=8e89bed6 delta_t=1003.997 ms
    systime=1439468503 freq=2402 addr=8e89bed6 delta_t=999.997 ms
    systime=1439468505 freq=2402 addr=8e89bed6 delta_t=2009.995 ms
    systime=1439468506 freq=2402 addr=8e89bed6 delta_t=1009.997 ms
    systime=1439468507 freq=2402 addr=8e89bed6 delta_t=1005.997 ms
    systime=1439468508 freq=2402 addr=8e89bed6 delta_t=1001.997 ms
    systime=1439468509 freq=2402 addr=8e89bed6 delta_t=1008.997 ms
    systime=1439468510 freq=2402 addr=8e89bed6 delta_t=102821.727 ms <---
    systime=1439468511 freq=2402 addr=8e89bed6 delta_t=1000.997 ms
    systime=1439468512 freq=2402 addr=8e89bed6 delta_t=1007.997 ms
    systime=1439468513 freq=2402 addr=8e89bed6 delta_t=1003.997 ms
    systime=1439468514 freq=2402 addr=8e89bed6 delta_t=999.997 ms
    systime=1439468515 freq=2402 addr=8e89bed6 delta_t=1006.997 ms
    systime=1439468516 freq=2402 addr=8e89bed6 delta_t=1002.997 ms
    systime=1439468517 freq=2402 addr=8e89bed6 delta_t=1009.997 ms
    systime=1439468518 freq=2402 addr=8e89bed6 delta_t=1005.997 ms
    systime=1439468519 freq=2402 addr=8e89bed6 delta_t=1001.997 ms
    systime=1439468520 freq=2402 addr=8e89bed6 delta_t=1008.999 ms

TorstenRobitzki, 2015-08-13:
And here a third example:

    systime=1439469489 freq=2402 addr=8e89bed6 delta_t=1002.997 ms
    systime=1439469490 freq=2402 addr=8e89bed6 delta_t=1010.011 ms
    systime=1439469491 freq=2402 addr=8e89bed6 delta_t=1005.984 ms
    systime=1439469492 freq=2402 addr=8e89bed6 delta_t=1001.997 ms
    systime=1439469493 freq=2402 addr=8e89bed6 delta_t=102825.727 ms
    systime=1439469494 freq=2402 addr=8e89bed6 delta_t=1004.997 ms
    systime=1439469495 freq=2402 addr=8e89bed6 delta_t=1000.997 ms
    systime=1439469496 freq=2402 addr=8e89bed6 delta_t=1007.997 ms
    systime=1439469497 freq=2402 addr=8e89bed6 delta_t=1004.008 ms
    systime=1439469499 freq=2402 addr=8e89bed6 delta_t=2006.993 ms
    systime=1439469500 freq=2402 addr=8e89bed6 delta_t=1002.988 ms
    systime=1439469502 freq=2402 addr=8e89bed6 delta_t=2015.994 ms
    systime=1439469503 freq=2402 addr=8e89bed6 delta_t=1002.055 ms
    systime=1439469504 freq=2402 addr=8e89bed6 delta_t=1008.939 ms

This time I was watching again on the screen and can confirm, that the printout came within 1 second to the previous one. I think there is a pattern in this three wrong values: 

    102822.727 ms
    102821.727 ms
    102825.727 ms

TorstenRobitzki, 2015-08-14:
While I was testing, I found an other interesting example ;-) :

    systime=1439563134 freq=2424 addr=506545a6 delta_t=3.310 ms
    01 00 bb 9e aa 
    Data / AA 506545a6 (valid) /  0 bytes
        Channel Index: 10
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 0  MD: 0

        Data: 
        CRC:   bb 9e aa

    systime=1439563134 freq=2424 addr=506545a6 delta_t=429496.649 ms
    05 00 68 98 aa 
    Data / AA 506545a6 (valid) /  0 bytes
        Channel Index: 10
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 0  MD: 0

        Data: 
        CRC:   68 98 aa

mikeryan, 2015-08-14:
This is almost certainly due to the internal clock rolling over at 32 bits. I'll set this as a milestone to be fixed in next month's release. Thanks for reporting!

TorstenRobitzki, 2015-08-14:
Let me know if I can help with testing or further analysis.

mikeryan, 2015-08-15:
Distressing behavior detected:
```
systime=1439658213 freq=2402 addr=8e89bed6 delta_t=100.631 ms
clk100ns=3276763582 prev_ts=3275757275
00 09 06 45 d1 f7 b1 34 02 01 05 6c ab af
<snip>

systime=1439658213 freq=2402 addr=8e89bed6 delta_t=101926.100 ms
clk100ns=1057289 prev_ts=3276763582
00 09 06 45 d1 f7 b1 34 02 01 05 6c ab af
<snip>
```

clk100ns is wrapping somewhere around 3276763582, or ```0xC3F71BE```. This is pretty foul, as a lot of the code makes assumptions about this value wrapping at 2^32. This will require a much deeper fix than anticipated.

mikeryan, 2015-08-15:
It turns out this is documented:

```
/*
 * CLK100NS is a free-running clock with a period of 100 ns.  It resets every
 * 2^15 * 10^5 cycles (about 5.5 minutes) - computed from clkn and timer0 (T0TC)
 *
 * clkn is the native (local) clock as defined in the Bluetooth specification.
 * It advances 3200 times per second.  Two clkn periods make a Bluetooth time
 * slot.
 */
```

This particular issue should be fixable, but the deeper problem remains.

commit 89d56cd9 (2015-08-16): le: measure delta_t correctly when clk100ns rolls over

Fixes #96
closed 2015-08-16 by mikeryan