greatscottgadgets/ubertooth #97: Wireshark issues with master ubertooth and master libbtbb

jbmontgomery4, 2015-08-13:
I am having wireshark issues. I pulled and built from the latest ubertooth and libbtbb from github (ubertooth_master and libbtbb_master as of 8/7). Everything built. Even built the firmware and downloaded it.  Kismet seems to work, it says it is receiving packets and the file is growing. Whenever I try and load the file to wireshark it only shows one packet that is malformed and stops. btbb shows up green in the filter window.  

Also for ubertooth-btle capture the web page mentions a -c option. That option no longer seems to be supported. What is needed to be done to capture btle?

Thanks,
John Montgomery

mikeryan, 2015-08-13:
If you're not seeing -c it's likely you didn't have libpcap-dev when you built libbtbb and Ubertooth. Can you check if that's installed?

jbmontgomery4, 2015-08-14:
Yes, it is installed:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
libpcap-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

jbmontgomery4, 2015-08-14:
Basically according to https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark the command used to send data to wireshark was suppose to be:

ubertooth-btle -f -c /tmp/pipe

However, -c is no longer a valid option for ubertooth-btle:

ubertooth-btle - passive Bluetooth Low Energy monitoring
Usage:
	-h this help

    Major modes:
	-f follow connections
	-p promiscuous: sniff active connections
	-a[address] get/set access address (example: -a8e89bed6)
	-s<address> faux slave mode, using MAC addr (example: -s22:44:66:88:aa:cc)
	-t<address> set connection following target (example: -t22:44:66:88:aa:cc)

    Data source:
	-i<filename> read packets from file
	-U<0-7> set ubertooth device to use

    Misc:
	-r<filename> capture packets to PCAPNG file
	-d<filename> dump packets to binary file
	-A<index> advertising channel index (default 37)
	-v[01] verify CRC mode, get status or enable/disable
	-x<n> allow n access address offenses (default 32)

If an input file is not specified, an Ubertooth device is used for live capture.
In get/set mode no capture occurs.

So, I understand the page was last updated on 4/2014. Just wondering what is the way to capture packets at this time. Is it the -r option?  

Seems to work, so that is solved.  

The last remaining question is that I do receive packets using kismet.  However, each packet is 14 bytes and are designated as malformed in wireshark. Why is that? 

I have included screen shots.

[image]
[image]
[image]

Thank you.

mikeryan, 2015-08-14:
The Wireshark issue is expected behavior. Until you've recovered the LAP and NAP it is not possible to recover packet data.

closed 2015-08-14 by jbmontgomery4