greatscottgadgets/ubertooth #100: Missing packages

TorstenRobitzki, 2015-08-14:
Hi,
I see sometimes situations where I miss a package. First I thought it could be a problem with my BLE stack, but I quite confident that this is an ubertooth-btle issue. Here is an example, of three consecutive BLE connection events. It looks like the slaves response is missing in the second event (on channel 7), but as the master is responding with its sequence number and its next expected sequence number being incremented, I think the master must have received the missing package:

    systime=1439569783 freq=2474 addr=506545de delta_t=12.268 ms
    01 00 41 84 2e 
    Data / AA 506545de (valid) /  0 bytes
        Channel Index: 34
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 0  MD: 0

        Data: 
        CRC:   41 84 2e

    systime=1439569783 freq=2474 addr=506545de delta_t=0.232 ms
    05 00 92 82 2e 
    Data / AA 506545de (valid) /  0 bytes
        Channel Index: 34
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 0  MD: 0

        Data: 
        CRC:   92 82 2e

    systime=1439569783 freq=2418 addr=506545de delta_t=12.327 ms
    0d 00 34 8f 2e 
    Data / AA 506545de (valid) /  0 bytes
        Channel Index: 7
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 1  MD: 0

        Data: 
        CRC:   34 8f 2e

    systime=1439569783 freq=2440 addr=506545de delta_t=12.442 ms
    01 00 41 84 2e 
    Data / AA 506545de (valid) /  0 bytes
        Channel Index: 17
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 0  MD: 0

        Data: 
        CRC:   41 84 2e

    systime=1439569783 freq=2440 addr=506545de delta_t=0.096 ms
    05 00 92 82 2e 
    Data / AA 506545de (valid) /  0 bytes
        Channel Index: 17
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 0  MD: 0

        Data: 
        CRC:   92 82 2e

cheers,

Torsten

mikeryan, 2015-08-18:
Hello @TorstenRobitzki, some packets will be discarded by default due to CRC failure. To capture these packets, use:

```
ubertooth-btle -v0
ubertooth-btle -f
```

In addition, it's possible some packets are simply being missed by Ubertooth.

TorstenRobitzki, 2015-08-18:
Hi @mikeryan,
I don't think that this are CRC failures because otherwise, the master would not acknowledge the not captured package. Is this "it's possible some packets are simply being missed by Ubertooth" a limitation we simply have to live with? Could it be something that is related to USB on OS/X?

TorstenRobitzki, 2015-08-18:
here is an other example: 

    $ ubertooth-btle -f -v 0 | grep CONNECT_REQ -B 5 -A 80
    
    systime=1439902358 freq=2402 addr=8e89bed6 delta_t=57.531 ms
    85 22 e4 fc 94 9d 96 5c 0b 60 4f d5 a6 f0 ae ab 9a af 13 19 0d 03 05 00 0c 00 00 00 c8 00 ff ff ff ff 1f ac a7 70 0d 
    Advertising / AA 8e89bed6 (valid)/ 34 bytes
        Channel Index: 37
        Type:  CONNECT_REQ
        InitA: 5c:96:9d:94:fc:e4 (public)
        AdvA:  f0:a6:d5:4f:60:0b (random)
        AA:    af9aabae
        CRCInit: 0d1913
        WinSize: 03 (3)
        WinOffset: 0005 (5)
        Interval: 000c (12)
        Latency: 0000 (0)
        Timeout: 00c8 (200)
        ChM: ff ff ff ff 1f
        Hop: 12
        SCA: 5, 31 ppm to 50 ppm

        Data:  e4 fc 94 9d 96 5c 0b 60 4f d5 a6 f0 ae ab 9a af 13 19 0d 03 05 00 0c 00 00 00 c8 00 ff ff ff ff 1f ac
        CRC:   a7 70 0d

    systime=1439902358 freq=2430 addr=af9aabae delta_t=7.665 ms
    01 00 c8 2c 7b 
    Data / AA af9aabae (valid) /  0 bytes
        Channel Index: 12
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 0  MD: 0

        Data: 
        CRC:   c8 2c 7b

    systime=1439902358 freq=2430 addr=af9aabae delta_t=0.232 ms
    05 00 1b 2a 7b 
    Data / AA af9aabae (valid) /  0 bytes
        Channel Index: 12
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 0  MD: 0

        Data: 
        CRC:   1b 2a 7b

    systime=1439902358 freq=2454 addr=af9aabae delta_t=15.064 ms
    09 00 6e 21 7b 
    Data / AA af9aabae (valid) /  0 bytes
        Channel Index: 24
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 1  MD: 0

        Data: 
        CRC:   6e 21 7b

    systime=1439902358 freq=2478 addr=af9aabae delta_t=14.736 ms
    02 07 03 00 04 00 02 68 00 f4 89 95 
    Data / AA af9aabae (valid) /  7 bytes
        Channel Index: 36
        LLID: 2 / LL Data PDU / L2CAP start
        NESN: 0  SN: 0  MD: 0

        Data:  03 00 04 00 02 68 00
        CRC:   f4 89 95

    systime=1439902358 freq=2428 addr=af9aabae delta_t=14.968 ms
    0d 00 bd 27 7b 
    Data / AA af9aabae (valid) /  0 bytes
        Channel Index: 11
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 1  SN: 1  MD: 0

        Data: 
        CRC:   bd 27 7b

    systime=1439902358 freq=2452 addr=af9aabae delta_t=15.000 ms
    01 00 c8 2c 7b 
    Data / AA af9aabae (valid) /  0 bytes
        Channel Index: 23
        LLID: 1 / LL Data PDU / empty or L2CAP continuation
        NESN: 0  SN: 0  MD: 0

        Data: 
        CRC:   c8 2c 7b

For the second connection event, it's the masters PDU that is missing and for the third connection event is again the slaves PDU that is not captured.

dominicgs, 2015-08-18:
CRC failures are caused by noise that the receiver sees.  It is possible for the Ubertooth's receiver to see noise that the master device does not.  Therefore the Ubertooth will fail the CRC check but the master device will not.  This is something that we just have to live with when using wireless technology.

TorstenRobitzki, 2015-08-18:
@dominicgs Silly me, of cause you are right! ;-)

closed 2015-08-18 by TorstenRobitzki