greatscottgadgets/ubertooth #104: Ubertooth Wireshark Plugin - Wireshark 1.12.6

blong0159, 2015-08-24:
Trying to build the ubertooth wireshark plugin on Kali Linux 2.0 VM on Mac OSX...  Using the following:
- Kismet-2013-03-R1B
- libbtbb-2014-02-R4
- Ubertooth-2014-02-R2

When attempting to build the wireshark plugin, I get the following error in the image: 

[image]

The files do exist ... Do I need to statically link the locations of the files?

dominicgs, 2015-08-24:
It's strange that it finds epan/proto.h but not epan/emem.h especially as the files are both in the same place.

I think you'll need to specify the path to the headers, above the epan directory.

dominicgs, 2015-08-24:
Additionally: the 2014-04 release version of the plugin does not build against wireshark 1.12 as one of the features that we use was removed.

You may use the git version of the plugin without using the git version of libbtbb to fix the problem.

blong0159, 2015-08-24:
Ran git clone https://github.com/greatscottgadgets/libbtbb.git

Navigated to libbtbb/wireshark/plugins/btbb

Ran the following: 

cmake .
make 
make install 

Get the following screen ...

[image]

Then copy btbb.so to /usr/local/lib/kismet as this is what is configured in the kismet_ui.conf file for loading plugins ...  Then get the following ...  Failed to find kis_plugin_info in btbb.so

[image]

Looks like I am still missing something ...

Thanks for any assistance

dominicgs, 2015-08-24:
The btbb.so plugin is the Wireshark plugin, it has nothing to do with Kismet.

The Kismet plugin is in the Ubertooth host/kismet/plugin-ubertooth directory.

blong0159, 2015-09-14:
Dominic ::

Thanks for the notes... Got myself caught up between Kismet and Wireshark.  Biggest thing is it appears Kismet was taking my settings from /usr/local/etc/kimset.conf versus /etc/kismet.conf....

Was able to get everything up and running as well as successfully did the updates to 2015-09-R2 ...  All set, thanks!

dominicgs, 2015-09-14:
That's great to hear, thanks for letting me know.  I'll close this issue now.

closed 2015-09-14 by dominicgs