greatscottgadgets/ubertooth #105: support partial channel maps

mikeryan, 2015-08-24:


TorstenRobitzki, 2015-08-25:
@mikeryan are you going to take the LL_CHANNEL_MAP_REQ into account too? Would be handy and I promise to do some tests.

mikeryan, 2015-08-25:
Phase 1 is to consider partial channel maps during connections, phase 2 will handle ```LL_CHANNEL_MAP_REQ``` link layer control procedures. If you have any insight into how to easily generate these for testing purposes, I'm all ears!

TorstenRobitzki, 2015-08-25:
I've seen LL_CHANNEL_MAP_REQs when testing bleutoe on OS/X, using noble as GATT client (a java script library). An other, more reliable and way more expensive way, would be to use HCI.

mikeryan, 2018-06-27:
This is implemented in the latest pre-release 2018-06-R1. Please give it a try.

closed 2018-06-27 by mikeryan