greatscottgadgets/ubertooth #106: Ubertooth One firmware upgrade

ajjsss, 2015-08-25:
Attempted to follow Update Firmware instructions and can't seem to navigate to  "ubertooth-one-firmware-bin" (doesn't seem to exist)  or find  bluetooth_rxtx.dfu file to write to ubertooth one.  please help.

dominicgs, 2015-08-25:
Did you download the release archive or clone from git?

If you cloned from git there are no binary firmware images and you would need to build it yourself.  Binary firmware files are included in the release here:https://github.com/greatscottgadgets/ubertooth/releases/tag/2014-04-R1

ajjsss, 2015-08-25:
I did download the release using the following instructions from the build guide:

wget https://github.com/greatscottgadgets/ubertooth/archive/2014-02-R2.tar.gz -O ubertooth-2014-02-R2.tar.gz
tar xf ubertooth-2014-02-R2.tar.gz
cd ubertooth-2014-02-R2/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

Then I attempted to use following instructions for firmware upgrade:

extract the archive and change into directory  ubertooth-one-firmware-bin .
You may then run the  ubertooth-dfu  command like so:
$ ubertooth-dfu --write bluetooth_rxtx.dfu 
Checking firmware signature
No DFU devices found - attempting to find Ubertooth devices
1) Found 'Ubertooth One' with address 0x1d50 0x6002
Select a device to flash (default:1, exit:0):

dominicgs, 2015-08-25:
Is the ubertooth-one-firmware-bin directory still missing?  Or did the firmware update succeed?

ajjsss, 2015-08-25:
I searched everything inside the ubertooth2014-02-R2 main folder,  but the ubertooth-one-firmware-bin directory and  bluetooth_rxtx.dfu are not present anywhere in this folder.

dominicgs, 2015-08-25:
I don't understand how this can be, I've just downloaded both of the most recent Ubertooth release and the firmware directory is present.

dominicgs, 2015-08-25:
Gah!  I've just looked at the URL in your comment!  That's the github prepared source archive which we have no control over.  You need to download this version: https://github.com/greatscottgadgets/ubertooth/releases/download/2014-04-R1/ubertooth-2014-04-R1.tar.xz

I'm aware that this is confusing, unfortunately I cannot prevent GitHub from creating those archives.

I have now fixed the build guide to point to the correct URL, I'm sorry that it was misleading.

closed 2015-08-25 by dominicgs