greatscottgadgets/ubertooth #108: most ubertooth tools fail with 'uh oh, full_buf not emptied'

Hainish, 2015-08-28:
I've seen it mentioned elsewhere that this happens often on low-memory VMs and the like.  For me, this is happening on an 8G X1 Carbon amd64 machine, running Ubuntu 14.04 LTS with an otherwise minimal memory footprint.

Most tools I've tested - `ubertooth-rx`, `ubertooth-scan`, and `kismet` with `btbb` support, get the repeated error: `uh oh, full_buf not emptied`.  Looks like `ubertooth-specan-ui` does not.  I've tested this using the following combinations, with the same result:

1. `libbtbb-2014-02-R4` with `ubertooth-2014-04-R1`
2. libbtbb https://github.com/greatscottgadgets/libbtbb/commit/f53131dbc561872c2dade17adf319594f1b7d209 with ubertooth https://github.com/greatscottgadgets/ubertooth/commit/825bc8901d4b36016cdab3fcc2f49b1487571bfc

I've also tested this on a Kali VM, given 2.6G memory.  All the same result...

Hainish, 2015-08-28:
```bash
# ubertooth-util -v
Firmware revision: 2014-04-R1
```

dominicgs, 2015-09-02:
@Hainish Those specs appear to be more than enough for Ubertooth and libbtbb processing, but this bug appears more often with low power systems because the USB controller doesn't keep up with the Ubertooth.

I think the real bug here is due to a bad assumption made in the code (by me), so I'll try to fix this today.  Unfortunately I haven't been able to reproduce this issue, so I will have to ask if you, or someone else seeing the problem, can test my fix for me.

Hainish, 2015-09-02:
Sounds good, I can test in tandem with your commits.

Hainish, 2015-09-02:
(just let me know when to pull from master)

dominicgs, 2015-09-04:
The current head of the git master branch has cleaner USB handling, which I believe will fix this issue.

At the very least, it should exit as soon as it hits this error and not get in to an infinite loop printing the same error message.  Please let me know if you have the opportunity to test this.

Hainish, 2015-09-04:
Still getting this sometimes, but not as often, and I can confirm it does exit as soon as it hits the error.

dominicgs, 2015-09-04:
Thank you for your feedback, I think I'll need to revisit this.

I'm going to release the current improvements today and leave this issue open (while moving it to the next release).

savy26, 2016-04-07:
Does it now return the RSSI values associated to each received packet

elli89, 2016-07-05:
I just uploaded a patch that uses a bigger receive buffer for usb bulk transfers and might solve this issue. @Hainish could you please test it and report if it makes a difference?

Please note that I have also changed the behaviour in the way that the software does not stop on a buffer overflow so you would have to search for the error message "uh oh, full_usb_buf not emptied" in the program output.

closed 2020-12-03 by mikeryan