greatscottgadgets/ubertooth #112: Duplicate btbb protocol.

swzealord, 2015-09-15:
The file "btbb.so" is causing an error. When following the build guide exactly as stated I can not open Wireshark.

Wireshark crashes when loading configuration files. Starting Wireshark from the command line will give an error that says "Duplicate plugin found". Searching my entire Ubuntu system I could only find 1 instance of btbb.so and that is in the usr/lib/......../wireshark/libwireshark/ (The path the guide said to install the plugins into)

dominicgs, 2015-09-15:
Have you had a previous version of the plugin installed?
Which version of Wireshark are you using?
Were there any errors or warnings when building or installing the btbb plugin?
Are you running Linux?  If so, which distribution?

swzealord, 2015-09-15:
Thank you for your response.

**Have you had a previous version of the plugin installed?**
No, it was a fresh install on a fresh VM.
**Which version of Wireshark are you using?**
The one your guide points me to: 1.10.6 (I am aware of the newer 1.12, but it did not compile on my machine)
**Were there any errors or warnings when building or installing the btbb plugin?**
No errors whatsoever.
**Are you running Linux? If so, which distribution?**
I'm running Ubuntu 14.04.3 LTS, 64-bit.

My solution was to just use ubertooth-btle command with dump to PCAPNG, then export that file out of my VM to my windows machine, then reading it with the Windows version of Wireshark.

By the way, I don't know if this is the right place to ask, but I'm having trouble sniffing my BLE connections; what sequence of actions must I perform to sniff data packets? (ATT and such). All I ever get out of these dump files are Link Layer packets. I have been combing the web for hours trying to figure this out.

Thanks again,

dominicgs, 2015-09-15:
Thanks for the information, I'm not sure why it would be finding duplicate plugins, especially for btbb.so.  If you are trying to inspect BLE packets you could simply delete the btbb.so plugin as it's only used for basic rate Bluetooth packets produced by the kismet plugins.

I can't think of a reason that this would be happening and how it would find two copies of the plugin.  Could you run the following and share the output?
```
sudo updatedb
locate btbb.so
```
This should give you a couple of locations, such as the build directory and the install path, but it may show an extra one.

I'm not sure that this is the best place to discuss BLE sniffing (especially as I don't know too much about it), it may be more productive to ask on the [mailing list](https://lists.sourceforge.net/lists/listinfo/ubertooth-general) or in #ubertooth on Freenode if you are an IRC user.

raulsiles, 2015-10-02:
@swzealord Regarding the duplicate bbtp protocol issue, please check a related issue in libbtbb (https://github.com/greatscottgadgets/libbtbb/issues/29) and the commit that fixes it (https://github.com/greatscottgadgets/libbtbb/commit/ae840325b61ad74181b079db288b4309ce96746b).

dominicgs, 2016-02-04:
As @raulsiles pointed out, this was fixed in git.  It will be released as part of 2016-03-R1.

closed 2016-02-04 by dominicgs