greatscottgadgets/ubertooth #113: Only getting BTLE Adv & Scan packets

robotastic, 2015-09-23:
I seem to only be getting Ind Adv, Scan Req, & Scan Resp packets when I run ubertooth-btle in both promiscuous and follow-connection mode. I successfully complete an BLE connection while capture, but nothing is picked up.

Am I doing something obviously wrong?
ubertooth-btle -f -c btle.cap

nandra, 2015-12-10:
Same here but I can see only AVD events nothing more (using latest release 2015):

 ubertooth-btle  -f -c /tmp/pipe 

systime=1449744810 freq=2402 addr=8e89bed6 delta_t=245136.589 ms
01 0c 38 a3 c3 29 6a bc 7e e8 ee 1e 27 54 87 da 7d 
Advertising / AA 8e89bed6 (valid)/ 12 bytes
    Channel Index: 37
    Type:  ADV_DIRECT_IND

    Data:  38 a3 c3 29 6a bc 7e e8 ee 1e 27 54
    CRC:   87 da 7d

Any idea what is wrong or what else should I check? Thanks.

dominicgs, 2015-12-14:
A common problem when sniffing BLE advertising channels is that there are three channels and you may not be on the correct channel for the connection.  The solution to this is to keep trying until you are lucky enough to be on the right channel at the right time.

danurna, 2016-01-06:
Can report the same issue as @robotastic 
It used to work before I've flashed the latest firmware on my ubertooth one (and had to upgrade the tools as well). I've sniffed using a macbook and after that a RPi2 while simulating a central device first on my macbook, then on my iphone 6 and was trying to connect using a android device. 
I am thinking about downgrading my ubertooth one again...

dominicgs, 2016-01-06:
@danurna Could you downgrade the firmware to the previous release and confirm that the sniffing performance improves?

danurna, 2016-01-06:
@dominicgs I'll try. Just having issues with flashing the firmware as on my RPi2 is saying:
```
Switching to DFU Mode...
Checking firmware signature
....
....
control message unsupported
```
And on my macbook it is failing with:
```
libUSB Error: Command Error:  (-1)
Switching to DFU mode...
Checking firmware signature
.............................................................................................................................................
Detached
```

The provided workaround for the libUSBError issue is not working for me as "make clean all" fails saying 
```
make: arm-none-eabi-gcc: No such file or directory
make: *** [gccversion] Error 1
```

Right now I am trying to work around the make issue.

dominicgs, 2016-01-06:
Don't worry about those error messages, they're hard to avoid because we're resetting the Ubertooth while we're connected to it.  It looks like the firmware flashing worked.  Try running `ubertooth-util -Vv` to check.

danurna, 2016-01-06:
Ahh....you are right! Got this now running `ubertooth-util -Vv`:
```
Firmware revision: 2014-02-R2
ubertooth 2014-02-R2 (dominicgs@mercury) Thu Feb 20 13:28:01 GMT 2014
```

On the first try it followed a new connection! Seems to work now.

dominicgs, 2016-01-06:
Ok, we'll have to look in to what's changed between 2014-02-R2 and 2015-09-R1.  @mikeryan Any suggestions?

danurna, 2016-01-06:
```
Firmware revision: 2015-09-R2
ubertooth 2015-09-R2 (dominicgs@hydrogen) Sat Sep  5 12:44:21 BST 2015
```
Works for me as well! I've experienced the reported issues with version `2015-10-R1`. 

Thanks for the help!

cryptax, 2016-01-14:
I have that same issue.
From time to time, I manage to capture some other packets (LL Data PDU, L2CAP Start, LL Control PDU, LL_TERMINATE_IND) but that does not happen frequently. If there are only 3 advertising channels, then, in my case, it looks like I am unlucky most of the time as it works only seldom.

I tried with the following firmware. In my case, no significant difference (?): it works "seldom":
- 2015-10-R2 with latest tools
- 2015-09-R2 with latest tools 
- 2015-09-R2 with 2015-09-R2 tools
- 2014-02-R2 with latest tools

tsvehagen, 2016-06-08:
I experienced the same problem but like @dominicgs already pointed out, BLE uses three advertising channels and you must be on the same one as the CONNECT_REQ in order to follow the connection. ubertooth-btle defaults to advertising channel 37 but try running ubertooth-btle with -A 38 or -A 39.

When I used my Nexus 5x to connect to a BLE device it seems to favor channel 39 for some reason. Not sure if it's Android, the BT module, the target device or maybe the environment that makes this channel favorable.

I experience the same behavior with 2015-09-R2 and 2015-10-R1.

brettg0396, 2018-03-16:
Have the same issue on Ubuntu 16.04 with 2017-03-R2 (could not get 2015-10-R1 setup right). We're trying to use the device to sniff bluetooth packets for a uni project, and - on all 3 channels - we're seeing nothing but scan and adv packets despite multiple tests connecting BTLE devices in the immediate vicinity of the ubertooth. Followed the build guide to the letter, firmware is up to date (2017-03-R2, March 13th 2017, etc), got the wireshark stuff set up right, our script looks like this:

#!/bin/bash
sudo mkfifo /tmp/pipe
sudo ubertooth-btle -A 37 -f -c /tmp/pipe &
sudo wireshark -k -i /tmp/pipe &

Any ideas or known fixes?

mikeryan, 2018-06-27:
Try the latest pre-release firmware 2018-06-R1 for better results.

closed 2018-06-27 by mikeryan

kristopolous, 2021-03-22:
I ran into this extremely vexing issue and after much debugging simply swapping my antenna out with a different one solved the issue.

Hopefully this helps people who have tried everything else

mickeyl, 2021-03-31:
Same problem here – I don't have an antenna to swap…

kristopolous, 2021-04-02:
do you have a way to do a bluetooth operation you know is working fine, such as say associating a set of headphones or transferring a file between two devices?

kristopolous, 2021-04-10:
so a few things, excuse me if this is known, it was news to me

antennas have gender. even though I screwed mine on tightly it was female/female which is what caused issues
[image]

I bought one of the ebay off-brand units and this is where it was happening. I went with the official one and so far it's been ok.

This may also be more antenna related. In the off brand one I'm able to apply slight pressure and get different results. Bad sign.

this is my 3rd ubertooth one actually. Top is 2013 model. 2nd is dec/2020 off-brand over ebay, 3rd is official one just bought last week.  
[image]

They seem to overall be really temperamental devices. Nice when they work though.

e-hat, 2021-06-05:
I have a newly purchased ubertooth one from [here](https://www.amazon.com/Ubertooth-One-Aluminum-Enclosure-Bundle/dp/B07GBDD53W/ref=sr_1_1_sspa?dchild=1&keywords=ubertooth+one&qid=1622388968&sr=8-1-spons&psc=1&smid=A2JO7YP9I9Y3D6&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFHTE82NzNaSzNKTVomZW5jcnlwdGVkSWQ9QTAwNTU0NTE5UkdNSEI1Uk05ODEmZW5jcnlwdGVkQWRJZD1BMDA2NjM0OTFLMktMU0hQUU4zVkkmd2lkZ2V0TmFtZT1zcF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl). I am also getting this issue and do not have access to additional antennae. 

I have attempted changing the channels to 37, 38, and 39. These attempts are with firmware version 2020-12-R1. I have a bluetooth headset (whose communications with my iPhone I am attempting to sniff, hence the ubertooth one) that I can reliably connect/disconnect, pair/unpair whathaveyou with no change for any of these actions. I also pick up the malformed CONNECT_REQ from time to time.

 Are there any new developments with this issue? Thanks in advance.