greatscottgadgets/ubertooth #120: static build not an option on linux

ZeroChaos-, 2015-10-27:
# FIXME: This may be a hack
# perhaps there should be separate ubertooth and ubertooth-static targets?

Please do create a target so I can make static libs on linux. Typically I would want to always make dynamic libs but also make static-libs if the user requests it.

dominicgs, 2015-12-01:
Just adding notes for myself:
We already build static libs when `${WIN32}` is defined.
Need to tidy this up so that it can be controlled from a configuration option.
Are there cases when we want to build both?  Or should we default to shared and only build static if the option is specified?
Why are we building static libs for Windows? - I imagine the answer to this is that the entire cmake setup needs to be re-examined for Windows support, but that's a different issue.

ZeroChaos-, 2016-02-02:
partial fix in https://github.com/greatscottgadgets/ubertooth/commit/78f48bc8ec2b84f3a1e0d561a518365ba4ab5494

this causes static libs to be built but the binaries appear to all be compiled dynamically according to file.

ZeroChaos-, 2016-02-02:
zero@gato ubertooth % file /usr/bin/ubertooth-util                                                                                                    (git)-[master] 
/usr/bin/ubertooth-util: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, stripped

dominicgs, 2016-02-02:
```
$ ldd /usr/local/bin/ubertooth-rx 
	linux-vdso.so.1 =>  (0x00007fff0dfe2000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f4f9ef9a000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4f9ebd0000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f4f9ebb0000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4f9e992000)
	/lib64/ld-linux-x86-64.so.2 (0x000055edc7322000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f4f9e78a000)
```

ZeroChaos-, 2016-02-02:
zero@gato ~ % lddtree /usr/bin/ubertooth-rx
/usr/bin/ubertooth-rx (interpreter => /lib64/ld-linux-x86-64.so.2)
    libubertooth.so.0 => /usr/lib64/libubertooth.so.0
    libusb-1.0.so.0 => /lib64/libusb-1.0.so.0
        libudev.so.1 => /lib64/libudev.so.1
            librt.so.1 => /lib64/librt.so.1
            libm.so.6 => /lib64/libm.so.6
            libresolv.so.2 => /lib64/libresolv.so.2
            libcap.so.2 => /lib64/libcap.so.2
                libattr.so.1 => /lib64/libattr.so.1
        libpthread.so.0 => /lib64/libpthread.so.0
    libbtbb.so.0 => /usr/lib64/libbtbb.so.0
        libpcap.so.1 => /usr/lib64/libpcap.so.1
            libdbus-1.so.3 => /usr/lib64/libdbus-1.so.3
    libc.so.6 => /lib64/libc.so.6
zero@gato ~ % ldd /usr/bin/ubertooth-rx 
	linux-vdso.so.1 (0x00007b90ff910000)
	libubertooth.so.0 => /usr/lib64/libubertooth.so.0 (0x00007b90ff4e6000)
	libusb-1.0.so.0 => /lib64/libusb-1.0.so.0 (0x00007b90ff2ce000)
	libbtbb.so.0 => /usr/lib64/libbtbb.so.0 (0x00007b90ff03c000)
	libc.so.6 => /lib64/libc.so.6 (0x00007b90fec9c000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007b90ff8a5000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007b90fea80000)
	libpcap.so.1 => /usr/lib64/libpcap.so.1 (0x00007b90fe83f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007b90ff6f1000)
	librt.so.1 => /lib64/librt.so.1 (0x00007b90fe637000)
	libm.so.6 => /lib64/libm.so.6 (0x00007b90fe33f000)
	libresolv.so.2 => /lib64/libresolv.so.2 (0x00007b90fe127000)
	libcap.so.2 => /lib64/libcap.so.2 (0x00007b90fdf21000)
	libdbus-1.so.3 => /usr/lib64/libdbus-1.so.3 (0x00007b90fdcdc000)
	libattr.so.1 => /lib64/libattr.so.1 (0x00007b90fdad7000)
zero@gato ~ % file /usr/bin/ubertooth-rx
/usr/bin/ubertooth-rx: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, stripped

ZeroChaos-, 2016-02-02:
this is my full cmake line, is it correct or did I mess up?

```
cmake --no-warn-unused-cli -C /var/tmp/portage/net-wireless/ubertooth-9999/work/ubertooth-9999_build/gentoo_common_config.cmake -G Unix Makefiles -DCMAKE_INSTALL_PREFIX=/usr -DENABLE_USE_BLUEZ=ON -DUSE_PCAP=ON -DBUILD_STATIC=ON -DENABLE_INSTALL_UDEV_RULES=ON -DDISABLE_PYTHON=true -DUDEV_RULES_GROUP=usb -DUDEV_RULES_PATH=/lib/udev/rules.d -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_INSTALL_DO_STRIP=OFF -DCMAKE_USER_MAKE_RULES_OVERRIDE=/var/tmp/portage/net-wireless/ubertooth-9999/work/ubertooth-9999_build/gentoo_rules.cmake -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/ubertooth-9999/work/ubertooth-9999_build/gentoo_toolchain.cmake  /var/tmp/portage/net-wireless/ubertooth-9999/work/ubertooth-9999/host
```

mossmann, 2021-06-04:
@ZeroChaos- Is this still a concern of yours?

ZeroChaos-, 2021-06-06:
This was a response to a commit years ago that added static build support for windows but not linux (iirc).  It would be nice if static build works for linux, has anything changed to allow it?

ZeroChaos-, 2021-06-06:
kinda looks like it only builds static bins if dynamic libs are disabled, is that right?
https://github.com/greatscottgadgets/ubertooth/blob/master/host/CMakeLists.txt#L47-L56

mossmann, 2021-06-18:
> kinda looks like it only builds static bins if dynamic libs are disabled, is that right?
> https://github.com/greatscottgadgets/ubertooth/blob/master/host/CMakeLists.txt#L47-L56

It automatically selects static bins if the dynamic lib is disabled, but it should all the user to select either static or dynamic bins if both static and dynamic lib is selected. I think that bit of the cmake configuration is correct except that one of the comments could be clearer.

However, static builds don't seem to be working for me, so I'm looking deeper.

closed 2021-07-18 by straithe