greatscottgadgets/ubertooth #121: Error with Ubertooth One

ninjaztar, 2015-10-27:
Hi,

I purchase an Ubertooth recently and had followed the latest build guide on github to install the necessary tools. After installing the necessary tools, I followed the instruction on github to update the firmware but receive the following error.

ubuntu@ubuntu:~/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ ubertooth-dfu -d bluetooth_rxtx.dfu -r
libUSB Error: Timeout:  (-7)
Switching to DFU mode...
Unable to find Ubertooth

Anyone able to help me out with this? Thanks in advance.

dominicgs, 2015-10-27:
Hi @ninjaztar are you running in a VM?  The Ubertooth has a different USB product ID when in DFU mode, so you'll need to set up a rule to pass both IDs through to the VM from your host.

ninjaztar, 2015-10-28:
Hi @dominicgs, thanks for the prompt reply.

Yes, I am running it in a VM. Based on openmoko the product and vendor ID for ubertooth one is 0x1d50 and 0x6002 and for  ubertooth one DFU is 0x1d50 and 0x6003. I had added the ID into the vmx file of my VM to auto-connect it. But I am still receiving the same error as followed.

ubuntu@ubuntu:~/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ ubertooth-dfu -d bluetooth_rxtx.dfu -r
libUSB Error: Timeout: (-7)
Switching to DFU mode...
Unable to find Ubertooth

Am I doing anything wrong?

dominicgs, 2015-10-28:
Depending on the age of your Ubertooth, you may not have the OpenMoko USB IDs in DFU mode.  Before we switched over to them we were using 0xffff 0x0004.

Assuming that the Ubertooth is entering DFU mode (the lights flash in a distinctive pattern), could you check which USB ID it has?

ninjaztar, 2015-10-28:
I was able to view the USB ID in the VM before entering DFU mode which is as shown below.
[image]

Upon entering DFU mode with the command "ubertooth-util -f" the ubertooth lost it connection from the vm and I was not able to view the USB ID information of the ubertooth anymore.

Is the information of any help?

dominicgs, 2015-10-28:
Yes, I think you need to look at the USB devices available to the VM while the Ubertooth is in DFU mode.  There will be another device ID, either an OpenMoko (0x1d50) or a generic one that we used before (0xffff)

parkerpatriot, 2015-10-31:
I had this problem using my VM. It happens to me because the device changes name when in dfu mode. I navigate to my list of USB devices and add the renamed device to my machine. In my case my device name changed from `ubertooth bluetooth_rxtx` to 'usb_bootloader`. Once I have re-added it I re-run the flash command `sudo ubertooth-dfu -d bluetooth_rxtx.dfu` and it works like a charm.

ninjaztar, 2015-11-05:
Thanks for the response. Also sorry for the late reply.

Hi @dominicgs, I can't seems to find any of the ID you mentioned. Once i tried entering DFU mode to update the firmware I will receive the following error with the LED light performing a distinctive chasing pattern.

[image]

[image]

I then tried entering this command "make clean all && make && ubertooth-dfu -r -d bluetooth_rxtx.dfu" to troubleshoot it but was given the following error message.

[image]

Hi @parkerpatriot, is the ubertooth device still connected to your VM once you entered DFU mode, as my ubertooth cannot be recognize once entering DFU mode. 

Any help is appreciated. Thanks.

dominicgs, 2015-11-05:
@ninjaztar Could you show us a list of the USB devices that are attached to your host system when the Ubertooth is in DFU mode?

ninjaztar, 2015-11-05:
The first image is the ubertooth being connected to my host machine before entering DFU mode.

[image]

The second image is the ubertooth being connected to my host machine, while entering DFU mode.

[image]

is the information of any help?

dominicgs, 2015-11-05:
It's strange that Windows is seeing it as a "USB SmartCard Reader".  If you look at the properties of that device, you should be able to find the hardware IDs - VID / PID.

Alternatively, pass the "smart card reader" through to the VM and look at the output of `lsusb`.

parkerpatriot, 2015-11-05:
@ninjaztar to answer your question, I essentially lose the device as well when I put it in DFU mode. I have to go re-add it as a new device. However, my device still has a recognizable name (`usb_bootloader` in my case). Like @dominicgs said it's strange that it's showing up as USB SmartCard reader. By the way, are you using a Linux VM? That's weird. In your VM settings you can sometimes specify what devices should be automatically associated with the VM on startup. I would go to those settings after putting it in DFU mode and try associating this "USB SmartCard Reader" with the VM. Then restart your VM and try the process of putting it in DFU mode again. When you do, the VM shouldn't end up losing the device even though it will now be renamed. Then you could see if  your VM recognizes the device correctly.

pmar9000, 2015-11-06:
Hi

I had the same problem doing dfu through a VM (in my case VirtualBox).
I had to add a filter for both 1d50:6002 and 1d50:6003 for the usbbootloader device.
Then I got the dfu to work.

/Peter

ninjaztar, 2015-11-09:
Hi all, thanks for the response.

It seem the firmware of my ubertooth had been updated to 2015-10-R1, as shown in the image below.

[image]

However, when I was updating the firmware with the command "ubertooth-dfu -d bluetooth_rxtx.dfu -r", I receive the following output. 

[image]

The expected output from github when the command was being entered was supposed to be similar to the one shown below. .

Checking firmware signature
No DFU devices found - attempting to find Ubertooth devices

1) Found 'Ubertooth One' with address 0x1d50 0x6002

Select a device to flash (default:1, exit:0):

Is there anything wrong with the update?

dominicgs, 2015-12-01:
@ninjaztar That update looks good.  There have been changes to the DFU tool recently that mean the output is not always the same.

As we have solved this issue, I will close this bug report.

closed 2015-12-01 by dominicgs

ninjaztar, 2015-12-03:
Alright. Thanks everyone for all the help.

trublsom, 2016-01-12:
I can't update my firmware either. 

I completed the build guide. I am using vmware workstation 12 (free). I have tried updating with kali 2.0 (desktop) and ubuntu 15.10 (laptop). 

blue@ubuntu:~/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ ubertooth-util -v
Firmware revision: 2012-10-R1
blue@ubuntu:~/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ ubertooth-util -V
control message unsupported
error: -9

ubertooth-one-firmware-bin$ ubertooth-dfu -d bluetooth_rxtx.dfu -r
libUSB Error: Timeout:  (-7)
Switching to DFU mode...
Unable to find Ubertooth

/ubertooth-one-firmware-bin$ make clean all && make && ubertooth-dfu -r -d bluetooth_rxtx.dfu
make: *** No rule to make target 'clean'.  Stop.

blue@ubuntu:~$ lsusb
Bus 002 Device 007: ID 1d50:6002 OpenMoko, Inc. 

Windows sees my device as: "bluetooth_rxtx"

[image]

dominicgs, 2016-01-12:
@trublsom Did you try the solution from @pmar9000 above: https://github.com/greatscottgadgets/ubertooth/issues/121#issuecomment-154429656

The Ubertooth uses two different USB IDs for normal operation and firmware update mode.  What's happening is that the Ubertooth resets itself at the start of the process, to reappear with the other ID.  Unfortunately this other ID is not passed through to the VM, so the rest of the tool fails.

Add a second USB pass through rule for the ID 1d50 6003

trublsom, 2016-01-13:
I couldn't find how to add a filter for the usb devices using vmware workstation... so i turned one of my old desktop into a kali 2.0 box (bare metal). I ran the script created by dinosec. https://github.com/dinosec/ubertooth-install

Double checked that the everything from the build guide was completed then ran the dfu command. 

It looks like my problem was VM related.  Since the firmware update work the very first time when I did it on bare metal.

Also I did get the same output as ninjaztar did (ninjaztar commented on Nov 8, 2015). However after running "ubertooth-util -v" I saw that it had been updated to 2015-10-R1

Thank you for responding to me even though this issue had been closed.

garyv100, 2016-03-11:
Hi,

I have just purchased Ubertooth One w/ firmware release dated back to 2012.

I am running Ubuntu 14.04 LTS in VMWare Workstation 11. It looks like that there were similar issues w/ VM that were resolved by adding filters to USB device which I do not know how to do in VMware. If someone could help me with that - it would be great.

This is what I get when I try to update the firmware for 2015.
///////////////////////////////////////////////////////////////////////////
garyv@ubuntu:~/ubertooth/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ sudo ubertooth-dfu -d bluetooth_rxtx.dfu -r
libUSB Error: Command Error: (-1)
Switching to DFU mode...
Checking firmware signature
.......................................................................................................................................
libUSB Error: Command Error: (-1)
/////////////////////////////////////////////////////////////////////////////
When I check rev:
garyv@ubuntu:/usr/local/bin$ ./ubertooth-util -v
Firmware revision: 2015-10-R1

First question, given the above, did the firmware updated correctly?

To see if it did, for starters i tried to replicate your youtube demo using:
////////////////////////////////////////////////////////
sudo ubertooth-scan -b hci0 -t 40 –x 
Ubertooth scan
systime=1457634961 ch=10 LAP=f19d9a err=2 clk100ns=2335214731 clk1=373634 s=-54 n=-62 snr=8
systime=1457634971 ch=57 LAP=a667b3 err=2 clk100ns=2429973401 clk1=388796 s=-52 n=-63 snr=11
??:??:??:F1:9D:9A
AFH map: 0x00040000000000000000
??:??:??:A6:67:B3
AFH map: 0x00000000000000020000
//////////////////////////////////////////////////////////////////
or sometimes I get noting, it just hangs
///////////////////////////////////////////////////////////////
garyv@ubuntu:~/ubertooth/ubertooth-2015-10-R1/ubertooth-one-firmware-bin$ sudo ubertooth-scan -b hci0 -t 40 –x

Ubertooth scan
///////////////////////////////////////////////////////////////////

Is this a VMWare issue, am I doing something wrong, how do I fix it?

chuzairps, 2022-06-29:
Hi there, 
Facing firmware updating issue

sudo ubertooth-dfu -d bluetooth_rxtx.dfu -r
Checking firmware signature
Unknown DFU suffix length: 0
Signature check failed, firmware will not be update