greatscottgadgets/ubertooth #122: ubertooth-follow: no PCAP support (WTAP_ENCAP = 0)

parkerpatriot, 2015-10-31:
I am trying to get ubertooth-follow to work and save the data to a pcap file. I have installed the latest version of the ubertooth tools, updated libpcap, and also installed the latest release of libbtbb. After updating libpcap I no longer get the old "link-layer type 255 isn't supported" error message. However, I get no packets showing up in the capture file when I use the `-q` option with ubertooth-follow. If I instead use the `-r` to capture to a pcapng file I do get packets showing up but wireshark fails to dissect the packet and just displays WTAP_ENCAP = 0 in the Info column.

Version:
libbtbb: 2015-10-R1
ubertooth: 2015-10-R1
OS: Ubuntu 14

[image]

parkerpatriot, 2015-12-03:
Any help on this?

dominicgs, 2015-12-07:
@parkerpatriot Do you have the btbredr dissector built for Wireshark?  If that is missing the packets will not be correctly decoded.

PCAP/PCAPNG for Basic Rate Bluetooth in libbtbb is in need of some work, it is one of our goals for the next release to streamline the code and make the output consistent between the formats.

parkerpatriot, 2015-12-12:
I get the following output when I build btbredr:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version.
libwireshark-dev is already the newest version.
wireshark is already the newest version.
wireshark-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 408 not upgraded.
mkdir: cannot create directory ‘build’: File exists
-- Plugin will be installed in: /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jparker/lib/libbtbb.git/wireshark/plugins/btbredr/build
[ 25%] Generating plugin.c
Scanning dependencies of target btbredr
[ 50%] Building C object CMakeFiles/btbredr.dir/plugin.c.o
[ 75%] Building C object CMakeFiles/btbredr.dir/packet-btbredr.c.o
[100%] Building C object CMakeFiles/btbredr.dir/packet-btlmp.c.o
/home/jparker/lib/libbtbb.git/wireshark/plugins/btbredr/packet-btlmp.c: In function ‘proto_register_btlmp’:
/home/jparker/lib/libbtbb.git/wireshark/plugins/btbredr/packet-btlmp.c:2262:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/jparker/lib/libbtbb.git/wireshark/plugins/btbredr/packet-btlmp.c:2267:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
/home/jparker/lib/libbtbb.git/wireshark/plugins/btbredr/packet-btlmp.c:2272:4: warning: large integer implicitly truncated to unsigned type [-Woverflow]
    FT_UINT64, BASE_HEX, NULL, 0x0000ffffffffffff,
    ^
Linking C shared module btbredr.so
[100%] Built target btbredr
[100%] Built target btbredr
Install the project...
-- Install configuration: ""
-- Installing: /usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins/btbredr.so
```
It seems to be installing ok, but I don't know what might be wrong with libbtbb.

dominicgs, 2015-12-15:
That build output is fine, although it would be nice to remove the warnings, they aren't going to be causing a problem here.

parkerpatriot, 2015-12-15:
Do you not get these types of errors when you compile? I feel like I get warnings like this all the time when I'm building the tools.

dominicgs, 2015-12-15:
I see these warnings for the Wireshark dissectors, but I don't remember seeing them for any of the other Ubertooth tools.  If you see them, could you send me build output and I'll attempt to fix them?

dominicgs, 2015-12-15:
Looking at your original issue, it appears that Wireshark isn't finding the dissector, although `/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins/btbredr.so` appears to be a valid path.

michaelp123, 2016-02-22:
Any news about this issue?

cecio, 2016-11-26:
Hi.
I have the same exact issue. Let me know if I can do some tests for you.
Thanks,

closed 2020-12-03 by mikeryan