greatscottgadgets/ubertooth #123: Error during building the Bluetooth baseband library

MrSarinth, 2015-11-06:
I'm trying to set up Ubertooth One. I followed steps in the Build Guide and got following error

Scanning dependencies of target btbb
[ 12%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_packet.c.o
[ 25%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_piconet.c.o
[ 37%] Building C object lib/src/CMakeFiles/btbb.dir/bluetooth_le_packet.c.o
[ 50%] Building C object lib/src/CMakeFiles/btbb.dir/pcap.c.o
In file included from /Users/ondrejziak/libbtbb-2015-10-R1/lib/src/pcap.c:25:
/Users/ondrejziak/libbtbb-2015-10-R1/lib/src/pcap-common.h:28:10: fatal error: 'pcap/bluetooth.h' file not found
#include <pcap/bluetooth.h>
         ^
1 error generated.
make[2]: *** [lib/src/CMakeFiles/btbb.dir/pcap.c.o] Error 1
make[1]: *** [lib/src/CMakeFiles/btbb.dir/all] Error 2
make: *** [all] Error 2

it occurred during $ make. 
I'm using Mac OS X El Capitan.

Any help welcome. Thanks

dominicgs, 2015-11-06:
It looks like you're missing the file `pcap/bluetooth.h` from your libpcap installation, of the pcap headers aren't in your include path.

Could you show us the output of the cmake command and then run `make VERBOSE=1` to enable verbose logging in the build process?

MrSarinth, 2015-11-06:
$ cmake ..  just step before $ make 

-- The C compiler identification is AppleClang 7.0.0.7000176
-- The CXX compiler identification is AppleClang 7.0.0.7000176
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
fatal: Not a git repository (or any of the parent directories): .git
-- Pcap include dirs set to /usr/include
-- Pcap library set to  /usr/lib/libpcap.dylib
-- Looking for pcap_version
-- Looking for pcap_version - found
-- Looking for pcap_open_dead
-- Looking for pcap_open_dead - found
-- Looking for pcap_freecode
-- Looking for pcap_freecode - found
-- Looking for pcap_breakloop
-- Looking for pcap_breakloop - found
-- Looking for pcap_create
-- Looking for pcap_create - found
-- Looking for pcap_datalink_name_to_val
-- Looking for pcap_datalink_name_to_val - found
-- Looking for pcap_datalink_val_to_description
-- Looking for pcap_datalink_val_to_description - found
-- Looking for pcap_datalink_val_to_name
-- Looking for pcap_datalink_val_to_name - found
-- Looking for pcap_findalldevs
-- Looking for pcap_findalldevs - found
-- Looking for pcap_free_datalinks
-- Looking for pcap_free_datalinks - found
-- Looking for pcap_get_selectable_fd
-- Looking for pcap_get_selectable_fd - found
-- Looking for pcap_lib_version
-- Looking for pcap_lib_version - found
-- Looking for pcap_list_datalinks
-- Looking for pcap_list_datalinks - found
-- Looking for pcap_set_datalink
-- Looking for pcap_set_datalink - found
-- Looking for pcap_open
-- Looking for pcap_open - not found
-- Looking for pcap_findalldevs_ex
-- Looking for pcap_findalldevs_ex - not found
-- Looking for pcap_createsrcstr
-- Looking for pcap_createsrcstr - not found
-- Found PCAP: /usr/include  
-- Found PythonInterp: /usr/bin/python (found version "2.7.10") 
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   btbb

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/ondrejziak/libbtbb-2015-10-R1/build 

this is output of $ make VERBOSE=1

/opt/local/bin/cmake -H/Users/ondrejziak/libbtbb-2015-10-R1 -B/Users/ondrejziak/libbtbb-2015-10-R1/build --check-build-system CMakeFiles/Makefile.cmake 0
/opt/local/bin/cmake -E cmake_progress_start /Users/ondrejziak/libbtbb-2015-10-R1/build/CMakeFiles /Users/ondrejziak/libbtbb-2015-10-R1/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f lib/src/CMakeFiles/btbb.dir/build.make lib/src/CMakeFiles/btbb.dir/depend
cd /Users/ondrejziak/libbtbb-2015-10-R1/build && /opt/local/bin/cmake -E cmake_depends "Unix Makefiles" /Users/ondrejziak/libbtbb-2015-10-R1 /Users/ondrejziak/libbtbb-2015-10-R1/lib/src /Users/ondrejziak/libbtbb-2015-10-R1/build /Users/ondrejziak/libbtbb-2015-10-R1/build/lib/src /Users/ondrejziak/libbtbb-2015-10-R1/build/lib/src/CMakeFiles/btbb.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f lib/src/CMakeFiles/btbb.dir/build.make lib/src/CMakeFiles/btbb.dir/build
[ 12%] Building C object lib/src/CMakeFiles/btbb.dir/pcap.c.o
cd /Users/ondrejziak/libbtbb-2015-10-R1/build/lib/src && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -DENABLE_PCAP -DRELEASE=\"GITDIR-NOTFOUND\" -DVERSION=\"0.3\" -Dbtbb_EXPORTS -fPIC   -o CMakeFiles/btbb.dir/pcap.c.o   -c /Users/ondrejziak/libbtbb-2015-10-R1/lib/src/pcap.c
In file included from /Users/ondrejziak/libbtbb-2015-10-R1/lib/src/pcap.c:25:
/Users/ondrejziak/libbtbb-2015-10-R1/lib/src/pcap-common.h:28:10: fatal error: 'pcap/bluetooth.h' file not found
#include <pcap/bluetooth.h>
         ^
1 error generated.
make[2]: *** [lib/src/CMakeFiles/btbb.dir/pcap.c.o] Error 1
make[1]: *** [lib/src/CMakeFiles/btbb.dir/all] Error 2
make: *** [all] Error 2

dominicgs, 2015-11-06:
Cmake seems to be finding libpcap without any problems, it believes that it's in /usr/include.  Is that where it is?  Do you have a directory called /usr/include/pcap ?  Is there a bluetooth.h in it?

MrSarinth, 2015-11-06:
I found those two files.
/System/Library/Frameworks/IOBluetooth.framework/Versions/A/Headers/Bluetooth.h
/System/Library/Frameworks/Kernel.framework/Versions/A/Headers/IOKit/bluetooth/Bluetooth.h

Folder /usr/include/pcap doesn't contain any bluetooth.h header.

dominicgs, 2015-11-06:
Ah, it looks like that's the problem, your version of libpcap is missing bluetooth.h, perhaps because it's older?

You can try to run cmake with -DUSE_PCAP=OFF to disable pcap use.

MrSarinth, 2015-11-06:
I don't see how it could be older. Is there any way to obtain the missing file? 

Running ```$ cmake -DUSE_PCAP=OFF``` results in following error

CMake Error: The source directory "/Users/ondrejziak/libbtbb-2015-10-R1/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

mikeryan, 2015-11-06:
Per the [build guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide), on OS X you have to install a version of libpcap from homebrew:

    brew install libusb wget cmake pkg-config homebrew/dupes/libpcap

MrSarinth, 2015-11-06:
Thanks, mikeryan
installation completed without any errors.

closed 2015-11-07 by mikeryan

jdetrick, 2016-11-30:
I had bluetooth.h at /opt/local/include/pcap/bluetooth.h

- just did a copy to the /usr/include/pcap directory and it compiled.

cp /opt/local/include/pcap/bluetooth.h /usr/include/pcap/bluetooth.h