greatscottgadgets/ubertooth #125: Problem capturing session in Wireshark

jlubeck, 2015-11-12:
Hi,

I'm following this guide:
https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark

But when I hit Start on Wireshark (step 8) I'm guetting this error:
`The capture session could not be initiated because "/tmp/pipe" is neither an interface nor a socket nor a pipe`

Any idea what I might be doing wrong?

Thanks

dominicgs, 2015-11-12:
To help us diagnose what's going wrong here, could you provide us with some more information about your setup?
OS + version?
Wireshark version?
Ubertooth tools version?
Ubertooth firmware version?

jlubeck, 2015-11-12:
Ubuntu 14.04 LTS inside a VirtualBox on OSX
Wireshark 1.10.6
Ubertooth Tools 2015-10-R1
Ubertooth Firmware 2015-10-R1

Thanks!

dominicgs, 2015-11-12:
Looking at the error message, it looks like `/tmp/pipe` isn't a pipe.  Could you `ls -l` it and see what it is?

jlubeck, 2015-11-12:
jan@jan:~$ ls -l /tmp/pipe
    -rw-rw-r-- 1 jan jan 180477 Nov 12 12:06 /tmp/pipe

dominicgs, 2015-11-12:
That looks like a regular file to me.  You should delete it and run `mkfifo /tmp/pipe` to replace it with a fifo.

jlubeck, 2015-11-12:
Mmmm that's strange. I made the fifo yesteday. I was trying to make the fifo again today and it was telling me it already existed. I didn't try recreating it though. That seems to have fixed it. Thanks!

Any idea if the fifo turns back to a file on computer reset or something?

dominicgs, 2015-11-12:
It's in `/tmp` so it probably got destroyed when you rebooted.  If you then ran `ubertooth-btle -f -c /tmp/pipe` it would find that the output file was missing and create it, defaulting to a regular file.

jlubeck, 2015-11-12:
Ahhh got it. Thanks!
Now I'm running into this issue again: https://github.com/greatscottgadgets/ubertooth/issues/61
but that'a another story... I'm closing this one. If you could check that one, I would really appreciate it. Thanks for your help!

closed 2015-11-12 by jlubeck