greatscottgadgets/ubertooth #127: uh oh, full_usb_buf not emptied AFH map: 0x00000000000000000000

Darokthar, 2015-11-22:
I just got an ubertooth one. I build the source from git and uploaded the firmware.
The output of ubertooth-util is:
`$ ubertooth-util -v`
`$ Firmware revision: git-fd9748c`
`$ubertooth-util -V `
`$ ubertooth git-fd9748c (darokthar@DaroktharLaptop) So 22. Nov 21:23:49 CET 2015`
I never got more than maybe 3-5 packets, I'm running debian 8. Has anyone the same issue?

dominicgs, 2015-11-23:
The "uh oh, full_usb_buf not emptied" error occurs when the host is not able to keep up with the rate that Ubertooth is sending data over USB.  This can be for a number of reasons, but in this case it may be an issue with the code from git - there was a recent patch that may have caused some performance issues.

Could you try using the 2015-10-R1 release version of libbtbb and Ubertooth?

Darokthar, 2015-11-23:
I tried both. Running `ubertooth-rx` without arguments and `ubertooth-rx -l foo -r bar` i get the error in both cases. I'll try the 2015-10-R1 release tomorrow.

dominicgs, 2015-11-23:
@elli89 I think this may be related to the changes in AFH/hopping pattern calculations.  What do you think?

isplitt, 2015-11-24:
elli89 will most likely be unavailable till the middle of December - i think if this issue still exists he will look into it then

Darokthar, 2015-11-24:
Ok, i removed the libbtbb with `cmake/cleanup.sh` and installed libbtbb-2015-10-R1/ afterwards. Then i compiled and installed 2015-10-R1 and checked with `ubertooth-util -v` and `ubertooth-util -V` if everything is correct. Both say i installed the 2015-10-R1 version.

The error still occurs after a while. Is the ubertooth data transfer rate increased, if it is unable to catch a lot of bluetooth packets?

dominicgs, 2015-11-25:
The Ubertooth data transfer rate is constant, it passes a stream of symbols to the host system, where the libbtbb code searches for packets.

I have some ideas to lessen the impact of the full_usb_buf error, so I'll take a look at that as soon as I can.

gripet, 2016-05-16:
I have the same problem
2015-10-R1, Firmware revision: git-f33b0ea
Is there any news about this problem?

asad-lesani, 2016-06-14:
I also have the same issue, even for 15 seconds scan I get this error. I tried different high performance processor with Ubuntu on it. Any new?

elli89, 2016-07-05:
I just uploaded a patch that uses a bigger receive buffer for usb bulk transfers and might solve this issue. @gripet @Darokthar could you please test it and report if it makes a difference?

Please note that I have also changed the behaviour in the way that the software does not stop on a buffer overflow so you would have to search for the error message "uh oh, full_usb_buf not emptied" in the program output.

asad-lesani, 2016-07-05:
Can I know where is the patch?

elli89, 2016-07-05:
Please clone my fork of the Ubertooth project 'https://github.com/elli89/ubertooth' and checkout the branch 'ringbuffer_fifo'. Be sure to use the git master of libbtbb and a current firmware on your Ubertooth stick. You can find a firmware binary at https://github.com/elli89/ubertooth/releases/tag/2016-07-alpha

dominicgs, 2016-07-05:
@asad-lesani Almost correct.  You will need to use the very latest (unreleased) version of libbtbb, which can be found here: https://github.com/greatscottgadgets/libbtbb/archive/master.zip

Use that instead of https://github.com/greatscottgadgets/libbtbb/archive/2015-10-R1.tar.gz

asad-lesani, 2016-07-05:
Perfect,
let me try that.
Should I still use the 2015-10-R1 firmware on Ubertooth dongle?

asad-lesani, 2016-07-05:
I tried to install the libraries, after that I get this error:
```
ubertooth-rx
libUSB Error: Command Error:  (-9)
Ubertooth running very old firmware found.
Please upgrade to latest released firmware.
```
The firmware of the module is the latest one and I can confirm it with `ubertooth-util -v` that shows: `Firmware revision: 2015-10-R1`. The version of the library is also (using `ubertooth-util -V`):  `ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:56 BST 2015`
Do you have any idea about the issue?

dominicgs, 2016-07-05:
Try this firmware that I've just built: https://dominicspill.com/ubertooth/bluetooth_rxtx.dfu

asad-lesani, 2016-07-05:
I have updated the firmware to new version and it shows: `Firmware revision: git-3c0f8ec`.
But still after a while I get the error of `uh oh, full_usb_buf not emptied`

tomsaul, 2016-07-27:
I'm new to GitHub, so perhaps I am missing something obvious, but I am looking for the stuff from elli89 referenced above, and I am not seeing the releases and tags he mentioned in his repo.  Specifically the ringbuffer_fifo branch, and the new firmware release.  For the latter, does the version posted by dominicgs contain the same fixes?

asad-lesani, 2016-07-27:
I tied this firmware and the new library but still I'm getting the same
error after a while.

tomsaul, 2016-07-27:
# ubertooth-scan
I installed the firmware provided by dominicgs above, and rebuilt libbtbb and the tools from the current GIT master.  When I run any of the tools I get something similar to the following.

Ubertooth API version 0 found, libubertooth requires 1.
Please upgrade to latest released firmware.

I'm now trying to build the firmware from the same pull, and it will not compile - problem finding crt0.o - I'll take a closer look later today, but any suggestions (on either problem)?

dominicgs, 2016-07-28:
The changes from @elli89 have been merged in to master.  You will need to use git versions of libbtbb, ubertooth and the ubertooth firmware.

I have no idea where crt0.o is meant to be coming from.  Could you give more information on the problem building the firmware (perhaps in a new issue report?).

tomsaul, 2016-07-28:
I will open a new ticket shortly

village-idiot, 2016-07-29:
I see this when piping data using 'stdbuf' or 'expect_unbuffer'. 
While I do not see this error message when using 'unbuffer'
it still stops frequently and unpredictably when 'unbuffer'
is used. I have found no time or sequence of events that
trigger this, it seems purely random. The reason for using
'unbuffer' and variants is to eliminate the delay of displayed
data when commands are piped. Seems Linux uses a 4K
buffer for efficiency when it detects piping thinking the command
result is not being viewed real time.

tomsaul, 2016-07-29:
re the buffer fix, I have been testing this for the last couple of days, and it does seem to have helped.  I am still seeing the messages occasionally, but it seems to recover.  The one possible issue is that it does seem to 'go to sleep' for about a minute after one of these errors occurs (using ubertooth-scan).  I never get any samples for a while after I see this - I'll continue to test, but this is definitely a large step forward.

dominicgs, 2016-07-29:
As @elli89 commented in another issue report, we're unlikely to be able to completely stop this from happening, but we can probably do a lot to reduce it.  It looks like those efforts are paying off.  I think we'll need to change the messaging to make it more clear what is happening.

village-idiot, 2016-08-01:
Will the firmware update alone increase the time between error messages? Or will it take all the steps outlined above to obtain the best results?

dominicgs, 2016-08-01:
You don't need the steps above as I've now merged these changes in to the master git repository.  However, you will need to build and run the latest code from git for both libbtbb and Ubertooth, including updating the firmware.

village-idiot, 2016-08-01:
Ok cloned : git clone git://github.com/greatscottgadgets/libbtbb
Created the 'build' directory, cd to the 'build' directory, then run:
cmake ..
make
make install

Indicated the installation was completed.

The cloned : git clone git://github.com/greatscottgadgets/ubertooth
First thing I noticed is there is no CMakeLists.txt in the 
'ubertooth' directory.

Did I grab the wrong files?

dominicgs, 2016-08-01:
You have the right files, you're just in the wrong part of the directory structure.  To build the host tools you need to be in the host/ subdirectory.  Then it's:
```
mkdir build
cd build
cmake ..
make
sudo make install
```

village-idiot, 2016-08-01:
Fantastic. Should have run 'find' and would have stumbled right into it. I am finding directories with the firmware source but not the bluetooth_rxtx.dfu name. Before I wander off and brick a card is the 'bluetooth_rxtx.dfu' filename lurking in the ubertooth directory or should I use the link to the version you listed above at:

https://dominicspill.com/ubertooth/bluetooth_rxtx.dfu

dominicgs, 2016-08-01:
`bluetooth_rxtx.dfu` is the compiled firmware, but we only put the source in the repository.  You can either build your own or download the linked version, I believe either should work.

village-idiot, 2016-08-01:
Many thanks. I attempted to use ubertooth-dfu to download the firmware from your link to the device but it kept balking about an Ubertooth-One not being found. At that point dropped into the source file directory were you stowed Bluetooth_rxtx and ran the line:
     make clean all && make && ubertooth-dfu -r -d bluetooth_rxtx.dfu

There was an error message of: 
   libUSB Error: No Device: Check Ubertooth is connected to host (-4) 
and then it dropped back to a prompt. In another terminal window running:
  ubertooth-util -U0 -v 
returned the line 

  Firmware Revision: git-2bc6502

I compared this to another card that was known to be running with the 2015-10-R1 firmware and it returned the information matching the original firmware. I took the second card and ran:
   make clean all && make && ubertooth-dfu -r -d bluetooth_rxtx.dfu
again. It still balked a device was not found after finish running but again a check revealed the more
recent firmware version 'git-2bc6502'.

The firmware is updating. Not sure what is amiss with my procedure.

Thanks for the guidance.

VI

tomsaul, 2016-08-01:
There were comments about this elsewhere - with the thought being that it was just a timing issue - the update triggers a device reboot, and it is not recovering fast enough for the program to get the status.  

From my experience it completes, so its really a false alarm.

dominicgs, 2016-08-01:
Try running `ubertooth-dfu -d bluetooth_rxtx.dfu`  What output do you get?

village-idiot, 2016-08-02:
It yields the same error message about the device not being found. ubertooth-dfu -d bluetooth_rxtx.dfu was the first attempt to load the firmware downloaded from your link and produces the message ' libUSB Error: No Device: Check Ubertooth is connected to host (-4)'. It was after that I tried the build instructions I found in another message chain. It makes me wonder if ubertooth-dfu -d bluetooth_rxtx.dfu is actually successful but the error message is masking reality. I think I have one more unmolested ubertooth-one card. Will try incrementally testing and see where the first hint of a successful firmware update takes place.

Thanks

VI

dominicgs, 2016-08-02:
Your previous comment said that you had an extra -r argument to
ubertooth-dfu. I was asking you to try it without.

Are you running in a virtual machine?

village-idiot, 2016-08-02:
My apology for lack of attention to the details. I will try it without the '-r'. 

No virtual machines are being used. Straight up loads of Kali or Fedora are used on bootable flash. I find virtual machines useful in production once the app is vetted but for development I just do not want the snake in the grass problems that seem to come with the additional variables in VM's. With USB 3.0, cheap flash drives etc it seems much more convenient to simply make bootable flash drives and run from them. If one is corrupted I simply reach in the drawer for a clone of yesterdays end of day effort.

village-idiot, 2016-08-02:
Dominic
Ran:
        'ubertooth-dfu -d bluetooth_rxtx.dfu 

Voila!  No error message.
Grab my son's ubertooth card and ran:
        'make clean all && make && ubertooth-dfu  -d bluetooth_rxtx.dfu'       (dropped the -r switch)
and it similarly exited without error.

Now what happens with the cards without running the '-r' switch is the card is found
to be running the row of colored LED's in a back and forth pattern (Green, Yellow, Red, Red).
This is addressed by unplugging the card and re-inserting it into the USB port.

On the URL at:
       https://github.com/greatscottgadgets/ubertooth/wiki/Firmware
it indicates that removing and reinserting is appropriate for resetting the card. It also 
states the command:
       ubertooth-dfu --detach 
will perform the same function however when I ran the above line
      'ubertooth-dfu'
simply displayed the help menu.

Then using the '-r' switch alone with ubertooth-dfu
returned the error message:
      libUSB Error: No Device: Check Ubertooth is connected to host (-4)

The back and forth LED pattern stopped however and normal
comms with the device seemed to resume as I could use
other ubertooth tools with the device after running 
    ubertooth-dfu -r

In any event all cards are now running the same firmware and
the other Ubertooth tools seem to be working as well as if
not better than before.

Thanks for your patience.

VI

bsamuels453, 2016-11-19:
for anyone who found this on google: since the firmware link on @dominicgs 's website died, you can compile that firmware yourself:

1. download @elli89 's ubertooth master branch here: https://github.com/elli89/ubertooth/archive/master.zip

2. follow compile instructions here under "Developing Firmware"
https://github.com/greatscottgadgets/ubertooth/wiki/Firmware

tomsaul, 2017-04-04:
Should this issue also be closed with the latest firmware/tools?

dominicgs, 2017-04-04:
I think it could be, but I'd like one of the users who has seen this issue to confirm that they no longer see it.

tomsaul, 2017-04-04:
OK - I have been using a build with some of the earlier patches and have been fine, but I'll try to install this new version and run for a while to confirm.  I'll report then!

asad-lesani, 2017-04-04:
Hello,

Could you tell me the location of latest firmware then I can upgrade it on
my modules to check if it is working?
Then I would be able confirm it.

Thanks.

dominicgs, 2017-04-04:
@asad-lesani There are binary firmware files included in the release package (ubertooth-2017-03-R2.tar.xz) available from: https://github.com/greatscottgadgets/ubertooth/releases/tag/2017-03-R2

You'll also need to upgrade libbtbb to the latest version, available from here: https://github.com/greatscottgadgets/libbtbb/releases/tag/2017-03-R2

asad-lesani, 2017-04-04:
> releases/tag/2017-03-R2

Perfect,

I'll try these versions and leave it running for some days to see how it
works.

Asad.

dominicgs, 2017-04-10:
@asad-lesani Any update on this?

tomsaul, 2017-04-10:
I've been running the updated version for a few days on two systems now with no noticeable problem.

dominicgs, 2017-04-10:
Thanks Tom!  I'm going to close this issue.

Please let me know, or reopen the issue, if you see the error again.

closed 2017-04-10 by dominicgs