greatscottgadgets/ubertooth #128: Getting RSSI with ubertooth-btle

1am, 2015-12-01:
Hi

I've been looking into the raw data returned by `ubertooth-btle` utility while writing ubertooth support for my simple scanning/testing utility. What I'm doing is setting up a pipe like described in [Capturing BLE in Wireshark](https://github.com/greatscottgadgets/ubertooth/wiki/Capturing-BLE-in-Wireshark) but instead of Wireshark I'm using my own piece of simple code. 

The thing is that I can't really find anything changing correspondingly to the distance of the advertising device. Is there any way to get obtain the raw RSSI of received package data from ubertooth FW passed on to ubertooth-btle? Is it there and I just can't seem to notice it?

Other thing I've tried was `ubertooth-specscan` but it doesn't return the values associated to each received packet which is what I need to get out of ubertooth.

I'm working on the following versions:

```
$ ubertooth-util -v
Firmware revision: 2015-10-R1
$ ubertooth-util -V
ubertooth 2015-10-R1 (dominicgs@hydrogen) Mon Oct 12 22:30:56 BST 2015
```

Best regards,
Piotr

akhilpo, 2016-01-20:
I am too trying to get RSSI values from the ubertooth-btle. It would be very helpful if RSSI value can be seen in wireshark. Have you made any progress on this?

mikeryan, 2016-02-04:
This is something I will include in our next release.

akhilpo, 2016-02-08:
@mikeryan could you let us know an approximate date for the next release?

dominicgs, 2016-02-08:
The next release is scheduled for March.

kandizzy, 2016-11-23:
is anyone still working on this enhancement?

mikeryan, 2016-11-23:
We still haven't got a date for the next release scheduled, but you can try the host code and firmware from master which have both had this feature for ages. They're currently quite stable.

kandizzy, 2016-12-17:
@mikeryan this is working great, thanks so much! I have a question about piping information into tshark, it is sort of working but only sending the frame and PPI data. Is the best place to ask on the IRC channel?

mikeryan, 2016-12-17:
On Fri, Dec 16, 2016 at 07:26:16PM -0800, Carrie Kengle wrote:
> @mikeryan this is working great, thanks so much! I have a question about piping information into tshark, it is sort of working but only sending the frame and PPI data. Is the best place to ask on the IRC channel?

Yes, although I don't know if we have any tshark expertise therein.

foxx, 2017-12-19:
Hey guys, any idea if this made it into a stable release? Struggling to get this working.

mikeryan, 2017-12-20:
The latest release 2017-03-R2 has this feature. Additionally the WIP le_phy branch has experimental firmware that does this more robustly.

mikeryan, 2018-06-27:
The latest pre-release has integrated the le_phy changes.

closed 2018-06-27 by mikeryan