greatscottgadgets/ubertooth #129: kismet_ubertooth build failure in git

ZeroChaos-, 2015-12-02:
repository:               https://github.com/greatscottgadgets/ubertooth.git
   at the commit:            9848cd9aefe104c890382d9c420704536ebf4c06
>>> Source unpacked in /var/tmp/portage/net-wireless/kismet-ubertooth-9999/work
>>> Preparing source in /var/tmp/portage/net-wireless/kismet-ubertooth-9999/work/kismet-ubertooth-9999 ...
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/kismet-ubertooth-9999/work/kismet-ubertooth-9999 ...
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/kismet-ubertooth-9999/work/kismet-ubertooth-9999 ...
make -j32 KIS_SRC_DIR=/usr/include/kismet/ 
x86_64-pc-linux-gnu-g++ -pthread -I/usr/include/ncurses -I/usr/include/libnl3  -Wall -Os -mtune=nocona -pipe -frecord-gcc-switches -Werror=format-security -I/usr/include/kismet/ -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packetsource_ubertooth.cc -o packetsource_ubertooth.o 
x86_64-pc-linux-gnu-g++ -pthread -I/usr/include/ncurses -I/usr/include/libnl3  -Wall -Os -mtune=nocona -pipe -frecord-gcc-switches -Werror=format-security -I/usr/include/kismet/ -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c packet_btbb_types.cc -o packet_btbb_types.o 
x86_64-pc-linux-gnu-g++ -pthread -I/usr/include/ncurses -I/usr/include/libnl3  -Wall -Os -mtune=nocona -pipe -frecord-gcc-switches -Werror=format-security -I/usr/include/kismet/ -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c phy_btbb.cc -o phy_btbb.o 
x86_64-pc-linux-gnu-g++ -pthread -I/usr/include/ncurses -I/usr/include/libnl3  -Wall -Os -mtune=nocona -pipe -frecord-gcc-switches -Werror=format-security -I/usr/include/kismet/ -I/usr/include/libusb-1.0 -I/usr/local/include/libusb-1.0 -I/usr/include -g -fPIC -c kismet_ubertooth.cc -o kismet_ubertooth.o 
packetsource_ubertooth.cc: In member function ‘virtual int PacketSource_Ubertooth::OpenSource()’:
packetsource_ubertooth.cc:267:12: error: cannot convert ‘ubertooth_t*’ to ‘libusb_device_handle*’ in assignment
  if ((devh = ubertooth_start(-1)) == NULL) {
            ^
packetsource_ubertooth.cc:280:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
packetsource_ubertooth.cc:288:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
packetsource_ubertooth.cc: In member function ‘virtual int PacketSource_Ubertooth::CloseSource()’:
packetsource_ubertooth.cc:317:22: error: cannot convert ‘libusb_device_handle*’ to ‘ubertooth_t*’ for argument ‘1’ to ‘void ubertooth_stop(ubertooth_t*)’
   ubertooth_stop(devh);
                      ^
Makefile:43: recipe for target 'packetsource_ubertooth.o' failed
make: *** [packetsource_ubertooth.o] Error 1
make: *** Waiting for unfinished jobs....
 * ERROR: net-wireless/kismet-ubertooth-9999::gentoo failed (compile phase):

dominicgs, 2015-12-02:
Ah, I know what's causing this, it's the PR that I merged yesterday. I'll
take a look later this week (I'm on vacation).

elli89, 2015-12-22:
Sorry, I forgot to change kismet_ubertooth when I created this PR. I've created a fix and just filed pull request #132. Unfortunately I don't have an ubertooth device here right now. Could you please check if my fix works?

ZeroChaos-, 2016-01-05:
fixed by https://github.com/greatscottgadgets/ubertooth/pull/132

closed 2016-01-05 by ZeroChaos-