greatscottgadgets/ubertooth #130: ubertooth-follow libUSB Error: Timeout:  (-7)

HannibalSmith13, 2015-12-04:
Hey,

when I try to use ubertooth-follow i continuously getting libUSB Errors. After the first occurrence I'm not able to use Ubertooth anymore until I reset it. This happens only when using ubertooth-follow.

> \# ubertooth-follow -lxxxx -uxx -a -r dd.pcapng -w7 -U0
Address given, assuming address is remote
Address: 00:00:xx:xx:xx:xx

Calculating complete hopping sequence.
Hopping sequence calculated.
AFH map: 0xffff0f000000ffffff7f
libUSB Error: Timeout:  (-7)

My Setup:

libbtbb (master git~525381)
ubertooth tools&firmware (2015-10-R1)
libusb (0.1.12)

tested on Linux Mint 17.2 & Kali GNU/Linux 2.0 with two Ubertooth One

dominicgs, 2015-12-06:
@HannibalSmith13 Did you clone libbtbb from git?  Or are you using the 2015-10-R1 release?  (the version number may be a bug if you're using the release)

HannibalSmith13, 2015-12-09:
@dominicgs 
I pulled libbtbb and ubertooth from git and tried the latest master as well as the latest release tag 2015-10-R1.

dominicgs, 2015-12-09:
Are you running Linux in a VM?

HannibalSmith13, 2015-12-11:
No, I've tried it with Mint as Host System. Only the tests with Kali were running in a VM environment.

dominicgs, 2015-12-14:
The timeout is caused by the Ubertooth not responding to the host quickly enough.  From the look of it, this is when we try to send the AFH pattern to the device.

I don't know what would be causing this to happen only for ubertooth-follow and nothing else, I don't see the issue with my version here, so I don't think it's an outright "sending AFH map does not work" sort of bug, but is probably something more subtle.

I'll try to reproduce the bug here, if you have any more details or observations that may help me to dot his, please let me know.

HannibalSmith13, 2015-12-15:
What exactly do you need?

I reproduced this with another Laptop and with Debian8, Kali2.0, Mint 17.2, Mint 17.1.

Sometimes I get this fault in ubertooth-rx too and when the fault appears, I have to wait a couple of seconds to be able to reset the device.

dominicgs, 2015-12-15:
Do you see the same problem with all USB ports?  Do you use a USB hub?

Could you try running with libUSB debug turned on?  You need to set LIBUSB_DEBUG to 3.
`LIBUSB_DEBUG=3 ubertooth_rx`

HannibalSmith13, 2015-12-15:
Yes I see it on all ports. Just checked it to be sure and no USB hub.

level 3 was a bit to silent, so I increased it to 10.

[libUSB-timeout.txt](https://github.com/greatscottgadgets/ubertooth/files/62576/libUSB-timeout.txt)

dominicgs, 2015-12-15:
@HannibalSmith13 Thank you for the output, it's very helpful.  Was this from ubertooth-rx or ubertooth-follow?

HannibalSmith13, 2015-12-16:
The output was produced by ubertooth-follow.

oxoocoffee, 2016-02-16:
I have something similar. If I connect ubertooth direct to my rMBP 15 USB port it works fine. It I connect to my Sorbent HB-UM43 USB 3.0 hub I get error below. I am not using latest github source.

14:29 $ ubertooth-rx -V
libubertooth 0.2 (GITDIR-NOTFOUND), libbtbb 0.3 (GITDIR-NOTFOUND)

ubertooth-util -v
Firmware revision: 2015-10-R1

13:42 $ LIBUSB_DEBUG=3  ubertooth-btle -f
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
control message unsupported
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libUSB Error: Command Error:  (-1)
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
libUSB Error: Command Error:  (-1)
libusb: error [submit_control_transfer] control request failed: unknown error (0xe00002d8)
libUSB Error: Command Error:  (-99)
libusb: error [submit_control_transfer] control request failed: unknown error (0xe00002d8)
libUSB Error: Command Error:  (-99)
USB error
libusb: error [submit_control_transfer] control request failed: unknown error (0xe00002d8)
libUSB Error: Command Error:  (-99)

I did try to measure amp draw and I see that ubertooth draw  0.08A @5V and with hub together (nothing else is plugged in to it) 0.13A @ 5V together. Also strange thing is that if I plug both of them in and sometimes ubertooth alone too (but less frequent) I get kicked out from WIFI. I know this port is fine since I plug and debug Android tablet and do not have problem. So it must be something else. 

Is there anything in github latest worth to try?

mike01, 2016-06-23:
Same problem here. Perhaps too much traffic lets Ubertooth halt?

closed 2020-12-03 by mikeryan