greatscottgadgets/ubertooth #131: Little errors under  IAR ARM compiler

sevstels, 2015-12-19:
Building configuration: ubertooth-one - Debug 
Updating build tree... 
usbhw_lpc.c  
Error[Pe042]: operand types are incompatible ("void (U16) *" and "void *") \Src\common\lpcusb\target\usbhw_lpc.c 546 
Error[Pe042]: operand types are incompatible ("void (U8) *" and "void *") \Src\common\lpcusb\target\usbhw_lpc.c 566 
Error[Pe042]: operand types are incompatible ("void (*)(U8, U8)" and "void *") \Src\common\lpcusb\target\usbhw_lpc.c 593 
Error while running C/C++ Compiler 
usbstdreq.c  
Warning[Pe177]: variable "bInterfaceNumber" was declared but never referenced \Src\common\lpcusb\target\usbstdreq.c 156 
Error[Pe042]: operand types are incompatible ("TFnHandleRequest *" and "void *") \Src\common\lpcusb\target\usbstdreq.c 584 
Error while running C/C++ Compiler 
bluetooth_rx-tx.c  
Warning[Pe068]: integer conversion resulted in a change of sign \Src\bluetooth_rx-tx.c 54 
Warning[Pe068]: integer conversion resulted in a change of sign \Src\bluetooth_rx-tx.c 55 
Error[Pe020]: identifier "INT8_MIN" is undefined \Src\bluetooth_rx-tx.c 187 
Error[Pe020]: identifier "INT8_MAX" is undefined \Src\bluetooth_rx-tx.c 188 
Warning[Pe177]: variable "ac_copy" was declared but never referenced \Src\bluetooth_rx-tx.c 381 
Error[Pe018]: expected a ")" \Src\bluetooth_rx-tx.c 1318 
Error[Pe020]: identifier "INT8_MIN" is undefined \Src\bluetooth_rx-tx.c 1500 
Warning[Pe177]: variable "i" was declared but never referenced \Src\bluetooth_rx-tx.c 1456 
Warning[Pe177]: variable "packet_offset" was declared but never referenced \Src\bluetooth_rx-tx.c 1457 
Warning[Pe174]: expression has no effect \Src\bluetooth_rx-tx.c 1627 
Error[Pe020]: identifier "INT8_MIN" is undefined \Src\bluetooth_rx-tx.c 1694 
Warning[Pe546]: transfer of control bypasses initialization of: \Src\bluetooth_rx-tx.c 1718 
            variable "ret" (declared at line 1767 of "\Src\bluetooth_rx-tx.c") 
Warning[Pe546]: transfer of control bypasses initialization of: \Src\bluetooth_rx-tx.c 1751 
            variable "ret" (declared at line 1767 of "\Src\bluetooth_rx-tx.c") 
Warning[Pe546]: transfer of control bypasses initialization of: \Src\bluetooth_rx-tx.c 1843 
            variable "p" (declared at line 1853 of "\Src\bluetooth_rx-tx.c") 
            variable "whit" (declared at line 1856 of "\Src\bluetooth_rx-tx.c") 
            variable "len" (declared at line 1865 of "\Src\bluetooth_rx-tx.c") 
            variable "total_transfers" (declared at line 1871 of "\Src\bluetooth_rx-tx.c") 
Warning[Pe546]: transfer of control bypasses initialization of: \Src\bluetooth_rx-tx.c 1867 
            variable "total_transfers" (declared at line 1871 of "\Src\bluetooth_rx-tx.c") 
Warning[Pe177]: variable "rssi" was declared but never referenced \Src\bluetooth_rx-tx.c 1788 
Warning[Pe177]: variable "adv_addr" was declared but never referenced \Src\bluetooth_rx-tx.c 2057 
Warning[Pe177]: variable "crc" was declared but never referenced \Src\bluetooth_rx-tx.c 2061 
Error[Pe513]: a value of type "int (*)()" cannot be assigned to an entity of type "data_cb_t" \Src\bluetooth_rx-tx.c 2371 
Warning[Pe177]: variable "epstat" was declared but never referenced \Src\bluetooth_rx-tx.c 2453 
Error[Pe167]: argument of type "uint8_t" is incompatible with parameter of type "ego_mode_t" \Src\bluetooth_rx-tx.c 2625 
Error while running C/C++ Compiler 
 
Total number of errors: 11 
Total number of warnings: 15

dominicgs, 2016-05-19:
While this is interesting, I don't plan to action any of these.  As I have said before, we target the GCC ARM compiler only.

If you have fixes to any of these warnings I will happily review a pull request, but for now I will close this issue.

closed 2016-05-19 by dominicgs

sevstels, 2016-05-20:
Hi.
Yes, I have fixs... but do not know how insert fix here. Need stady github doc. I ask pool litle later.