greatscottgadgets/ubertooth #133: specan and ui works but scan or util = Unable to find hci0 (-1)

ad5qa, 2015-12-27:
I have installed ubertooth build following the instructions. When I use the ubertooth-specan and ubertooth-specan-ui works however when I use ubertooth-scan none of the options work and display "error Unable to find hci0 (-1)". 

hcitool dev
Devices:

Any suggestions? I am running Kali 2.0 and all the libraries seemed to compile and install without errors.

mikeryan, 2015-12-27:
ubertooth-scan requires a normal Bluetooth device (e.g., builtin Bluetooth adapter, USB Bluetooth dongle). It looks like you don't have any Bluetooth devices in your Kali system. Either add a USB Bluetooth dongle, or if you're testing from a VM try sending your host's Bluetooth adapter into the VM system.

ad5qa, 2015-12-27:
Thanks, I connected the Bluetooth USB and it worked, overlooked that one.

closed 2015-12-27 by ad5qa

liut12, 2016-11-01:
I am curious what did you do to make it work?

dominicgs, 2016-11-04:
@liut12 I am unsure what you are asking.  This was a question rather than a bug.  The question was about tools that require a Bluetooth controller to be attached to the system.  The solution to make them work is, as Mike said, to attach a Bluetooth controller to the system.

If you are having a different problem, please open a new issue for it.