greatscottgadgets/ubertooth #135: ubertooth-rx fails to follow

elli89, 2016-01-06:
During my diploma thesis I found out that ubertooth-rx fails to follow a piconet most of the time. After debugging with blinking LEDs and a logic analyzer I found some serious bugs in the firmware.

1. Ubertooth also captures date while hopping. While hopping, the cc2400 sends no data for ~130 µs and also stops his SPI clock. So one USB packet could contain Bluetooth data from two different channels but tells the host only one channel. Also the bit offset, which I introduced in #84 can't be calculated correctly because of the ~130 µs gap between two hops.

My solution for this issue distinguishes between the two modes sweep and follow. While sweeping every hop sets a flag to the current USB packet's status that tells the host to discard this packet. When following, the time between two hops is much shorter (625µs) and only one packet (50 bytes = 400 bits = 400 µs) can be captured on each channel. So I start the capturing process right after each hop and stop it as soon as the DMA is full.

2. Hops are not equidistant when capturing data from the cc2400, because the main loop in the firmware function bt_stream_rx() first fills an entire USB packet before checking if a hop should be done. This is especially bad when following a piconet because the hops of the Ubertooth and the observed piconet are misaligned that way.

3. The channel of the USB packet is set when the packet is sent to the host. So the channel of the packet does not represent the actual channel of the captured data. My solution is setting the channel in the DMA interrupt so it represents the channel of the last bit of the packet data (which should be the channel of every bit in one packet).

4. When following, the firmware initializes the cc2400 with the function cc2400_rx_sync(u32 sync). This function tells the chip to only listen for packets that fit to the syncword. However the syncword is never set in the firmware. The function also tells the chip to work in a "packet mode". In that mode the cc2400 does not deliver a raw bitstream but only a matching bluetooth packet without an access code and without the crc checksum. So even if the correct syncword is set, the captured data would lack the access code which is necessary for libbtbb to detect bluetooth packets and do all the calculations for following. So I think even when following the firmware should initialize the cc2400 with the function cc2400_rx().

5. CLKN of the Ubertooth is not aligned to the clock of the piconet. So sometimes when following the Bluetooth access codes are always sent by the piconet when the Ubertooth is hopping. In that case no bluetooth packet is detected at all by libbtbb. The solution is to stretch/trim clk100ns before telling the Ubertooth to follow so the AC is always sent 225 µs (CLK_TUNE_TIME) after the beginning of a CLKN cycle. I achieved that by calculating the offset between the first bit of the access code and the first edge of CLKN and sending the value to the Ubertooth using a new command cmd_trim_clock().

I solved those issues in my experimental branch "fix-clock". Now ubertooth-rx is able to follow a piconet more reliably and captures much more packets from the piconet. Though sometimes it still fails. In about one third of all cases I observe a strange misalignment between the Bluetooth data and the Ubertooth clock even when not stretching/trimming clk100ns. In one pass of ubertooth-rx the alignment is good, when I stop the program and start it again, the Bluetooth data are misaligned by either 225 µs or -225 µs. Sometimes this sudden misalignment also occurs when changing from sweep to follow. I will continue working on that issue. Any hints or ideas are highly appreciated.

dominicgs, 2016-01-09:
@mikeryan and I have been taking a look at the diff between the fix-clock branch and master.  Overall we think the changes look good and would be happy to merge them.

However, we'd like to leave the `usb_pkt_types` enum in `ubertooth_interface.h`as it is because it's used by some code in Wireshark which we do not control but which is used by a large enough group of people that we don't want to break it at this time.

https://github.com/greatscottgadgets/ubertooth/compare/master...elli89:fix-clock?expand=1#diff-dfb36f6e858a3d1123c8f33f22575c05L107

elli89, 2016-01-12:
I totally agree. I changed a lot of things and left quite a mess in the code. I will clean it up and file a PR with only the changes that are necessary to fix the issues.

mikeryan, 2016-02-04:
Closing this issue as it appears to be resolved. Please reopen if that's not the case.

closed 2016-02-04 by mikeryan