greatscottgadgets/ubertooth #141: 2015-10-R1 Ubertooth-rx captures too few packets

dinesh-muvvala, 2016-01-14:
Hello,

Recently I have upgraded firmware of my Ubertooth-one device to 2015-10-R1 version and noticing that ubertooth-rx command captures too little traffic when compared to 2014-2-R1 release.

I am using the command following way:

$ sudo ubertooth-rx -U 0 -l e29a0d > capture_file.text

I tried to capture the Bluetooth classic logs side by side with two ubertooth's with two different versions of ubertooth firmware with old and new firmware versions above,

I see that everytime ubertooth captures made with older firmware 2014-02-R1 seems to capture good amount of traffic when I compare it with captures made with newer firmware.

I also upgraded the btbb library to the latest release.

When capturing traffic with new firmware I'm seeing different kinds of behaviour,

i) The capture seems to stall or happen very slowly once the device locks onto master's clocks CLK27,

     a) after this it shows few NULL or POLL or DM packets and shows USB error
     b) Sometimes there is no USB error but capture seems to happen quite slowly i.e. out 20 packets sent from device only one or two are captured

Can somebody help pls,

Best regards,
Dini

dinesh-muvvala, 2016-01-14:
[2015-10-R1_phs.txt](https://github.com/greatscottgadgets/ubertooth/files/90357/2015-10-R1_phs.txt)
[2015-10-R1_phs1.txt](https://github.com/greatscottgadgets/ubertooth/files/90359/2015-10-R1_phs1.txt)

[2014-02-R2_phs.txt](https://github.com/greatscottgadgets/ubertooth/files/90361/2014-02-R2_phs.txt)
[2014-02-R2_phs1.txt](https://github.com/greatscottgadgets/ubertooth/files/90362/2014-02-R2_phs1.txt)

elli89, 2016-02-10:
@dinesh-muvvala could you try using the current git master of libbtbb and ubertooth? I have committed some changes in the past few weeks and package capturing should work much better now.

closed 2020-12-03 by mikeryan